JuulLabs/kable #22: Ability to provide filters to the scanner

davidtaylor-juul, 2020-12-08:
The `Scanner` API presently exposes a flow of unfiltered peripherals which can of course be filtered after the fact. Internally the scan mechanisms typically provide for filtering *during* the scan process (part of the BLE spec) which is intended to be more efficient/save battery etc.

When instantiating the scanner, the API should allow injection of said filters to be fed through to the underlying systems.

burnhamd, 2021-06-19:
@twyatt I believe passing the service filter is required on iOS to be able to scan in background.  According to apple documentation 
_Your app can scan for Bluetooth devices in the background by specifying the bluetooth-central background mode. To do this, your app must explicitly scan for one or more services by specifying them in the serviceUUIDs parameter. The CBCentralManager scan option has no effect while scanning in the background._

Just wanted to mention this as it also blocks background scanning at least on iOS.

twyatt, 2021-06-19:
@burnhamd thanks for the heads up. This issue is getting a number of 👍  so I'll likely be re-prioritizing issues to be able to get to this one sooner.

closed 2021-06-25 by twyatt