JuulLabs/kable #36: Provide ability to request connection priority

twyatt, 2020-12-15:
[Android BLE API supports requesting various connection priorities](https://developer.android.com/reference/android/bluetooth/BluetoothGatt#requestConnectionPriority(int)). Kable should expose this functionality for platforms that support it.

Implemented in Able via https://github.com/JuulLabs/able/pull/95.

mtrewartha, 2021-06-23:
Hey, just checking on status for this... Any updates? We're experimenting with kable in our app and so far, liking it quite a bit! I think this is the one feature it's missing that we'd find really helpful, as we transfer large amounts of data over BLE and this would speed things up quite a bit. Thanks for the great library!

twyatt, 2021-06-23:
We try to prioritize issues based on number of 👍  &nbsp;. After each release we try to shuffle issues into appropriate target milestones based on their upvote count (as much as feasible at least).

That being said, internally we'll be wanting this feature as well, so we'll be getting to it sooner rather than later regardless of number of upvotes.

> Thanks for the great library!

Thanks for the kind words. 😄

mtrewartha, 2021-06-24:
Great, thanks for the update!

twyatt, 2021-06-24:
@mtrewartha looked into it really quick and found the implementation to be super simple. As mentioned in #125, `requestConnectionPriority` is a bit bizarre: AFAIK there isn't a way to know if the connection priority request was successful. The `requestConnectionPriority` itself returns a `Boolean`, but that only indicates if the request went to the OS, not if it was completed successfully (i.e. whether or not connection parameters were successfully negotiated with the remote peripheral). 🤷 

When you have a chance, you can give it a try:

```kotlin
repositories {
    maven("https://oss.sonatype.org/content/repositories/snapshots")
}

dependencies {
    implementation("com.juul.kable:core:0.5.1-issue-125-1-SNAPSHOT")
}
```

mtrewartha, 2021-06-24:
Works great 👍🏼 

Thanks again for the great library! The API is quite intuitive and so far, it's been reliable for me. I was able to replace all BLE code in our app and move it from our Android app module to a shared module in a matter of a day, all thanks to this library.

PR #125 (2021-06-24): Support requesting connection priority on Android
Closes #36

Unlike nearly every other BLE request on Android, this one does **not** (or at least it isn't documented to) invoke a function on [`BluetoothGattCallback`](https://developer.android.com/reference/android/bluetooth/BluetoothGattCallback) to indicate when the process is complete (and whether successful or not). It seems you just have to blindly hope it worked. 🤷 

Per [Android ble, requestConnectionPriority not working](https://stackoverflow.com/a/47491885), it is recommended that `requestConnectionPriority` not be called during servi [... cut]
closed 2021-06-24 by twyatt