JuulLabs/kable #42: Add notification setup mode for devices that do not contain client characteristic config descriptor

kostya29-strikersoft, 2020-12-23:
There are some BLE devices which does not have client characteristic config descriptor. It's a violation of BLE specification and the bug of the firmware of peripheral, but there is no possibility to change it for mobile developers. For a workaround, it's good to have the possibility to subscribe on notification by only setCharacteristicNotification(characteristic, boolean) call without writing to the descriptor.
Library RXAndroidBle has this possibility (NotificationSetupMode.Compact) and earlier I have used it but unfortunately, it's not friendly for kotlin implementation.

twyatt, 2020-12-23:
Would it be enough to toggle this option for all observations for a `Peripheral` object, or is there a need to not write to the descriptor only for some observes for a specific `Peripheral`.

In others, trying to decide if it would make sense to allow a configuration of something like:

```kotlin
val peripheral = scope.peripheral {
    writeObserveDescriptors = false
}
```

...or, should it be an optional argument of the [`Peripheral.observe`](https://juullabs.github.io/kable/core/core/com.juul.kable/-peripheral/index.html#%5Bcom.juul.kable%2FPeripheral%2Fobserve%2F%23com.juul.kable.Characteristic%2FPointingToDeclaration%2F%5D%2FFunctions%2F-328684452) function?

kostya29-strikersoft, 2020-12-23:
@twyatt I think the first option will be enough to toggle this option for all observations for a Peripheral.

twyatt, 2020-12-23:
> I think the first option will be enough to toggle this option for all observations for a Peripheral.

Great, thanks for the feedback. 😄 

> When you can add it and republish lib?

Sorry, don't really have a timeline for this issue. I want to look into how RxAndroidBle supports this and make a clear decision on what the shape of the API should be for this.

I'll try to get to it soon, but can't make any promises.

kostya29-strikersoft, 2020-12-23:
@twyatt thanks! RxAndroidBle supports this as the optional parameter for each setup notification method and checks if this parameter is true do not execute writing to the descriptor.

twyatt, 2020-12-23:
Preliminary research into RxAndroidBle's implementation to support this feature:

- [NotificationSetupMode.java](https://github.com/Polidea/RxAndroidBle/blob/75b854fa028380d0d6cb44ec9c4c2f27ac5533bf/rxandroidble/src/main/java/com/polidea/rxandroidble2/NotificationSetupMode.java)
- [RxBleConnectionImpl.java#L163-L173](https://github.com/Polidea/RxAndroidBle/blob/8550bd66493c8e12449c784ca28ed0b167e67fc3/rxandroidble/src/main/java/com/polidea/rxandroidble2/internal/connection/RxBleConnectionImpl.java#L163-L173)

twyatt, 2020-12-29:
Exploring providing this configuration builder. Current experimental implementation on branch [`twyatt/configuration`](https://github.com/JuulLabs/kable/tree/twyatt/configuration).

Initial impression is that builder adds a lot of unnecessary complexity (haven't yet explored other configurations that might be needed on other platforms, but `writeObserveDescriptors` is only needed for Android).

Considering as a simpler alternative is adding optional parameter to `CoroutineScope.peripheral` functions in Android source set.

kostya29-strikersoft, 2020-12-29:
adding the optional parameter is more flexible and easier to implement. Please explain how I can add the library from the branch to my project? I don't have experience with kotlin multiplatform before, I have tried yesterday to clone project and build artifacts but found that core-release.aar artifact does not enough...

twyatt, 2020-12-29:
> adding the optional parameter is more flexible and easier to implement.

That's what I'm leaning towards so far.

> Please explain how I can add the library from the branch to my project?

Sorry, was more to just share a status update. I can publish the branch (as a SNAPSHOT) if you want to give it a try?

> I have tried yesterday to clone project and build artifacts but found that core-release.aar artifact does not enough...

Easiest way to try it locally, is to clone, then switch to the branch. Then you can install it to your local maven: `./gradlew installArchives`. This will install it to your local Maven as version `main-SNAPSHOT`. In your project, if you add `mavenLocal()` as a repository then it should be able to pull in the `main-SNAPSHOT` version.

twyatt, 2020-12-31:
@kostya29-strikersoft when you have a chance, can you try using the following dependency:

```kotlin
import java.net.URI

repositories {
    maven { url = URI("https://oss.sonatype.org/content/repositories/snapshots") }
}

dependencies {
    implementation("com.juul.kable:core:0.1.0-issue-42-SNAPSHOT")
}
```

The default behavior is to only write to the config descriptor if it's available, so you shouldn't need to change your code (it should _just work_ 🤞 ).

twyatt, 2021-01-10:
Wrote a quick GATT server test app that sends out notifications on a characteristic that does **not** have an associated config descriptor.

From Core Bluetooth (macOS), attempting to observe the characteristic results in a crash:

```
Uncaught Kotlin exception: com.juul.kable.IOException: Error Domain=CBErrorDomain Code=1 "One or more parameters were invalid." UserInfo={NSLocalizedDescription=One or more parameters were invalid.}
...
```

The BLE sniffer showed the error response (due to absence of config characteristic):

<details>
<summary>Wireshark log</summary>

```
Bluetooth Attribute Protocol
    Opcode: Error Response (0x01)
        0... .... = Authentication Signature: False
        .0.. .... = Command: False
        ..00 0001 = Method: Error Response (0x01)
    Request Opcode in Error: Read By Type Request (0x08)
        0... .... = Authentication Signature: False
        .0.. .... = Command: False
        ..00 1000 = Method: Read By Type Request (0x08)
    Handle in Error: 0x002b (Unknown: Unknown: Unknown)
        [Service UUID: 44158efc2d3a4425ad1999c801fe6a76]
        [Characteristic UUID: 3a59e5305bb1445a9ec53e752857f9f1]
    Error Code: Attribute Not Found (0x0a)
    [UUID: GATT Characteristic Declaration (0x2803)]
```
</details>

On Web Bluetooth (Chrome), results in the following error:

```
DOMException: GATT Error: Not supported.
```

It appears that neither Web Bluetooth nor Core Bluetooth (macOS) are permissive about observing a characteristic **without** the config descriptor.

---

**tl;dr**: I'll likely change the default to `Always` to have all platforms behave the same (by default).

PR #49 (2021-01-11): Add ability to configure writing to config descriptor for observations
Adds ability to configure if the config descriptor for a characteristic is written to when starting/stopping an observation.

> ~I do not have a peripheral that supports notifications on a characteristic that is missing the config descriptor, so I am unable to testes this, nor can I confirm behavior for Apple and JavaScript platforms (which do not have this as configurable).~

**UPDATE:** Tested via Android test app (see https://github.com/JuulLabs/kable/issues/42#issuecomment-757428891 for details).

> ~I suspect that Appl [... cut]
closed 2021-01-11 by twyatt