JuulLabs/kable #44: Force disconnecting peripheral causes crash by ConnectionLostException

kostya29-strikersoft, 2020-12-28:
Steps to reproduce:
Start connecting to peripheral by method peripheral.connect and quickly stop Bluetooth server on the peripheral side.

Stack trace:
`com.juul.kable.ConnectionLostException
        at com.juul.kable.PeripheralKt$suspendUntilConnected$2.invokeSuspend(Peripheral.kt:220)
        at com.juul.kable.PeripheralKt$suspendUntilConnected$2.invoke(Unknown Source:10)
        at kotlinx.coroutines.flow.FlowKt__TransformKt$onEach$$inlined$unsafeTransform$1$2.emit(Collect.kt:134)
        at kotlinx.coroutines.flow.StateFlowImpl.collect(StateFlow.kt:348)
        at kotlinx.coroutines.flow.StateFlowImpl$collect$1.invokeSuspend(Unknown Source:12)
        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
        at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
        at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)`

twyatt, 2020-12-28:
What platform are you seeing this on, and what function call are you seeing it thrown from (`connect` or `disconnect`)?

kostya29-strikersoft, 2020-12-28:
@twyatt, Android. Connect. It happens when disconnect happens before discovering services.
this method generates exception

`private suspend fun Peripheral.suspendUntilConnected() {
    state
        .onEach { if (it is State.Disconnected) throw ConnectionLostException() }`
and there are not any handlers of it

twyatt, 2020-12-28:
It is intended behavior that `connect` would throw an exception if the connection fails (or is cancelled) as there needs to be some way to inform the caller of `connect` of the failure state.

However it should **not** throw at the `disconnect` call-site (which it appears that it currently is).

kostya29-strikersoft, 2020-12-28:
@twyatt do you have ideas what shoud I do?

twyatt, 2020-12-28:
Really depends on what your use-case is. I assume that in your code a connection failure can be safely ignored? In which case you could have an extension function similar to:

```kotlin
fun Peripheral.connectOrFalse(): Boolean = try {
    connect()
    true
} catch (e: IOException) {
    false
}
```

...but it really depends on how you want your code (control flow) to act when a connection isn't successful.

kostya29-strikersoft, 2020-12-28:
I already have try catch around peripheral.connect but despite of this crash is happened

twyatt, 2020-12-28:
> I already have try catch around peripheral.connect but despite of this crash is happened

Oh, I see. _That_ is not intended behavior.

Do you have a simple code snippet that reproduces the problem? Unfortunately with the nature of BLE I know these issues can be hard to demonstrate. 😢 

I'll try to reproduce on my end. Simple `connect` then disconnect on the peripheral side, right?

kostya29-strikersoft, 2020-12-28:
@twyatt yes

kostya29-strikersoft, 2020-12-28:
@twyatt Also when I try often in raw connect & disconnect on the peripheral side I get the crash with `kotlinx.coroutines.flow.internal.AbortFlowException: Flow was aborted, no more elements needed` exception

kostya29-strikersoft, 2020-12-28:
@twyatt I see in code of Android Peripheral and think the reason of unintended ConnectionLostException - that connectJob is live after connect call is finished and `state
        .onEach { if (it is State.Disconnected) throw ConnectionLostException()` continue works

kostya29-strikersoft, 2020-12-28:
@twyatt Or perhaps need call cancel for connectJob below (like `if (connectJob.value?.isActive) connectJob.value?.cancel(null) connectJob.value = null`)
`  private fun establishConnection(): Connection =
        bluetoothDevice.connect(
            applicationContext,
            _state,
            invokeOnClose = { connectJob.value = null }
        ) ?: throw ConnectionRejectedException()`

twyatt, 2020-12-28:
Thanks for all the info, and good suggestions. Looking into it. 👍

kostya29-strikersoft, 2020-12-28:
@twyatt Some more info. When the app disconnects from the peripheral I see following bluetoothgatt logs:
`
BluetoothGatt: close()
BluetoothGatt: unregisterApp() - mClientIf=10
BluetoothGatt: close()
BluetoothGatt: unregisterApp() - mClientIf=0
`
you can see gatt close calls twice but it should be one time I think or not?
For example, well-known BLE app - "NRF connect" when I connect to the same peripheral and disconnects from it 
calls BluetoothGatt: close only one time

twyatt, 2020-12-28:
The dual `close` calls is because it is trigger by multiple conditions within Kable. We could add a checks to prevent the multiple calls but I don't believe extra calls to `close` have any effect (i.e. are not harmful).

kostya29-strikersoft, 2020-12-28:
@twyatt I have reproduced the crash with ConnectionLostException with SensorTag example (I have only added try catch around peripheral.connect()).

twyatt, 2020-12-28:
> @twyatt I have reproduced the crash with ConnectionLostException with SensorTag example (I have only added try catch around peripheral.connect()).

Are you having the SensorTag cancel the connection by holding the power button?
I wasn't able to reproduce with that process, though timing is hard to get right (I kept seeing the peripheral drop the connection during service discovery, whereas it propagated the exception as expected).

kostya29-strikersoft, 2020-12-28:
@twyatt I have used android app as peripheral - SensorTag, and I have button: start, stop Bluetooth server, so I can press Stop very quickly.

twyatt, 2020-12-28:
> @twyatt I have used android app as peripheral - SensorTag, and I have button: start, stop Bluetooth server, so I can press Stop very quickly.

Do you happen to have a project available that acts as Bluetooth server that I could use to reproduce?
If not, no worries, I can draft something up when I have time.

twyatt, 2020-12-28:
I've found a way to reproduce the issue on my side. Currently investigating a fix.

twyatt, 2020-12-28:
@kostya29-strikersoft I believe I have a fix that makes `connect` properly propagate the connection failures.
Can you try it out and let me know if it works as expected?

```kotlin
import java.net.URI

repositories {
    maven { url = URI("https://oss.sonatype.org/content/repositories/snapshots") }
}

dependencies {
    implementation("com.juul.kable:core:0.1.0-issue-44-SNAPSHOT")
}
```

Thanks!

kostya29-strikersoft, 2020-12-29:
@twyatt  yes I could not reproduce this crash but I have found another crash when does connect/disconnect a lot of times in raw.
`kotlinx.coroutines.flow.internal.AbortFlowException: Flow was aborted, no more elements needed`
Unfortunately that there are not other lines of the stack trace. 
It happens just after call method connect.

`kotlinx.coroutines.flow.internal.AbortFlowException: Flow was aborted, no more elements needed
2020-12-29 12:39:45.027 6461-6511/com.xx D/BluetoothGatt: connect() - device: 43:95:9A:31:C7:6C, auto: false
2020-12-29 12:39:45.028 6461-6511/com.xx D/BluetoothAdapter: isSecureModeEnabled
2020-12-29 12:39:45.029 6461-6511/com.xx D/BluetoothGatt: registerApp()
2020-12-29 12:39:45.030 6461-6511/com.xx D/BluetoothGatt: registerApp() - UUID=1228b816-0f28-48f8-a49d-ef4c752b9027
2020-12-29 12:39:45.034 6461-6476/com.xx D/BluetoothGatt: onClientRegistered() - status=0 clientIf=10`

twyatt, 2020-12-29:
Thanks for confirming the fix @kostya29-strikersoft. Once #46 is approved (and merged) then we'll probably release [`0.1.1`](https://github.com/JuulLabs/kable/milestone/3).

I've created #47 to continue discussion re: `AbortFlowException` issue you discovered.

---

Thanks for reporting all these issues and providing details!

PR #46 (2020-12-30): Fix propagation of ConnectionLostException on connect
Was previously using `launch` internally to establish connection. An early failure (before `join` was called) would cause the failure to propagate to the parent scope (rather than the `connect` caller). Switched to using `async` so that failures are properly propagated on call to `await` (within `connect`).

Fixes #44
commit d040c41f (2020-12-30): Fix propagation of ConnectionLostException on connect (#46)

Fixes #44
closed 2020-12-30 by twyatt