JuulLabs/kable #47: AbortFlowException crash on repeated connect/disconnect calls

twyatt, 2020-12-29:
@twyatt  yes I could not reproduce this crash but I have found another crash when does connect/disconnect a lot of times in raw.
`kotlinx.coroutines.flow.internal.AbortFlowException: Flow was aborted, no more elements needed`
Unfortunately that there are not other lines of the stack trace. 
It happens just after call method connect.

`kotlinx.coroutines.flow.internal.AbortFlowException: Flow was aborted, no more elements needed
2020-12-29 12:39:45.027 6461-6511/com.xx D/BluetoothGatt: connect() - device: 43:95:9A:31:C7:6C, auto: false
2020-12-29 12:39:45.028 6461-6511/com.xx D/BluetoothAdapter: isSecureModeEnabled
2020-12-29 12:39:45.029 6461-6511/com.xx D/BluetoothGatt: registerApp()
2020-12-29 12:39:45.030 6461-6511/com.xx D/BluetoothGatt: registerApp() - UUID=1228b816-0f28-48f8-a49d-ef4c752b9027
2020-12-29 12:39:45.034 6461-6476/com.xx D/BluetoothGatt: onClientRegistered() - status=0 clientIf=10`

_Originally posted by @kostya29-strikersoft in https://github.com/JuulLabs/kable/issues/44#issuecomment-752030738_

kostya29-strikersoft, 2020-12-29:
I have attached simple android peripheral app. Name of this peripheral "SensorTag" :-)
[peripheral.apk.zip](https://github.com/JuulLabs/kable/files/5750440/peripheral.apk.zip)

twyatt, 2020-12-29:
Without a stacktrace I'm not totally confident what the issue is, but looks closely related to https://github.com/Kotlin/kotlinx.coroutines/issues/1454. Hopefully fixed in #48. 🤞 

@kostya29-strikersoft can you test it out and see if it fixes the issue?

```kotlin
import java.net.URI

repositories {
    maven { url = URI("https://oss.sonatype.org/content/repositories/snapshots") }
}

dependencies {
    implementation("com.juul.kable:core:0.1.0-issue-47-SNAPSHOT")
}
```

Thanks so much!

kostya29-strikersoft, 2020-12-30:
@twyatt Now the app does not get disconnected state from peripheral.state flow after BluetoothGatt: close() BluetoothGatt: unregisterApp()

twyatt, 2020-12-30:
> @twyatt Now the app does not get disconnected state from peripheral.state flow after BluetoothGatt: close() BluetoothGatt: unregisterApp() but don't see the crash with abortflowexception. Please fix propagating peripheral state

Good to know, thanks! I'll look into it.

mattbutlar, 2021-02-26:
@twyatt Just wanted to mention that I am seeing this issue and it is making it impossible to move forward on anything... sometime I get it to connect and the exception isn't thrown, but most of the time it throws the AbortFlowException right after registering the app, and no try/catch prevents it from crashing the app, and there is no stacktrace of any sort.  It is pretty horrible.

twyatt, 2021-02-26:
@mattbutlar thanks for sharing. I'll get the associated PR and a version pushed out either today or tomorrow.

mattbutlar, 2021-02-26:
@twyatt I did try that snapshot and saw no improvement just so you know

twyatt, 2021-02-26:
> @twyatt I did try that snapshot and saw no improvement just so you know

Oh, yikes. Ok, I'll need to investigate further then. Super helpful to know that the SNAPSHOT didn't help.

twyatt, 2021-02-26:
@mattbutlar is your failure occurring on connect, or during scan? Or, more specifically, what method call is causing the crash?

mattbutlar, 2021-02-26:
@twyatt I hadn't see #67 but I just tried it's snapshot and it didn't crash, so it looks like it was the scan even though it happened well after connection had started

twyatt, 2021-02-26:
> @twyatt I hadn't see #67 but I just tried it's snapshot and it didn't crash, so it looks like it was the scan even though it happened well after connection had started

Great, thanks for checking. I'll get a version (that has the #67 fix) released ASAP.

twyatt, 2021-02-27:
If anyone continues to see this issue in `0.3.0` (or newer) can you please comment on this issue?

_I have a suspicion that #67 might have been the cause for the crash reported in this issue, and it was just observed after a short delay making it seem like it was due to quick reconnect (rather than from the scanner)._

Based on https://github.com/JuulLabs/kable/issues/47#issuecomment-752360748 I'm hesitant to merge #48 until I've received confirmation that this is still an issue (and warrants further investigation).

zhourenjun, 2021-03-18:
0.3.0   If the device is far away from the disconnection and close to the reconnection, the crash will reproduce after multiple operations

twyatt, 2021-03-18:
@zhourenjun thanks for reporting that issue still exists.

Can you try the following SNAPSHOT and let me know if it fixes the issue?

```kotlin
import java.net.URI

repositories {
    maven { url = URI("https://oss.sonatype.org/content/repositories/snapshots") }
}

dependencies {
    implementation("com.juul.kable:core:0.3.0-issue-47-SNAPSHOT")
}
```

Thank you!

zhourenjun, 2021-03-19:
Still not working, I crashed the second time I reconnected, and my log did not capture the specific reason

twyatt, 2021-03-19:
If at all possible to get the failure (and ideally a stacktrace) then it'd be super helpful. Otherwise it'll be a bit challenging to track down.

In the mean time, I'll try to find some time to reproduce.

zhourenjun, 2021-03-19:
2021-03-19 11:43:14.967 20449-20449/? I/.juul.sensorta: Late-enabling -Xcheck:jni
2021-03-19 11:43:14.979 20449-20449/com.juul.sensortag E/.juul.sensorta: Unknown bits set in runtime_flags: 0x8000
2021-03-19 11:43:15.368 20449-20449/com.juul.sensortag W/.juul.sensorta: Accessing hidden method Landroid/view/View;->computeFitSystemWindows(Landroid/graphics/Rect;Landroid/graphics/Rect;)Z (greylist, reflection, allowed)
2021-03-19 11:43:15.368 20449-20449/com.juul.sensortag W/.juul.sensorta: Accessing hidden method Landroid/view/ViewGroup;->makeOptionalFitsSystemWindows()V (greylist, reflection, allowed)
2021-03-19 11:43:15.388 20449-20449/com.juul.sensortag D/SensorTag: Scan status: com.juul.sensortag.features.scan.ScanStatus$Stopped@46b4028
2021-03-19 11:43:15.393 20449-20449/com.juul.sensortag I/SurfaceFactory: [static] sSurfaceFactory = com.mediatek.view.impl.SurfaceFactoryImpl@d7de427
2021-03-19 11:43:15.398 20449-20449/com.juul.sensortag D/ViewRootImpl[ScanActivity]: hardware acceleration = true , fakeHwAccelerated = false, sRendererDisabled = false, forceHwAccelerated = false, sSystemRendererDisabled = false
2021-03-19 11:43:15.402 20449-20449/com.juul.sensortag V/PhoneWindow: DecorView setVisiblity: visibility = 0, Parent = android.view.ViewRootImpl@ada9bc3, this = DecorView@65bec40[ScanActivity]
2021-03-19 11:43:15.415 20449-20449/com.juul.sensortag E/GraphicExt: Can't load libboost_ext_fwk
2021-03-19 11:43:15.415 20449-20449/com.juul.sensortag E/GraphicExt: GraphicExtModuleLoader::CreateGraphicExtInstance false
2021-03-19 11:43:15.416 20449-20480/com.juul.sensortag I/GPUD: @gpudInitialize: successfully initialized with GL, dbg=0 mmdump_dbg=0 mmpath_dbg=0
2021-03-19 11:43:15.421 20449-20480/com.juul.sensortag I/GED: [GT]_get_procNameprocess pid(20449)
2021-03-19 11:43:15.421 20449-20480/com.juul.sensortag I/GED: [GT]_getprocess name(com.juul.sensortag)
2021-03-19 11:43:15.424 20449-20480/com.juul.sensortag D/Surface: Surface::connect(this=0x73b6392000,api=1)
2021-03-19 11:43:15.425 20449-20480/com.juul.sensortag D/mali_winsys: EGLint new_window_surface(egl_winsys_display *, void *, EGLSurface, EGLConfig, egl_winsys_surface **, EGLBoolean) returns 0x3000
2021-03-19 11:43:15.425 20449-20480/com.juul.sensortag D/Surface: Surface::setBufferCount(this=0x73b6392000,bufferCount=3)
2021-03-19 11:43:15.425 20449-20480/com.juul.sensortag D/Surface: Surface::allocateBuffers(this=0x73b6392000)
2021-03-19 11:43:15.430 20449-20480/com.juul.sensortag W/Gralloc3: mapper 3.x is not supported
2021-03-19 11:43:15.431 20449-20480/com.juul.sensortag E/gralloc: Arm Module v1.0
2021-03-19 11:43:15.432 20449-20480/com.juul.sensortag E/ion: ioctl c0044901 failed with code -1: Invalid argument
2021-03-19 11:43:15.432 20449-20480/com.juul.sensortag W/gralloc: WARNING: internal format modifier bits not mutually exclusive. AFBC basic bit is always set, so extended AFBC support bits must always be checked.
2021-03-19 11:43:19.333 20449-20449/com.juul.sensortag D/SensorTag: Scan status: com.juul.sensortag.features.scan.ScanStatus$Started@a68e89c
2021-03-19 11:43:19.358 20449-20449/com.juul.sensortag D/BluetoothAdapter: isLeEnabled(): ON
2021-03-19 11:43:19.360 20449-20467/com.juul.sensortag D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=11 mScannerId=0
2021-03-19 11:43:22.411 20449-20449/com.juul.sensortag D/BluetoothAdapter: isLeEnabled(): ON
2021-03-19 11:43:22.463 20449-20449/com.juul.sensortag W/ActivityThread: handleWindowVisibility: no activity for token android.os.BinderProxy@8c30db8
2021-03-19 11:43:22.483 20449-20449/com.juul.sensortag W/SensorTag: State Disconnected(status=null)
2021-03-19 11:43:22.483 20449-20449/com.juul.sensortag I/SensorTag: Waiting 250 ms to reconnect...
2021-03-19 11:43:22.484 20449-20449/com.juul.sensortag I/SensorTag: connect
2021-03-19 11:43:22.487 20449-20449/com.juul.sensortag D/BluetoothGatt: connect() - device: D1:D3:2A:C5:C3:16, auto: false
2021-03-19 11:43:22.487 20449-20449/com.juul.sensortag D/BluetoothGatt: registerApp()
2021-03-19 11:43:22.487 20449-20449/com.juul.sensortag D/BluetoothGatt: registerApp() - UUID=73db7297-ac88-4018-a7bb-8c06758aabe6
2021-03-19 11:43:22.488 20449-20467/com.juul.sensortag D/BluetoothGatt: onClientRegistered() - status=0 clientIf=11
2021-03-19 11:43:22.512 20449-20449/com.juul.sensortag V/PhoneWindow: DecorView setVisiblity: visibility = 4, Parent = null, this = DecorView@f414bf5[]
2021-03-19 11:43:22.514 20449-20449/com.juul.sensortag D/ViewRootImpl[SensorActivity]: hardware acceleration = true , fakeHwAccelerated = false, sRendererDisabled = false, forceHwAccelerated = false, sSystemRendererDisabled = false
2021-03-19 11:43:22.516 20449-20449/com.juul.sensortag V/PhoneWindow: DecorView setVisiblity: visibility = 0, Parent = android.view.ViewRootImpl@a97c818, this = DecorView@f414bf5[SensorActivity]
2021-03-19 11:43:22.527 20449-20449/com.juul.sensortag E/GraphicExt: GraphicExtModuleLoader::CreateGraphicExtInstance false
2021-03-19 11:43:22.528 20449-20480/com.juul.sensortag D/Surface: Surface::connect(this=0x73b7f0d000,api=1)
2021-03-19 11:43:22.529 20449-20480/com.juul.sensortag D/mali_winsys: EGLint new_window_surface(egl_winsys_display *, void *, EGLSurface, EGLConfig, egl_winsys_surface **, EGLBoolean) returns 0x3000
2021-03-19 11:43:22.529 20449-20480/com.juul.sensortag D/Surface: Surface::setBufferCount(this=0x73b7f0d000,bufferCount=3)
[... cut ...]
2021-03-19 11:46:36.250 20449-20467/com.juul.sensortag D/BluetoothGatt: onConnectionUpdated() - Device=D1:D3:2A:C5:C3:16 interval=168 latency=0 timeout=600 status=0
2021-03-19 11:46:41.016 20449-20480/com.juul.sensortag D/Surface: Surface::disconnect(this=0x73b7f0d000,api=1)
2021-03-19 11:46:41.030 20449-20480/com.juul.sensortag I/GED: ged_boost_gpu_freq, level 100, eOrigin 2, final_idx 27, oppidx_max 27, oppidx_min 0
2021-03-19 11:46:41.039 20449-20449/com.juul.sensortag V/PhoneWindow: DecorView setVisiblity: visibility = 4, Parent = android.view.ViewRootImpl@a97c818, this = DecorView@f414bf5[SensorActivity]

twyatt, 2021-03-19:
@zhourenjun I believe you're referring to the `ConnectionLostException` you're seeing?

I'm guessing you're seeing this propagate from a `connect` call (which aligns with your description of it happening when peripheral is nearly out-of-range and attempting to reconnect)?

This is by design, to propagate the fact that the connection was lost while attempting to connect.

I see that the logs are from the SensorTag example app...I should update that sample to demonstrate how to handle this failure from within sample.

How _you_ handle it (within your app) really depends on what you want to do in the event that a connection attempt fails.

In the SensorTag sample, since we re-attempt (with exponential back-off) to connect by listening to the `state` Flow, we can generally ignore this failure, since the reconnect logic should kick in for us.

In the case of the SensorTag sample app, the code may look something like:

```kotlin
try {
    peripheral.connect()
    sensorTag.enableGyro()
    sensorTag.writeGyroPeriodProgress(periodProgress.get())
} catch (e: ConnectionLostException) {
    // Silently ignore connection lost as we'll reconnect automatically.
    return@launch
}
```

...but, for your app, it really depends on how you want to handle a connect failure.

twyatt, 2021-06-24:
I don't believe this issue still exists in the latest versions of Kable. Please comment on this issue if I am mistaken (and it _is_ still an issue).

closed 2021-06-24 by twyatt