JuulLabs/kable #50: Conversion Guide Able -> Kable

sarn0ld, 2021-01-04:
Is there something like a tutorial how to move from Able to Kable?

twyatt, 2021-01-04:
There is not. It's a good idea though.

To get an idea of the guide should include: did you use the standard Able module (`core`) or other modules of Able as well (e.g. `keep-alive`, `processor`, `throw`, etc)?

sarn0ld, 2021-01-04:
Thank you for the fast reply and the good work!
A little guide would be nice to quickly see how much effort it takes to exchange Able.

So far we just used the core module to connect, readCharacteristics, writeCharacteristics and observe characteristics, ...

twyatt, 2021-04-26:
@sarn0ld I've written a guide that essentially compares the functionality of Able and Kable, which should hopefully provide an understanding of what is involved in migrating from Able to Kable. Let me know if it is what you had in mind (and if it is helpful), as well as any gaps that it may have.

https://github.com/JuulLabs/able/pull/97

Thank you!

twyatt, 2021-04-29:
[Migration guide](https://github.com/JuulLabs/able/blob/main/migration.md) is available on the Able repo. Let me know if anyone wants any further details added to it.

closed 2021-04-29 by twyatt

sarn0ld, 2021-04-29:
Hey @twyatt !
Thank you for your efford! The guide looks very good and encouraged us to discuss switching to Kable again.
We will have to discuss this with our customer first and that will take a while, but it sounds like Kable could fix some trouble we are facing at the moment (with scanning and asnyc waiting for characteristic changes).

Thank you again! ❤️

twyatt, 2021-04-29:
Great to hear. Happy to add more information if you find there are gaps in the guide. I'm also available on kotlinlang Slack (in `#juul-libraries`).