JuulLabs/kable #51: Kable to support advertising as a peripheral

tom-pratt, 2021-01-05:
Is it or will it be possible in the future to use Kabel to adertise the device as the peripheral? e.g. a mobile app that can communicate with other instances of itself on multiple phones.

twyatt, 2021-01-05:
There are no short term plans to have Kable act as a GATT server, though if Kable's feature set becomes more complete I could see us working on such support.

One of the hurdles would be that (at least AFAIK) JavaScript does not have GATT server support, so the library platform support would have to diverge a bit.

We'll continue to focus on the GATT client side for a while, so don't expect much movement on this issue for some time; but I do like the idea of Kable being a full turn key BLE solution, so hopefully some day we'll be able to support this.

Happy to discuss possible designs if the community wants to try and tackle this effort and submit PRs.

tom-pratt, 2021-01-06:
Thanks for the info. Not actually an essential feature for me since the peripheral side is Android only in my project.

Have barely got started so not super familiar with BLE concepts. But my understanding was that being peripheral/central device is not coupled to gatt server/client. e.g. peripheral can be server or client. Is that true? At least that was my interpretation of these docs https://developer.android.com/guide/topics/connectivity/bluetooth-le#roles

twyatt, 2021-01-06:
> But my understanding was that being peripheral/central device is not coupled to gatt server/client. e.g. peripheral can be server or client. Is that true?

That is my understanding as well.

I suppose my response was a bit wordy; merely trying to convey that things get a little muddy (from an API perspective) to introduce "support advertising as a peripheral", purely because BLE in general is a bit convoluted. So, just feel it takes a bit of care to build up the API to support the requested feature **and do it in a way that is intuitive to developers using Kable**.