JuulLabs/kable #52: Support iOS target?

francismariano, 2021-01-06:


tom-pratt, 2021-01-06:
Beat me to it. I was just coming here to ask the same question :D

Would love to use this in my current project and just focus on Android for a while until iOS Kable support is available. Doesn't need to be 100% stable, happy to provide testing and feedback, just keen to go multiplatform from the beginning on this new project!

twyatt, 2021-01-06:
We plan to add iOS targets soon. Being that we already support macOS target it should only be minor (if any) changes to add additional Apple related targets.

I currently only have access to an older iOS phone, so I'll be testing against it for adding iOS target. Then I'll either have to acquire other Apple devices (such as Apple watch, etc) or have the community help with testing/validating for additional Apple based targets.

francismariano, 2021-01-28:
Excited for waiting the milestone 0.3.0

twyatt, 2021-02-01:
I have not tested it yet, but for those who have time to test it and report back, it'd be much appreciated:

```kotlin
import java.net.URI

repositories {
    maven { url = URI("https://oss.sonatype.org/content/repositories/snapshots") }
}

dependencies {
    implementation("com.juul.kable:core:0.2.0-issue-52-SNAPSHOT")
}
```

The following targets are available:

https://github.com/JuulLabs/kable/blob/7ce5b55d61c910fe277d228e00363b75c0ad6e58/core/build.gradle.kts#L14-L20

francismariano, 2021-02-17:
Hello @twyatt 

I have tested the SNAPSHOT above with Kotlin 1.4.30 on iOS 14. Scan, connection, write (WITH_RESPONSE), read and notification work like a charm. In my tests I have some problems with multithread on kotlin/native so I decided to use just main thread.

I think to perform tests with multithread later.

PR #64 (2021-02-18): Add iOS targets via sourceset symlinks
Approach in #63 is preferable, but due to potential issues with dependencies, leaning on symlinks to share code across targets.

Closes #52
closed 2021-02-18 by twyatt