JuulLabs/kable #54: Add support for enable indication

DoruAdryan, 2021-01-11:
There are devices which requests enabling indication (BluetoothGattDescriptor.ENABLE_INDICATION_VALUE) instead of notification (BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE) in order to receive (observe in kable) values from them. 
Maybe kable-api should allow such thing, as I do not see any kind of workaround at the moment.

twyatt, 2021-01-11:
Agree, Kable should support indications (in addition to the current notification support).

We'll investigate how the various platforms differentiate between notification vs. indication and decide on a unified API to expose in Kable.

DoruAdryan, 2021-01-22:
Is there an estimation for 0.3.0? At the moment, unfortunately this is a thing blocking me from using the library :(
Also, if I was to create a pull request for Android changes, would it be ok, or you want to analyse all platforms beforehand?

twyatt, 2021-01-22:
PRs are welcomed.

To reduce the amount of effort, it might be helpful to investigate how all the platforms configure notification vs. indication ahead of time, to have a game plan for how to create a unified API for all the platforms.

Otherwise we may have to request significant changes on a PR you create, which is also fine, if you're OK with a bit of back-and-forth on a PR.

---

I'm hopeful this is a minor change though, so how about holding on your PR for now. I'll try to find time to look into it this weekend. If I don't have anything after this weekend, then feel free to put up a PR and I'll prioritize reviewing it.

As for estimated release of [0.3.0](https://github.com/JuulLabs/kable/milestone/3), hopefully in 3-4 weeks.

twyatt, 2021-01-25:
I didn't have time to put together a PR, but did look into this a bit further. It appears that Web Bluetooth API handles this in a fairly elegant way and I'll probably carry over its behavior for an implementation on both Android and Apple:

[image]

— https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetoothremotegattcharacteristic-startnotifications

Essentially (and pretty much as you, @DoruAdryan, described) check the characteristic properties to determine how to write to the Client Characteristic Configuration Descriptor (CCCD) — it appears that this is how Web Bluetooth handles this, and additionally properly supporting when CCCD is absent.

---

The changes will likely result in `observe` being completely automatic. Though I'm not quite sure how to handle case **8** shown above, whereas a characteristic supports both **indicate** and **notify**. Currently brainstorming around one of the following options:

- When characteristic supports both **indicate** and **notify**, default to **notify**
- When characteristic supports both **indicate** and **notify**, throw an exception
- Add optional parameter to `observe` function to explicitly use **indicate** or **notify**, if requested mode is not available then throw an exception
    - Parameter default would be something along the lines of `Auto` which would exhibit the behavior of the first option

twyatt, 2021-02-24:
Apologies I haven't gotten to this bug. Been focused on other projects, but I'll be dedicating some time to Kable soon.

DoruAdryan, 2021-03-01:
I see this couldn't make it into 0.3.0. As I said, this is the only thing that blocks me from using the library. Could you provide a workaround for the issue, or should you fix it in 0.4.0, meaning next month probably ?

twyatt, 2021-03-01:
I'll do a quick follow up release that incorporates this. Hopefully sooner rather than later.
Once this is integrated I'll push a release (even if this is the only issue resolved in the new version).

twyatt, 2021-03-04:
I have not yet tested it, but the following `SNAPSHOT` has preliminary support for properly configuring **notify** and/or **indicate** based on characteristic properties.

```kotlin
import java.net.URI

repositories {
    maven { url = URI("https://oss.sonatype.org/content/repositories/snapshots") }
}

dependencies {
    implementation("com.juul.kable:core:0.3.0-issue-54-SNAPSHOT")
}
```

closed 2021-03-22 by twyatt