JuulLabs/kable #58: Apple targets should invalidate CBPeripheral objects on disconnect

twyatt, 2021-01-15:
It has been pointed out that Apple platform has some awkward behavior where CBPeripheral may be invalid upon disconnecting (e.g. when it has gone out of range).

petruswrango, 2021-08-24:
Any more info on this, or available workaround?

I'm having trouble when powering of my peripheral. When I power it back on it wont be detected a second time. Other peripherals can still be detected. If I restart the scanning when I detect a disconnect (powered off peripheral) it works as expected.  
This behavior is only seen on iOS, Android works as expected.

twyatt, 2021-08-24:
Unfortunately the Apple side of this library has not been given much attention lately. I wasn't aware that this issue was severe enough that it is preventing peripherals from being detected; so thanks for bringing that to my attention. I'll prioritize this and try to get to it soon.

Do you know of a way I can reproduce the issue on my side? For example, does taking a peripheral out-of-range also cause the same issue? Or does the peripheral have to power off and back on to see this issue?

petruswrango, 2021-08-24:
Yes, going out-of-range gives the exact same behavior for me. 
I will try to test with another type of peripheral tomorrow to check if it behaves the same.

Restarting the scan by collecting advertisements from `Scanner.advertisement` again makes the peripheral show up again, so that might be my workaround. But it's a bit backwards, and I'm not sure what kind of side effects I might run into.

petruswrango, 2021-08-25:
I checked another peripheral, a prototype board based on a completely different chipset. It shows the same behavior as the other.

twyatt, 2021-08-25:
> I checked another peripheral, a prototype board based on a completely different chipset. It shows the same behavior as the other.

Thanks for checking on that.

I'm currently swamped on a work project at the moment, but will try to get to this ASAP.

petruswrango, 2021-09-16:
Any update on this? Thanks

twyatt, 2021-09-17:
> Restarting the scan by collecting advertisements from `Scanner.advertisement` again makes the peripheral show up again

Can you clarify what you meant with "peripheral show up again"?

Does this mean that `peripheral.connect()` calls were suspending indefinitely, or at least until another scan was initiated?

Or were you seeing an error message? Was the failure when calling a function other than `connect`?

---

I attempted to reproduce the issue on my side, although I don't have an iPhone handy for testing — so, using BLE on my MacBook and connecting to a SensorTag. Using the [sample app](https://github.com/JuulLabs/sensortag), I am able to take the SensorTag out-of-range and then bring it back into range and it reconnects and communicates as expected.

I wish I had referenced where I heard that invalidating `CBPeripheral`s may be necessary when I created this issue, but I can't find any details at this point.

[This thread](https://lists.apple.com/archives/bluetooth-dev/2013/Dec/msg00032.html) indicates it is fine to reuse `CBPeripheral` objects (as Kable) is currently doing.

Any hints as to how to reproduce it on my end would be helpful. Or, iOS/Swift sample code of how the reference to the `CBPeripheral` object should be held (i.e. when/if the reference should be updated upon reconnection).

I'll continue researching as I have time, but the official documentation also doesn't provide much guidance on this topic as well. 😢

petruswrango, 2021-09-18:
> Can you clarify what you meant with "peripheral show up again"?

I just meant that I was receiving advertisements in the `Scanner.advertisements` Flow again (when I restarted the scan).

So let me try to explain what I'm seeing again.

- I start the scanner, create peripherals, connect and collect state changes. Simplified my code looks like this:

```
        scanner.advertisements.map {
            scope.peripheral(it)
        }.onEach { 
            it.connect()
        }.flatMapLatest { 
            it.state
        }.collect { 
            //Handle state changes
        }
```

- I power ON my BLE device as usual (receive the advertisement and call .connect() as shows above)
- I power OFF my device
- I receive an updated state: `State.Disconnected`
- I power my device back ON
- No new advertisements are received for that device, and thus .connect() is never called. I simply dont have a connection to my device anymore, and cant create one (since I'm not receiving it's advertisements).

I get the same behavior if go out of range. 100% repeatable, and works perfect on Android.

While testing this now I notice something that I didn't before; when I start the scan I get a lot of different BLE devices instantly, but after a while, all I see is new advertisements from my neighbor's Samsung TV. Could it be that the iPhone is somehow only emitting "updated" devices/advertisements? And if my device just advertises the same thing as before the power on/off cycle, the iPhone simply wont give it to me. I'm just guessing here.

Thanks for your time. I'll write here if I think of anything else.

twyatt, 2021-09-26:
Thanks for adding details and a simplified example — really helped me to understand the issue.

Just to clarify, this works on Android but not iOS, right?

I need to look into more detail how Core Bluetooth handles scanning while connected to a peripheral. I'm guessing there is some kind of limitation that I was unaware of (and maybe need to handle in Kable).

petruswrango, 2021-09-27:
> Just to clarify, this works on Android but not iOS, right?

Yes, never had any of these problem on Android.

twyatt, 2024-10-27:
It has been a long time, but if someone can reproduce this issue, can they test that it still exists on Kable [0.35.0-rc](https://github.com/JuulLabs/kable/releases/tag/0.35.0-rc), TY!