JuulLabs/kable #65: To expose device's bond state

francismariano, 2021-02-08:
Hello guys,

Is it possible to expose the device's bond state on Advertisement class?

twyatt, 2021-02-08:
Seems reasonable. From a quick search, it appears this status is [unavailable for Apple](https://punchthrough.com/core-bluetooth-basics/#Pairing-and-bonding) and JavaScript, so this can be an Android specific property.

twyatt, 2021-03-11:
Until [0.4.0](https://github.com/JuulLabs/kable/milestone/4) is released, this can be tested via:

```kotlin
import java.net.URI

repositories {
    maven { url = URI("https://oss.sonatype.org/content/repositories/snapshots") }
}

dependencies {
    implementation("com.juul.kable:core:0.3.0-issue-65-SNAPSHOT")
}
```

PR #76 (2021-03-22): Expose bond state on Advertisement (Android only)
Closes #65
closed 2021-03-22 by twyatt