JuulLabs/kable #66: Retrieve peripheral on iOS/macOS from previous identifier peripheral

francismariano, 2021-02-09:
Hello guys.

Currently the peripheral is retrieved only with advertisement data. I think interesting to be possible to retrieve the peripheral with its UUID identifier even knowing it is not 100% relied upon for peripheral re-identification.

Example:

```
Peripheral.kt
public fun CoroutineScope.peripheral(
    uuid: Uuid
): Peripheral {
    val peripheral = CentralManager.Default.retrievePeripheral(uuid)
    return ApplePeripheral(coroutineContext, peripheral)
}
```

```
CentralManager.kt
internal fun retrievePeripheral(withIdentifiers: Uuid) : CBPeripheral {
    val peripheral = cbCentralManager.retrievePeripheralsWithIdentifiers(listOf(withIdentifiers))
    return peripheral.firstOrNull() as CBPeripheral
}
```

Thank you

twyatt, 2021-02-09:
This does seem like a useful feature. I'll see if we can provide a common identifier to allow for a common function across platforms for retrieving a known peripheral. If the other platforms doesn't provide a similar functionality then at very least, this addition does make sense as an Apple-only function. :thu

burnhamd, 2021-06-23:
This would be a useful feature.

Android at least supports something similar.
From the documentation

> To get a BluetoothDevice, use BluetoothAdapter.getRemoteDevice(String) to create one representing a device of a known MAC address (which you can get through device discovery with BluetoothAdapter) or get one from the set of bonded devices returned by BluetoothAdapter.getBondedDevices(). 

I think realistically this should be two features.
getBondedDevices
and this peripheral from identifier.

QT has a similar capability where the identifier property is the actual Mac address on Android and the masked identifier on iOS.

burnhamd, 2022-03-15:
I have added the ability to get a peripheral from an identifier to my fork. Ill raise another PR shortly.
I think handling the ID as a string makes sense.  Of course this ID cannot be used between different devices but for the same device it seems to persist at least for multiple days as a way to identify the peripheral.

closed 2022-03-24 by twyatt