JuulLabs/kable #70: Using Kable in an Android Project

Computr0n, 2021-02-27:
I have an android gradle module and I am following the platform specific instructions in the readme that say to add:

```groovy
dependencies {
    implementation("com.juul.kable:core-android:0.2.0")
}
```

Gradle successfully syncs, but I don't see the library show up in the project window under external libraries and I cannot seem to use any kable classes in my project. 

I'm not sure whats missing, I have never used a multiplatform library from an android project. 

If I set the version to 0.1.0, android studio will recommend the next available highest version. (0.2.0 in my case although I see 0.3.0 was published yesterday). Because of this, I don't think its gradle failing to sync silently. not sure where to look next.

twyatt, 2021-02-28:
From what you've described, it _should_ work.

I find that sometimes Android Studio doesn't give the clearest failure as to what went wrong w/ Gradle sync, or it silently fails.

Can you try from the command line and let me know what kind of failure you're getting when you try to use:

```groovy
dependencies {
    implementation("com.juul.kable:core-android:0.3.0")
}
```

Example Gradle command line invocation:

```
./gradlew clean assemble
```

twyatt, 2021-02-28:
Looking at Maven, it looks like the `-metadata` artifact failed to publish w/ `0.3.0`. Although multiplatform seems to still work (tested by bumping dependency to `0.3.0` in [SensorTag sample app](https://github.com/JuulLabs/sensortag) and it compiled). I don't know a lot about the Android Studio dependency version suggestions, but it's possible that the missing metadata artifact for `0.3.0` is why it isn't suggestion the version bump to that version. 🤷 

In either case, I believe `0.3.0` should still work via the process you've described, so do let me know what the result of the Gradle command line invocation is. If we still can't figure it out, then I'll investigate the publication issue and/or publish another version.

Computr0n, 2021-02-28:
OK, I tried what you said on a freshly reset stable copy of the latest stable build of
```
Android Studio 4.1.2
Build #AI-201.8743.12.41.7042882, built on December 19, 2020
Runtime version: 1.8.0_242-release-1644-b3-6915495 x86_64
```

using a brand new project with only this line added to the app gradle modules dependency block:
```
    implementation("com.juul.kable:core-android:0.X.0")
```

Here is the output after running 
```shell
./gradlew clean assemble
```

with version 0.2.0:
```
~/AndroidStudioProjects/MyApplication  ./gradlew clean assemble

FAILURE: Build failed with an exception.

* What went wrong:
Could not determine the dependencies of task ':app:lintVitalRelease'.
> Could not resolve all artifacts for configuration ':app:debugCompileClasspath'.
   > Could not resolve com.juul.kable:core-android:0.2.0.
     Required by:
         project :app
      > No matching variant of com.juul.kable:core-android:0.2.0 was found. The consumer was configured to find an API of a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'debug', attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm' but:
          - Variant 'android-releaseApiElements' capability com.juul.kable:core-android:0.2.0 declares an API of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm':
              - Incompatible because this component declares a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'release' and the consumer needed a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'debug'
          - Variant 'android-releaseRuntimeElements' capability com.juul.kable:core-android:0.2.0 declares a runtime of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm':
              - Incompatible because this component declares a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'release' and the consumer needed a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'debug'
          - Variant 'metadata-api' capability com.juul.kable:core-android:0.2.0:
              - Incompatible because this component declares a usage of 'kotlin-metadata' of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'common' and the consumer needed an API of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm'
              - Other compatible attribute:
                  - Doesn't say anything about com.android.build.api.attributes.BuildTypeAttr (required 'debug')
          - Variant 'metadata-commonMainMetadataElements' capability com.juul.kable:core-android:0.2.0:
              - Incompatible because this component declares a usage of 'kotlin-api' of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'common' and the consumer needed an API of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm'
              - Other compatible attribute:
                  - Doesn't say anything about com.android.build.api.attributes.BuildTypeAttr (required 'debug')
```

and with version 0.3.0:

```
~/AndroidStudioProjects/MyApplication  ./gradlew clean assemble
Starting a Gradle Daemon, 1 incompatible and 1 stopped Daemons could not be reused, use --status for details

FAILURE: Build failed with an exception.

* What went wrong:
Could not determine the dependencies of task ':app:lintVitalRelease'.
> Could not resolve all artifacts for configuration ':app:debugCompileClasspath'.
   > Could not resolve com.juul.kable:core-android:0.3.0.
     Required by:
         project :app
      > No matching variant of com.juul.kable:core-android:0.3.0 was found. The consumer was configured to find an API of a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'debug', attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm' but:
          - Variant 'commonMainMetadataElements-published' capability com.juul.kable:core-android:0.3.0:
              - Incompatible because this component declares a usage of 'kotlin-api' of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'common' and the consumer needed an API of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm'
              - Other compatible attribute:
                  - Doesn't say anything about com.android.build.api.attributes.BuildTypeAttr (required 'debug')
          - Variant 'metadataApiElements-published' capability com.juul.kable:core-android:0.3.0:
              - Incompatible because this component declares a usage of 'kotlin-metadata' of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'common' and the consumer needed an API of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm'
              - Other compatible attribute:
                  - Doesn't say anything about com.android.build.api.attributes.BuildTypeAttr (required 'debug')
          - Variant 'releaseApiElements-published' capability com.juul.kable:core-android:0.3.0 declares an API of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm':
[... cut ...]
          - Variant 'releaseRuntimeElements-published' capability com.juul.kable:core-android:0.3.0 declares a runtime of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm':
              - Incompatible because this component declares a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'release' and the consumer needed a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'debug'
```

twyatt, 2021-02-28:
@Computr0n something in your Gradle configuration is expecting the `debug` version of the Kable library. With non-multiplatform Android projects generally I'm used to seeing `debugImplementation("...")` for **debug** dependencies, so I'm not quite sure what is going on there.

For a quick fix for your particular configuration, the `com.juul.kable:core-android-debug:0.3.0` should work (at least for what Gradle is expecting, although it is still strange to me that it wants the `debug` version.

I haven't tried it yet, but perhaps with newer versions of Gradle it knows to pull in the appropriate artifact based on the undecorated (i.e. `core` in this case) artifact? 🤷  In other words, does it maybe work to specify `com.juul.kable:core:0.3.0` as your dependency?

I'm no expert at Gradle, so I appreciate your patience and thoroughness with your comments in trying to figure this out.

Computr0n, 2021-02-28:
> I haven't tried it yet, but perhaps with newer versions of Gradle it knows to pull in the appropriate artifact based on the undecorated (i.e. `core` in this case) artifact? 🤷 In other words, does it maybe work to specify `com.juul.kable:core:0.3.0` as your dependency?

Indeed, gradle is apparently smart enough to figure it out:
[image]

TIL. Perhaps the readme should suggest this method instead.

> I'm no expert at Gradle, so I appreciate your patience and thoroughness with your comments in trying to figure this out.

Thank you as well. :-)

closed 2021-02-28 by Computr0n