JuulLabs/kable #75: IllegalStateException: Services have not been discovered for Peripheral

francismariano, 2021-03-10:
Hello guys.

I get the IllegalStateException when trying read a characteristic inside peripheral state collect.

```
peripheral.state.collect { state ->
    if (state == State.Connected) {
        // delay(200); // this line fix the exception
        peripheral.read(characteristic)
    }
}
```

This behavioural is normal because the services are discovered after peripheral connection is established. The commented line fixes the exception because the delay permits the peripheral to discover the services before the reading is performed.

Maybe we can change this behaviour. My suggestion is: update the peripheral state to CONNECTED after the services discovery is performed or creating a new peripheral state SERVICES_DISCOVERED.

What do you think?

twyatt, 2021-03-10:
I was really torn on the design of the `Peripheral`'s `state` `Flow`.

I went the route of having it provide a stream of the connection state as seen by the underlying platform. In this case, the BLE connection has technically "connected" (even before services have been discovered), but as you're finding, that doesn't quite align (or isn't a friendly API) when trying to act on that state.

I was hesitant to add another state into the mix (e.g. `ServicesDiscovered`) as that would pollute the current simple representation of just being the underlying BLE connection state.

Alternatively, could change what `State.Connected` means (to only be emitted _after_ services have been discovered) but then it removes the ability to know when BLE has "connected but not yet discovered services". Not sure how often that state would be needed, but I originally thought it would a good idea, as the other state could be determined separately (by "injected" a connected state into another `Flow` after the `connect` call completes, see below).

In the mean time, I'll share what we do with one of our internal libraries to accomplish what I believe you're looking for: we have a layer above Kable that manages connections, at that layer we provide a "connection state" that does represent a "services discovered connected state":

```kotlin
class Example(
    private val peripheral: Peripheral
) {

    // `MutableSharedFlow` (instead of `MutableStateFlow`) to get non-distinctUntilChanged behavior.
    private val isReady = MutableSharedFlow<State>(replay = 1, onBufferOverflow = DROP_OLDEST)

    val state: Flow<State> = merge(
        peripheral.state.filter { it != State.Connected },
        isReady
    ).distinctUntilChanged()

    suspend fun connect() {
        peripheral.connect()
        isReady.emit(State.Connected)
    }
}
```

...All that being said, definitely open to changing the API to make it friendlier and more intuitive.

I am leaning more towards changing the `state` `Flow` to be more intuitive, so you could react to a "connected **and** services discovered" state).

twyatt, 2021-08-27:
Apologies for the continuous changes, trying to remain transparent on design decisions, but another change:

We'll be adding sub-states to `Connecting` and `Connected` will be emitted _after_ services discovery **and** observations have been re-wired.

For example:

```
public sealed class State {

    public sealed class Connecting : State() {
        public object Bluetooth : Connecting()
        public object Services : Connecting()
        public object Observes : Connection()
    }

    public object Connected : State()

    // ...
}
```

closed 2021-08-28 by sdonn3

twyatt, 2021-08-28:
Updated connection states are available in [0.9.0](https://github.com/JuulLabs/kable/releases/tag/0.9.0).

twyatt, 2021-08-28:
@francismariano your original code example:

```kotlin
peripheral.state.collect { state ->
    if (state == State.Connected) {
        // delay(200); // this line fix the exception
        peripheral.read(characteristic)
    }
}
```

...should not work without the need for the `delay`.

PR #153 (2021-08-28): Add Connecting sub-states; delay Connected state until ready
Closes #75.