JuulLabs/kable #78: com.android.tools.build:gradle:7.0.0-alpha09 problem

isaya1910, 2021-03-11:
getting No matching variant of com.juul.kable:core-android:0.3.0 was found. The consumer was configured to find a runtime of a component, as well as attribute when updated my project gradle to 7.0.0 for using Jetpack Compose

isaya1910, 2021-03-11:
No matching variant of com.juul.kable:core-android:0.3.0 was found. The consumer was configured to find a runtime of a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'debug', attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm' but:
          - Variant 'commonMainMetadataElements-published' capability com.juul.kable:core-android:0.3.0:
              - Incompatible because this component declares a usage of 'kotlin-api' of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'common' and the consumer needed a runtime of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm'
              - Other compatible attribute:
                  - Doesn't say anything about com.android.build.api.attributes.BuildTypeAttr (required 'debug')
          - Variant 'metadataApiElements-published' capability com.juul.kable:core-android:0.3.0:
              - Incompatible because this component declares a usage of 'kotlin-metadata' of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'common' and the consumer needed a runtime of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm'
              - Other compatible attribute:
                  - Doesn't say anything about com.android.build.api.attributes.BuildTypeAttr (required 'debug')
          - Variant 'releaseApiElements-published' capability com.juul.kable:core-android:0.3.0 declares a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm':
              - Incompatible because this component declares an API of a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'release' and the consumer needed a runtime of a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'debug'
          - Variant 'releaseRuntimeElements-published' capability com.juul.kable:core-android:0.3.0 declares a runtime of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm':
              - Incompatible because this component declares a component, as well as attribute 'com.andr

twyatt, 2021-03-11:
Is this for a multiplatform project or single platform?

It looks like your Gradle configuration is expecting the `com.juul.kable:core-android-debug:0.3.0` artifact (note the `-debug` postfix).

isaya1910, 2021-03-11:
Thank you so much for this helping and for this amazing library!

isaya1910, 2021-03-11:
Its single android platform, -debug helped, closing this issue

closed 2021-03-11 by isaya1910

twyatt, 2021-03-11:
Glad that fixed your issue, just to add more detail...

Depends on your Gradle configuration, but you may want:

```
releaseImplementation("com.juul.kable:core-android:0.3.0")
debugImplementation("com.juul.kable:core-android-debug:0.3.0")
```