JuulLabs/kable #80: Identifier in common code

ln-12, 2021-03-22:
First of all: awesome library, it works perfect on iOS and Android!

For my project, I need want to show the available devices. For that, I want get all advertisements in common code, filter them and publish the result as list.  It would be really useful to have an identifier in this case to distinguish the devices. I can see from the source code that on iOS [the UUID is used](https://github.com/JuulLabs/kable/blob/c469772eb0660c96e2722e7c952b3f7ac8bbf331/core/src/appleMain/kotlin/Advertisement.kt#L21) while on Android [the address is returned](https://github.com/JuulLabs/kable/blob/bc1c7200cac94538ab7256806fd138f75ef65952/core/src/androidMain/kotlin/Advertisement.kt#L19). I know that on iOS you cannot get the physical device address. But what about a platform dependent id for the common code? Currently my simple approach looks like this:
```
// common
expect fun getIDForBluetoothDevice(advertisement: Advertisement): String

// Android
actual fun getIDForBluetoothDevice(advertisement: Advertisement): String = advertisement.address

// iOS
actual fun getIDForBluetoothDevice(advertisement: Advertisement): String = advertisement.identifier.toString()
```

I think it might be a good idea to add something like this as it seems to me that it's a common use case to distinguish devices by unique ids.

twyatt, 2021-03-22:
Totally agree that something like this would be helpful in the API. The current delay is the need to research Android and if it always returns the correct MAC address.

https://security.stackexchange.com/a/202893

Being that you may not get a consistent/correct MAC address, I feel that should influence the naming of the method. Or, need to determine if there is a way to get a stable ID.

ln-12, 2021-03-22:
Ah, I see, thanks for that hint. Bluetooth is such a pain to work with :D

twyatt, 2023-06-27:
Closed by #509. Will be available in the next release.

closed 2023-06-27 by twyatt