JuulLabs/kable #82: AndroidPeripheral.requestMtu() throws OutOfOrderGattCallbackException

doudasek, 2021-03-24:
When requesting MTU change, it internally succeeded but throws an exception. After that call, I am able to send for example 170B (20B before the call). You can see in logcat that it has status 0, which is [GATT_SUCCESS](https://developer.android.com/reference/android/bluetooth/BluetoothGatt#GATT_SUCCESS) in [that callback](https://developer.android.com/reference/android/bluetooth/BluetoothGattCallback#onMtuChanged(android.bluetooth.BluetoothGatt,%20int,%20int)).

**Library version**: 0.3.0
**Phone**: Pixel 5, Android 11

**Code:**
```
try {
    Timber.d("Requesting MTU: $REQUESTED_MTU")
    peripheral.requestMtu(REQUESTED_MTU)
} catch (e: Exception) {
    e.printStackTrace()
}
```

**Stack trace:**
> Requesting MTU: 512
> configureMTU() - device: 7D:CC:76:A0:A5:A0 mtu: 512
> onConfigureMTU() - Device=7D:CC:76:A0:A5:A0 mtu=512 status=0
> com.juul.kable.OutOfOrderGattCallbackException: Unexpected response type OnMtuChanged received
>   at com.juul.kable.AndroidPeripheral.requestMtu(Peripheral.kt:374)
>   at com.juul.kable.AndroidPeripheral$requestMtu$1.invokeSuspend(Unknown Source:12)
>   at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
>   at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
>   at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
>   at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
>   at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
>   at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)

twyatt, 2021-03-24:
Thanks for reporting. Was an oversight in the `requestMtu` implementation.

Can you try the following and let me know if it fixes the issue?

```kotlin
import java.net.URI

repositories {
    maven { url = URI("https://oss.sonatype.org/content/repositories/snapshots") }
}

dependencies {
    implementation("com.juul.kable:core:0.4.0-issue-82-SNAPSHOT")
}
```

Thanks!

---

_`SNAPSHOT` published from #83._

doudasek, 2021-03-24:
Thank you, it is working. Do you have an estimation for the 0.5.0 release?

twyatt, 2021-03-24:
Having issues w/ GitHub Actions publishing `0.4.0`, probably just going to roll this fix into `0.4.1` and release once I get another approval on #83. _So should be released by end-of-day today._

PR #83 (2021-03-24): Fix expected response type for requestMtu
Closes #82
closed 2021-03-24 by twyatt