JuulLabs/kable #84: Reads and writes give me "inappropriate blocking method call" lint warning in Android Studio

solvek, 2021-03-30:
I guess this is because these functions throw IOException so AS thinks these methods are blocking.
What is the preffered way to fix this? Should I suppress this warning?

`    @Suppress("BlockingMethodInNonBlockingContext") // This is how I can suppress
    private suspend fun readData() {
        Timber.tag(TAG).i("Reading data from the device")
        val time = device.read(CHARACTERISTIC_TIME) // Here is the warning
        Timber.tag(TAG).i("Time from device ${time.toHex()}")
        val capture = device.read(CHARACTERISTIC_CAPTURE)
        Timber.tag(TAG).i("Capturing state ${capture.toHex()}")
        val config = device.read(CHARACTERISTIC_OXIMETER_CONFIG)
        Timber.tag(TAG).i("Oximeter config ${config.toHex()}")
    }`

twyatt, 2021-03-30:
Until [KTIJ-838](https://youtrack.jetbrains.com/issue/KTIJ-838) is fixed, [suppressing seems to be the recommended approach](https://kotlinlang.slack.com/archives/C1CFAFJSK/p1600452070018000?thread_ts=1600451730.017900&cid=C1CFAFJSK). 🤷

twyatt, 2021-03-30:
The purpose behind marking functions as throwing `IOException` was for [Exception handling in Objective-C/Swift interop](https://kotlinlang.org/docs/whatsnew14.html#exception-handling-in-objective-c-swift-interop).

It might make sense to move the `@Throws(IOException::class)` annotations from `commonMain` to `appleMain`, which may uphold the Objective-C/Swift interop while also removing the warning on the Android side.

twyatt, 2021-07-12:
Revisited this issue and made another attempt at removing the `@Throws` from the `common` `Peripheral` (which I'm hoping would remove the erroneous warnings) but ran into some new snags:

I tried to follow the guidance in https://github.com/JetBrains/kotlin-native/issues/2122#issuecomment-475717050 (using an optional annotation) but could not make the compiler happy.

<details>
<summary><code>commonMain/../Peripheral.kt</code></summary>

```kotlin
/** @throws NotReadyException if invoked without an established [connection][connect]. */
@com.juul.kable.Throws(CancellationException::class, IOException::class, NotReadyException::class)
public suspend fun write(
    characteristic: Characteristic,
    data: ByteArray,
    writeType: WriteType = WithoutResponse,
): Unit
```
</details>

<details>
<summary><code>commandMain/../Throws.kt</code></summary>

```kotlin
package com.juul.kable

import kotlin.reflect.KClass

@OptIn(ExperimentalMultiplatform::class)
@OptionalExpectation
@Target(AnnotationTarget.FUNCTION, AnnotationTarget.CONSTRUCTOR)
@Retention(AnnotationRetention.SOURCE)
public expect annotation class Throws(vararg val exceptionClasses: KClass<out Throwable>)
```
</details>

`appleMain/../Peripheral.kt`

[image]

twyatt, 2023-03-10:
Now that [KTIJ-838](https://youtrack.jetbrains.com/issue/KTIJ-838) is fixed, is this still an issue?

twyatt, 2024-08-16:
Related to #724.

twyatt, 2025-02-20:
Please comment if this is still an issue.

closed 2025-02-20 by twyatt