JuulLabs/kable #85: Cannot connect to a device (status code 133)

solvek, 2021-03-30:
I have a device to which I cannot connect using Kable however [BLE Scanner](https://play.google.com/store/apps/details?id=com.macdom.ble.blescanner&hl=en) connects fine to it.

It crashes on `connect` call function with an exception.
With some other device my code works fine.

Also as a side note the stack trace does not show the place where the exception raisen until I added try/catch. Is there a good solution for this?

```
2021-03-30 13:13:26.364 29915-29915/com.myprojectdev I/HomeViewModel: Connecting to device 00:02:72:C8:73:3E, attempt 1
2021-03-30 13:13:26.385 29915-29915/com.myprojectdev D/BluetoothGatt: connect() - device: 00:02:72:C8:73:3E, auto: false
2021-03-30 13:13:26.385 29915-29915/com.myprojectdev D/BluetoothGatt: registerApp()
2021-03-30 13:13:26.385 29915-29915/com.myprojectdev D/BluetoothGatt: registerApp() - UUID=97455b6f-bfc2-412d-8926-d3899622a2bd
2021-03-30 13:13:26.391 29915-29940/com.myprojectdev D/BluetoothGatt: onClientRegistered() - status=0 clientIf=8
2021-03-30 13:13:26.408 29915-29915/com.myprojectdev I/HomeViewModel: Changed connect state to com.juul.kable.State$Connecting@5ab9aab
2021-03-30 13:13:26.408 29915-29915/com.myprojectdev D/HomeViewModel: Unhandled state com.juul.kable.State$Connecting@5ab9aab
2021-03-30 13:13:26.519 29915-29915/com.myprojectdev I/Choreographer: Skipped 33 frames!  The application may be doing too much work on its main thread.
2021-03-30 13:13:26.681 29915-30063/com.myprojectdev I/OpenGLRenderer: Davey! duration=725ms; Flags=1, IntendedVsync=22043658834306, Vsync=22044208834317, OldestInputEvent=9223372036854775807, NewestInputEvent=0, HandleInputStart=22044223732544, AnimationStart=22044223836236, PerformTraversalsStart=22044224763467, DrawStart=22044330493852, SyncQueued=22044341897390, SyncStart=22044342854775, IssueDrawCommandsStart=22044343481467, SwapBuffers=22044383623775, FrameCompleted=22044385422467, DequeueBufferDuration=640000, QueueBufferDuration=670000, 
2021-03-30 13:13:34.093 29915-29940/com.myprojectdev D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=8 device=00:02:72:C8:73:3E
2021-03-30 13:13:34.094 29915-30113/com.myprojectdev D/BluetoothGatt: close()
2021-03-30 13:13:34.094 29915-30113/com.myprojectdev D/BluetoothGatt: unregisterApp() - mClientIf=8
2021-03-30 13:13:34.097 29915-30113/com.myprojectdev D/BluetoothGatt: close()
2021-03-30 13:13:34.097 29915-30113/com.myprojectdev D/BluetoothGatt: unregisterApp() - mClientIf=0
2021-03-30 13:13:34.105 29915-29915/com.myprojectdev D/BluetoothGatt: close()
2021-03-30 13:13:34.105 29915-29915/com.myprojectdev D/BluetoothGatt: unregisterApp() - mClientIf=0
2021-03-30 13:13:34.111 29915-29915/com.myprojectdev E/HomeViewModel: Failed to connect device
    com.juul.kable.ConnectionLostException
        at com.juul.kable.PeripheralKt$suspendUntilConnected$2.invokeSuspend(Peripheral.kt:281)
        at com.juul.kable.PeripheralKt$suspendUntilConnected$2.invoke(Unknown Source:10)
        at kotlinx.coroutines.flow.FlowKt__TransformKt$onEach$$inlined$unsafeTransform$1$2.emit(Collect.kt:134)
        at kotlinx.coroutines.flow.StateFlowImpl.collect(StateFlow.kt:348)
        at kotlinx.coroutines.flow.StateFlowImpl$collect$1.invokeSuspend(Unknown Source:12)
        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
        at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:214)
        at android.app.ActivityThread.main(ActivityThread.java:7386)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:980)
2021-03-30 13:13:34.112 29915-29915/com.myprojectdev I/HomeViewModel: Changed connect state to Disconnected(status=Unknown(status=133))
2021-03-30 13:13:34.113 29915-29915/com.myprojectdev D/HomeViewModel: Waiting 500 ms to reconnect...
```

solvek, 2021-03-30:
I tried connect to this device directly not via Kable and if I add `BluetoothDevice.TRANSPORT_LE` parameter I can connect fine.

twyatt, 2021-03-30:
Thanks for the bug report, sounds like it's related to #61.
I'll try to prioritize getting a PR up that at least allows configuring the transport type.

twyatt, 2021-04-17:
Closing as duplicate of #61.

closed 2021-04-17 by twyatt