JuulLabs/kable #86: Peripheral initiated MTU Changes can cause OutOfOrderGattCallbackException

mediaController, 2021-04-05:
The BluetoothGattCallback() method for onMtuChanged needs to be serialized with the execute method when the libraries consumer calls readMtu(), but unlike other onBlankEvent() methods, this one can also be called in response to a peripheral device requesting an MTU change.

Note the UPPERCASED part of the documentation: 

```java
    /**
     * Callback indicating the MTU for a given device connection has changed.
     *
     * This callback is triggered in response to the requestMtu function, 
     * OR IN RESPONSE TO A CONNECTION EVENT.
     *
     * @param gatt GATT client invoked {@link BluetoothGatt#requestMtu}
     * @param mtu The new MTU size
     * @param status {@link BluetoothGatt#GATT_SUCCESS} if the MTU has been changed successfully
     */
    public void onMtuChanged(BluetoothGatt gatt, int mtu, int status) {
    }
```

I forked the old able library and fixed it there by exposing MTU Updates through a separate stateflow that is exposed publicly and is used for retrieving a serialized response to readMtu() with the execute method.

mediaController, 2021-04-05:
I should note that unless your peripheral device can request an MTU change on its own behalf, you likely would never see this problem show up. Our device happens to do this automatically after a few seconds into the connection, or if the OS happens to perform this request for a profile that it manages.

twyatt, 2021-04-06:
Thanks for reporting. We had only ever initiated MTU changes from the client side, so I was unaware that it could be initiated from the server side. TIL.

I'll brainstorm good ways to handle this, but as you described, using a `Flow`, is probably a good approach.

PR #135 (2021-07-12): Handle peripheral initiated MTU changes
Fixes #86.

Prevents `OutOfOrderGattCallbackException` after peripheral initiated MTU by implementing a dedicated channel for MTU events.

Additionally, exposes `mtu: StateFlow<Int?>` on `AndroidPeripheral`.
closed 2021-07-12 by twyatt