JuulLabs/kable #87: Add support for different build types

nikname, 2021-04-06:
Hi, I am trying to build an Android Gradle module selecting the "staging" build type and I am getting the following error

```
Could not determine the dependencies of task ':app:lintVitalStaging'.
> Could not resolve all artifacts for configuration ':app:stagingCompileClasspath'.
   > Could not resolve com.juul.kable:core:0.4.1.
     Required by:
         project :app
      > No matching variant of com.juul.kable:core:0.4.1 was found. The consumer was configured to find an API of a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'staging', attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm' but:
          - Variant 'commonMainMetadataElements-published' capability com.juul.kable:core:0.4.1:
              - Incompatible because this component declares a usage of 'kotlin-api' of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'common' and the consumer needed an API of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm'
              - Other compatible attribute:
                  - Doesn't say anything about com.android.build.api.attributes.BuildTypeAttr (required 'staging')
          - Variant 'debugApiElements-published' capability com.juul.kable:core:0.4.1 declares an API of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm':
              - Incompatible because this component declares a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'debug' and the consumer needed a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'staging'
          - Variant 'debugRuntimeElements-published' capability com.juul.kable:core:0.4.1 declares a runtime of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm':
              - Incompatible because this component declares a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'debug' and the consumer needed a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'staging'
          - Variant 'iosArm64ApiElements-published' capability com.juul.kable:core:0.4.1:
              - Incompatible because this component declares a usage of 'kotlin-api' of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'native' and the consumer needed an API of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm'
              - Other compatible attribute:
                  - Doesn't say anything about com.android.build.api.attributes.BuildTypeAttr (required 'staging')
          - Variant 'iosArm64MetadataElements-published' capability com.juul.kable:core:0.4.1:
              - Incompatible because this component declares a usage of 'kotlin-metadata' of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'native' and the consumer needed an API of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm'
              - Other compatible attribute:
                  - Doesn't say anything about com.android.build.api.attributes.BuildTypeAttr (required 'staging')
          - Variant 'iosX64ApiElements-published' capability com.juul.kable:core:0.4.1:
              - Incompatible because this component declares a usage of 'kotlin-api' of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'native' and the consumer needed an API of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm'
              - Other compatible attribute:
                  - Doesn't say anything about com.android.build.api.attributes.BuildTypeAttr (required 'staging')
          - Variant 'iosX64MetadataElements-published' capability com.juul.kable:core:0.4.1:
              - Incompatible because this component declares a usage of 'kotlin-metadata' of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'native' and the consumer needed an API of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm'
              - Other compatible attribute:
                  - Doesn't say anything about com.android.build.api.attributes.BuildTypeAttr (required 'staging')
          - Variant 'jsApiElements-published' capability com.juul.kable:core:0.4.1:
              - Incompatible because this component declares a usage of 'kotlin-api' of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'js' and the consumer needed an API of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm'
              - Other compatible attribute:
                  - Doesn't say anything about com.android.build.api.attributes.BuildTypeAttr (required 'staging')
          - Variant 'jsRuntimeElements-published' capability com.juul.kable:core:0.4.1:
              - Incompatible because this component declares a usage of 'kotlin-runtime' of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'js' and the consumer needed an API of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm'
              - Other compatible attribute:
                  - Doesn't say anything about com.android.build.api.attributes.BuildTypeAttr (required 'staging')
          - Variant 'macosX64ApiElements-published' capability com.juul.kable:core:0.4.1:
              - Incompatible because this component declares a usage of 'kotlin-api' of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'native' and the consumer needed an API of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm'
              - Other compatible attribute:
                  - Doesn't say anything about com.android.build.api.attributes.BuildTypeAttr (required 'staging')
          - Variant 'macosX64MetadataElements-published' capability com.juul.kable:core:0.4.1:
              - Incompatible because this component declares a usage of 'kotlin-metadata' of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'native' and the consumer needed an API of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm'
              - Other compatible attribute:
                  - Doesn't say anything about com.android.build.api.attributes.BuildTypeAttr (required 'staging')
          - Variant 'metadataApiElements-published' capability com.juul.kable:core:0.4.1:
              - Incompatible because this component declares a usage of 'kotlin-metadata' of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'common' and the consumer needed an API of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm'
              - Other compatible attribute:
                  - Doesn't say anything about com.android.build.api.attributes.BuildTypeAttr (required 'staging')
          - Variant 'releaseApiElements-published' capability com.juul.kable:core:0.4.1 declares an API of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm':
              - Incompatible because this component declares a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'release' and the consumer needed a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'staging'
          - Variant 'releaseRuntimeElements-published' capability com.juul.kable:core:0.4.1 declares a runtime of a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm':
              - Incompatible because this component declares a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'release' and the consumer needed a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'staging'
```

The problem seems related to https://github.com/JuulLabs/kable/issues/70. How can I fix the problem?

twyatt, 2021-04-06:
What version of Gradle are you using?

I'm not super familiar with Gradle, but I believe older versions need the explicit platform identifiers (e.g. `-android-debug` postfix on the artifactId) while newer versions do not.

For older versions, you can try:

```
releaseImplementation("com.juul.app:core-android:0.4.1")
debugImplementation("com.juul.app:core-android-debug:0.4.1")
```

For newer versions of Gradle, the simple `implementation("com.juul.app:core:0.4.1")` _should_ work.

nikname, 2021-04-06:
I was using Gradle 6.5.0 but even after I upgrade to 6.8.1 I get the same error. Anyway I have found a workaround by adding
```
matchingFallbacks = ["release"]
```
to the "staging" build type configuraiton

twyatt, 2021-04-06:
Apologies I couldn't be of more help (Gradle is not my strong suit) but great to hear you found a workaround, and thanks for sharing it!

closed 2021-11-15 by twyatt