JuulLabs/kable #88: NullPointerException when trying to read data

solvek, 2021-04-09:
I am getting the exception below when trying to read data from charactertistic

```
    java.lang.NullPointerException: value must not be null
        at com.juul.kable.gatt.Callback.onCharacteristicRead(Callback.kt:110)
        at android.bluetooth.BluetoothGatt$1$6.run(BluetoothGatt.java:394)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:214)
        at android.os.HandlerThread.run(HandlerThread.java:67)
```

solvek, 2021-04-09:
Actually in my case it crashes on first operation I am doing with the peripheral after connecting.
Stacktrace from above happens when I do read data from characteristic.

If my first operation is `observe` then crash looks like this

```
    com.juul.kable.GattStatusException: OnDescriptorWrite(descriptor=android.bluetooth.BluetoothGattDescriptor@ebd860e, status=GATT_INVALID_HANDLE(1))
        at com.juul.kable.AndroidPeripheral.write(Peripheral.kt:493)
        at com.juul.kable.AndroidPeripheral$write$4.invokeSuspend(Unknown Source:12)
        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
        at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
        at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:750)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)
```

and this is crash if I do `write`

```
    com.juul.kable.GattStatusException: OnCharacteristicWrite(characteristic=android.bluetooth.BluetoothGattCharacteristic@1969a7d, status=GATT_INVALID_HANDLE(1))
        at com.juul.kable.AndroidPeripheral.write(Peripheral.kt:433)
        at com.juul.kable.AndroidPeripheral$write$1.invokeSuspend(Unknown Source:12)
        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
        at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:214)
        at android.app.ActivityThread.main(ActivityThread.java:7386)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:980)
```

solvek, 2021-04-09:
Can we just specify some parameter to ignore response status of callbacks?

twyatt, 2021-04-09:
Thanks for reporting the bug!

Based on the `observe` and `write` stacktraces, it looks like Android is unable to properly communicate with your BLE peripheral (`GATT_INVALID_HANDLE(1)`).

Kable should definitely be providing better (more informative logs) for the `read` operation (other than `NullPointerException`) but I'm not sure what functionality you'd gain by having Kable "ignore response status of callback" when the `GATT_STATUS` is a failure? It seems any action with your BLE peripheral wouldn't work if you're getting the `GATT_INVALID_HANDLE` failures, right?

Perhaps after a short time your BLE peripheral starts to respond with gatt statuses that aren't failures? In which case, you could wrap the Kable I/O operations in `try`-`catch` and provide some retry logic? For the `observe`, I believe the `catch` and/or `retry` Flow operators might help you there.

I'll update the `read` function to throw `GattStatusException` exception when a gatt status failure occurs (instead of `NullPointerException`), is that the change you're looking for here? Or were you hoping for some additional changes?

francismariano, 2021-04-26:
Hi @twyatt 

I am getting the same exception on the specific scenario:

1- Try read a characteristic which requires a secure connection.
2- User types wrong PIN code or cancelling the pair request.

Let me know if you need more information.

twyatt, 2021-04-26:
@francismariano thanks for the info. It confirms that the fix to at least throw the appropriate (more informative) `GattStatusException` is working.

Aside from the failure due to the `GATT_AUTH_FAIL` status, what should Kable be doing differently?

Seems like something a consumer of the library should handle? Or is there a standard approach to resolving the auth issue that Kable should be doing?

francismariano, 2021-04-27:
I think `GATT_AUTH_FAIL` status depends on user interaction when it accept the pairing request and typing the correct PIN. So, it is ok the consumer of the library handle it.

closed 2021-04-28 by twyatt

twyatt, 2021-04-28:
Fixed (to throw the appropriate exception) in [0.5.0](https://github.com/JuulLabs/kable/releases/tag/0.5.0).