JuulLabs/kable #89: Indicate does not work

solvek, 2021-04-16:
A characteristic of my device is Indicate.
This differs from Notify in such a way that an update from characteristic requires an acknoladgement.
Seems Kable does not send such acknowladgement.

This is a fragment from log
```
2021-04-16 22:54:06.721 6155-6155/com.pulsariodev D/BluetoothGatt: setCharacteristicNotification() - uuid: 00008a82-0000-1000-8000-00805f9b34fb enable: true
2021-04-16 22:54:06.862 6155-6155/com.pulsariodev V/ZewaDevice: Received data A0C22656D4000000000000000000000000000000
2021-04-16 22:54:06.952 6155-6155/com.pulsariodev D/BluetoothGatt: setCharacteristicNotification() - uuid: 00008a82-0000-1000-8000-00805f9b34fb enable: false
2021-04-16 22:54:06.964 6155-6155/com.pulsariodev V/ZewaDevice: Writing int data 215BD16264
2021-04-16 22:54:07.039 6155-6155/com.pulsariodev D/ZewaDevice: Pairing, password -1037674796, Account 1540448868
2021-04-16 22:54:07.040 6155-6155/com.pulsariodev D/ZewaDevice: Verification. Reading random number
2021-04-16 22:54:07.066 6155-6155/com.pulsariodev D/BluetoothGatt: setCharacteristicNotification() - uuid: 00008a82-0000-1000-8000-00805f9b34fb enable: true
2021-04-16 22:54:07.148 6155-6173/com.pulsariodev D/BluetoothGatt: onClientConnectionState() - status=19 clientIf=13 device=CA:41:D4:56:26:C2
2021-04-16 22:54:07.150 6155-6350/com.pulsariodev D/BluetoothGatt: close()
2021-04-16 22:54:07.150 6155-6350/com.pulsariodev D/BluetoothGatt: unregisterApp() - mClientIf=13
```
status 19 means GATT_RSP_EXEC_WRITE that my device closed the connection since it does not receive an acknowledgment for an indicate.

twyatt, 2021-04-17:
Acknowledgement of indications is handled automatically by Android.

What Kable does do, is configure the Client Characteristic Configuration descriptor (CCCD) based on what the characteristic supports.

It's possible Kable isn't configuring the CCCD properly, or perhaps your peripheral's characteristic doesn't show as supporting **Indicate**?

solvek, 2021-04-17:
There was my mistake.
The scenario I want to achieve is

1. Write data to a chara A (a command to receive some data)
2. Wait until an Indicate chara is updated
3. Use this data

I used such approach

1. peripheral.write(...)
2. val r = peripheral.observe(...).first()

I am new in coroutines and flows and seems `first` receive expected data but makes the flow to cancel, which is not what I want.
So I rewrited my code to use channel. Not sure whether this is the best approach.

What is the best approach to achive my scenario above?

twyatt, 2021-04-17:
If I understand correctly: you want to continue observing the characteristic but _also_ retrieve a single element from the observation?

There are many ways to accomplish that, [`shareIn`](https://kotlin.github.io/kotlinx.coroutines/kotlinx-coroutines-core/kotlinx.coroutines.flow/share-in.html) operator sounds like it might fit your needs? It basically will subscribe to the `Peripheral.observe` `Flow` but then allow for numerous subscribers. This mean a subscriber can come along and receive some items, and then close without closing the `SharedFlow`.

Something to the effect of:

```kotlin
// Be sure to cancel `scope` when you want to stop observing.
val observation = peripheral.observe(...).shareIn(scope, SharingStarted.Eagerly)

val response = observation
    .onSubscription { peripheral.write(...) }
    .first()
```

The `onSubscription` + `first` combination is a good fit for write-for-response style communication. It ensures that the subscription is hooked up **before** performing the write.

You can read more on the topic in [this](https://github.com/Kotlin/kotlinx.coroutines/issues/1758#issuecomment-631917778) thread.

twyatt, 2021-04-29:
Closing as I think the outstanding questions have been addressed. Please let me know if that's not the case and you still have any more questions related to this issue.

closed 2021-04-29 by twyatt