JuulLabs/kable #96: Peripheral.observe function throws NotReadyException on iOS

francismariano, 2021-05-04:
Hello,

Peripheral.observe function throws NotReadyException on iOS when the peripheral is shutdown or the application disconnect from it. That behaviour do not happen on Android.

I did not investigate if that is correct, when I have some time I can investigate it.
Issue opened for tracking and discussion.

twyatt, 2021-05-04:
Thanks for the issue report!

The Android code is definitely more robust and more thoroughly tested, so there is likely still gaps in the iOS code.

Any chance you can provide more detail re: reproduction steps?

More specifically, when you say "peripheral is shutdown or the application disconnect from it", do you mean when `peripheral.disconnect` is called?

francismariano, 2021-05-04:
> Any chance you can provide more detail re: reproduction steps?
Watch a notification characteristic calling peripheral.observe() function. 

scope.launch {
      peripheral.observe(characteristic).catch {
            println("observe:catch")
      }.collect {
            ............
      }
}

For getting the NotReadyException there are two scenarios:

1- Turn off the peripheral (physical device)
2- Calling the peripheral.disconnect() function before without cancelling the job of the observe.

twyatt, 2021-05-17:
@francismariano when you have a chance, can you try the following `SNAPSHOT` to check if it fixes the issue:

```kotlin
repositories {
    maven("https://oss.sonatype.org/content/repositories/snapshots")
}

dependencies {
    implementation("com.juul.kable:core:0.5.0-issue-96-SNAPSHOT")
}
```

Thank you!

closed 2021-05-25 by twyatt

francismariano, 2021-05-25:
@twyatt Sorry. I was unable to test. my week was very troubled.

Nice job.

twyatt, 2021-05-25:
> @twyatt Sorry. I was unable to test. my week was very troubled.
> 
> Nice job.

No worries. Just let me know if it isn't actually fixed and I can look into it more.

PR #101 (2021-05-25): Ignore `NotReadyException` when stopping observations on Apple
Closes #96