JuulLabs/kable #99: Peripheral.observe has not been notified with a characteristic without descriptor

nspeeD, 2021-05-14:
We are trying to make a proof of concept of our app using Kable library (0.5.0) for connecting with the devices. Right now, we have devices with different bluetooth modules and of course, different services/characteristics/descriptors.

In our devices with a characteristic which contains a descriptor, if the device notify a response, we are able to receive the response in the "peripheral.observe" method.

On other hand, we have some old devices which not have a descriptor for each characteristic and we are not able to receive the response in the "peripheral.observe" method. I was taking a look to the issues and I found an issue from December that says that you implemented a piece of code to automatically detect if the characteristic contains a descriptor or not and act in consequence. 

We were wondering if we need to have something in consideration to solve this or if exist any impediment.

twyatt, 2021-05-14:
I'll assume this is in reference to Android, so I'll respond with that assumption in mind.

Observations are spun up via the following steps:

1. Enable characteristic notifications (`BluetoothGatt. setCharacteristicNotification(..., true)`)
2. Lookup "config descriptor"
    - If "config descriptor" exists:
        - If characteristic supports notify: write to "config descriptor" `ENABLE_NOTIFICATION_VALUE`
        - If characteristic supports indication: write to "config descriptor" `ENABLE_INDICATION_VALUE`

Whereas all sub-steps of step 2 will simply be skipped if the "config descriptor" does not exist, but step 1 will still be executed.

The relevant code snippets are:

https://github.com/JuulLabs/kable/blob/151e54d255bf5595c67023927084d083e6180706/core/src/androidMain/kotlin/Peripheral.kt#L287-L293

https://github.com/JuulLabs/kable/blob/151e54d255bf5595c67023927084d083e6180706/core/src/androidMain/kotlin/Peripheral.kt#L303-L324

I'm not sure if additional steps are needed for your device (for it to send notifications and/or indications) aside from `BluetoothGatt. setCharacteristicNotification(..., true)` which should be happening regardless of absence of a "config descriptor".

Are you able to receive notifications or indications using another library and/or interacting directly with Android BLE (without Kable)? If so, then I would love to know what setup process is used for the "observation" (indications and/or notifications) so that I can update Kable as needed.

Thanks!

nspeeD, 2021-05-14:
Yes, we are starting this proof of concept with the android platform. Regarding your question, in our app, we are using this library (https://github.com/Jasonchenlijian/FastBle) and we can receive the notification without a descriptor in the characteristic.

If you want to take a look to the internal process of the library, you should check this two files:
BleManager: https://github.com/Jasonchenlijian/FastBle/blob/master/FastBleLib/src/main/java/com/clj/fastble/BleManager.java
BleConnector: https://github.com/Jasonchenlijian/FastBle/blob/master/FastBleLib/src/main/java/com/clj/fastble/bluetooth/BleConnector.java

If need more information or something, I'm here to help!

twyatt, 2021-05-14:
Thanks for the links!

At quick glance, FastBle seems to follow the same process as Kable, so it's odd that it works w/ FastBle but not Kable.

I'll try to find some time soon to create a Kable build with significant logging around the observation, so that you can try it and report back with logs, then hopefully we can figure out what is going wrong.

nspeeD, 2021-05-14:
Perfect, thanks!!

twyatt, 2021-05-17:
Looking into FastBle a bit more, the only difference I can see with how it spins up notifications/indications (i.e. Kable's `observe`) is if `useCharacteristicDescriptor` is `true` then it writes `ENABLE_NOTIFICATION_VALUE` to the characteristic's descriptor (with the same UUID as the characteristic) — [`BleConnector.java#L294-L296`](https://github.com/Jasonchenlijian/FastBle/blob/d89126caee2326ebe8c2102eecbbeddc9eae3274/FastBleLib/src/main/java/com/clj/fastble/bluetooth/BleConnector.java#L294-L296):

```java
if (useCharacteristicDescriptor) {
    descriptor = characteristic.getDescriptor(characteristic.getUuid());
} else {
```

@nspeeD are you calling FastBle's [`enableCharacteristicNotify`](https://github.com/Jasonchenlijian/FastBle/blob/d89126caee2326ebe8c2102eecbbeddc9eae3274/FastBleLib/src/main/java/com/clj/fastble/bluetooth/BleConnector.java#L244-L245) with `userCharacteristicDescriptor` being `true`?

---

I've also pushed up a `SNAPSHOT` of Kable that has more logging around observations, if you want to give it a run and report back on the logs it produces:

```kotlin
import java.net.URI

repositories {
    maven { url = URI("https://oss.sonatype.org/content/repositories/snapshots") }
}

dependencies {
    implementation("com.juul.kable:core:0.5.0-issue-99-SNAPSHOT")
}
```

Thank you!

nspeeD, 2021-05-18:
Answering your last question, we are not doing nothing especial with enableCharacteristicNotify nor userCharacteristicDescriptor, we are only calling startNotify on the BleManager.class.

After using your snapshot, I can see this logs:
connect() - device: 4C:55:CC:1F:87:7E, auto: false
D/BluetoothGatt: registerApp()
D/BluetoothGatt: registerApp() - UUID=95986f34-d083-448b-ac45-fe71ad5ba17b
D/BluetoothGatt: onClientRegistered() - status=0 clientIf=11
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=11 device=4C:55:CC:1F:87:7E
D/BluetoothGatt: discoverServices() - device: 4C:55:CC:1F:87:7E
D/BluetoothGatt: onConnectionUpdated() - Device=4C:55:CC:1F:87:7E interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt: onSearchComplete() = Device=4C:55:CC:1F:87:7E Status=0
I/System.out: Set notification observer
D/BluetoothGatt: onConnectionUpdated() - Device=4C:55:CC:1F:87:7E interval=9 latency=0 timeout=400 status=0
D/Kable: setCharacteristicNotification(true) for LazyCharacteristic(serviceUuid=175f8f23-a570-49bd-9627-815a6a27de2a, characteristicUuid=cacc07ff-ffff-4c48-8fae-a9ef71b75e26)
D/BluetoothGatt: setCharacteristicNotification() - uuid: cacc07ff-ffff-4c48-8fae-a9ef71b75e26 enable: true
D/Kable: Wrote ENABLE_NOTIFICATION_VALUE to config PlatformDescriptor(serviceUuid=175f8f23-a570-49bd-9627-815a6a27de2a, characteristicUuid=cacc07ff-ffff-4c48-8fae-a9ef71b75e26, descriptorUuid=00002902-0000-1000-8000-00805f9b34fb)
D/Kable: Wrote ENABLE_INDICATION_VALUE to config PlatformDescriptor(serviceUuid=175f8f23-a570-49bd-9627-815a6a27de2a, characteristicUuid=cacc07ff-ffff-4c48-8fae-a9ef71b75e26, descriptorUuid=00002902-0000-1000-8000-00805f9b34fb)
I/System.out: SetMode: 01
I/System.out: Command sent
D/BluetoothGatt: cancelOpen() - device: 4C:55:CC:1F:87:7E
D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=11 device=4C:55:CC:1F:87:7E
D/BluetoothGatt: close()
D/BluetoothGatt: unregisterApp() - mClientIf=11
D/BluetoothGatt: close()
D/BluetoothGatt: unregisterApp() - mClientIf=0
D/BluetoothGatt: close()
D/BluetoothGatt: unregisterApp() - mClientIf=0

In your logs, you are logging the properties of the notification characteristic but I don't know if you want to see all the descriptors for all the characteristics, in that case:

DiscoveredService(
    serviceUuid=175f8f23-a570-49bd-9627-815a6a27de2a,
    characteristics=[
        //This is our characteristic to write data
        DiscoveredCharacteristic(
            serviceUuid=175f8f23-a570-49bd-9627-815a6a27de2a,
            characteristicUuid=1cce1ea8-bd34-4813-a00a-c76e028fadcb,
            descriptors=[]
        ),

        //This is our characteristic for notifications from the charger
        DiscoveredCharacteristic(
            serviceUuid=175f8f23-a570-49bd-9627-815a6a27de2a,
            characteristicUuid=cacc07ff-ffff-4c48-8fae-a9ef71b75e26,
            descriptors=[
                LazyDescriptor(
                    serviceUuid=175f8f23-a570-49bd-9627-815a6a27de2a,
                    characteristicUuid=cacc07ff-ffff-4c48-8fae-a9ef71b75e26,
                    descriptorUuid=00002902-0000-1000-8000-00805f9b34fb
                )
            ]
        ),

        //This is our characteristic to read data
        DiscoveredCharacteristic(
            serviceUuid=175f8f23-a570-49bd-9627-815a6a27de2a,
            characteristicUuid=20b9794f-da1a-4d14-8014-a0fb9cefb2f7,
            descriptors=[
                LazyDescriptor(
                    serviceUuid=175f8f23-a570-49bd-9627-815a6a27de2a,
                    characteristicUuid=20b9794f-da1a-4d14-8014-a0fb9cefb2f7,
                    descriptorUuid=00002902-0000-1000-8000-00805f9b34fb
                )
            ]
        )
    ]
)

For being more clear, we are writing data on the write characteristic and expected the response in the notification characteristic, I hope it help!

twyatt, 2021-05-18:
Thanks for the logs! It looks like it is properly setting up the observation (by writing the appropriate values to the config descriptor — albeit enabling both Notify and Indicate, but at worst your peripheral would send duplicates rather than you receiving nothing, as you've reported).

It appears that you're disconnecting after the observation is spun up, is that intentional? _(perhaps the timeframe between the observation being configured and the disconnect had given it a reasonable amount of time? — just wanted to be sure since timestamps are absent)_

I realized that I'd like to get a little more logging in there to see if it is an issue with the responses being picked up properly. If you could refresh your dependencies (i.e. pull down the `SNAPSHOT` again) and try again. You should get additional logs around writes and characteristic changes.

nspeeD, 2021-05-18:
We are disconnecting after 20 sec from the last write operation and that should not be the problem, I'm connected to the device via SSH and I'm able to see the response after 1 sec from the last write operation.

I downloaded again the snapshot but I'm seeing the same logs, I did it after invalidate all the caches from the Android Studio, perhaps the cache is in the repository side so could you upgrade the version?

twyatt, 2021-05-18:
Added additional logs around the connection and pushed up a new `SNAPSHOT` as `com.juul.kable:core:0.5.0-issue-99-2-SNAPSHOT`.

nspeeD, 2021-05-18:
2021-05-18 09:47:30.858 D/BluetoothGatt: connect() - device: 4C:55:CC:1F:87:7E, auto: false
    2021-05-18 09:47:30.858 D/BluetoothGatt: registerApp()
    2021-05-18 09:47:30.859 D/BluetoothGatt: registerApp() - UUID=e758a6b0-fcb9-43d2-9551-170c6c2b4084
    2021-05-18 09:47:30.863 D/BluetoothGatt: onClientRegistered() - status=0 clientIf=16
    2021-05-18 09:47:30.872 D/Kable: Connecting to 4C:55:CC:1F:87:7E
    2021-05-18 09:47:31.144 D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=16 device=4C:55:CC:1F:87:7E
    2021-05-18 09:47:31.148 D/Kable: Discovering services for 4C:55:CC:1F:87:7E
    2021-05-18 09:47:31.150 D/BluetoothGatt: discoverServices() - device: 4C:55:CC:1F:87:7E
    2021-05-18 09:47:31.580 D/BluetoothGatt: onConnectionUpdated() - Device=4C:55:CC:1F:87:7E interval=6 latency=0 timeout=500 status=0
    2021-05-18 09:47:32.038 D/BluetoothGatt: onSearchComplete() = Device=4C:55:CC:1F:87:7E Status=0
    2021-05-18 09:47:32.044 D/Kable: Rewiring observers for 4C:55:CC:1F:87:7E
    2021-05-18 09:47:32.045 D/Kable: Connected to 4C:55:CC:1F:87:7E
    2021-05-18 09:47:32.049 I/System.out: Set notification observer
    2021-05-18 09:47:32.075 D/BluetoothGatt: onConnectionUpdated() - Device=4C:55:CC:1F:87:7E interval=9 latency=0 timeout=400 status=0
    2021-05-18 09:47:32.139 D/Kable: setCharacteristicNotification(true) for LazyCharacteristic(serviceUuid=175f8f23-a570-49bd-9627-815a6a27de2a, characteristicUuid=cacc07ff-ffff-4c48-8fae-a9ef71b75e26)
    2021-05-18 09:47:32.139 D/BluetoothGatt: setCharacteristicNotification() - uuid: cacc07ff-ffff-4c48-8fae-a9ef71b75e26 enable: true
    2021-05-18 09:47:32.142 D/Kable: Writing 2 bytes to descriptor 00002902-0000-1000-8000-00805f9b34fb
    2021-05-18 09:47:32.188 D/Kable: Wrote OnDescriptorWrite(descriptor=android.bluetooth.BluetoothGattDescriptor@f82461d, status=GATT_SUCCESS(0))
    2021-05-18 09:47:32.189 D/Kable: Write to 00002902-0000-1000-8000-00805f9b34fb complete
    2021-05-18 09:47:32.189 D/Kable: Wrote ENABLE_NOTIFICATION_VALUE to config PlatformDescriptor(serviceUuid=175f8f23-a570-49bd-9627-815a6a27de2a, characteristicUuid=cacc07ff-ffff-4c48-8fae-a9ef71b75e26, descriptorUuid=00002902-0000-1000-8000-00805f9b34fb)
    2021-05-18 09:47:32.189 D/Kable: Received OnCharacteristicChanged(characteristic=cacc07ff-ffff-4c48-8fae-a9ef71b75e26, value=14 bytes)
    2021-05-18 09:47:32.190 D/Kable: Writing 2 bytes to descriptor 00002902-0000-1000-8000-00805f9b34fb
    2021-05-18 09:47:32.211 D/Kable: Wrote OnDescriptorWrite(descriptor=android.bluetooth.BluetoothGattDescriptor@f82461d, status=GATT_SUCCESS(0))
    2021-05-18 09:47:32.211 D/Kable: Write to 00002902-0000-1000-8000-00805f9b34fb complete
    2021-05-18 09:47:32.212 D/Kable: Wrote ENABLE_INDICATION_VALUE to config PlatformDescriptor(serviceUuid=175f8f23-a570-49bd-9627-815a6a27de2a, characteristicUuid=cacc07ff-ffff-4c48-8fae-a9ef71b75e26, descriptorUuid=00002902-0000-1000-8000-00805f9b34fb)
    2021-05-18 09:47:35.866 D/Kable: Writing 1 bytes to characteristic 20b9794f-da1a-4d14-8014-a0fb9cefb2f7
    2021-05-18 09:47:35.899 D/Kable: Wrote OnCharacteristicWrite(characteristic=android.bluetooth.BluetoothGattCharacteristic@342e092, status=GATT_SUCCESS(0))
    2021-05-18 09:47:35.900 D/Kable: Write to 20b9794f-da1a-4d14-8014-a0fb9cefb2f7 complete
    2021-05-18 09:47:36.028 I/System.out: SetMode: 01
    2021-05-18 09:47:36.032 D/Kable: Writing 21 bytes to characteristic 1cce1ea8-bd34-4813-a00a-c76e028fadcb
    2021-05-18 09:47:36.035 D/Kable: Wrote OnCharacteristicWrite(characteristic=android.bluetooth.BluetoothGattCharacteristic@6d24a63, status=GATT_SUCCESS(0))
    2021-05-18 09:47:36.035 D/Kable: Write to 1cce1ea8-bd34-4813-a00a-c76e028fadcb complete
    2021-05-18 09:47:36.038 D/Kable: Writing 18 bytes to characteristic 1cce1ea8-bd34-4813-a00a-c76e028fadcb
    2021-05-18 09:47:36.040 D/Kable: Wrote OnCharacteristicWrite(characteristic=android.bluetooth.BluetoothGattCharacteristic@6d24a63, status=GATT_SUCCESS(0))
    2021-05-18 09:47:36.041 D/Kable: Write to 1cce1ea8-bd34-4813-a00a-c76e028fadcb complete
    2021-05-18 09:47:36.041 I/System.out: Command sent
    2021-05-18 09:47:50.854 D/BluetoothGatt: cancelOpen() - device: 4C:55:CC:1F:87:7E
    2021-05-18 09:47:50.860 D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=16 device=4C:55:CC:1F:87:7E
    2021-05-18 09:47:50.861 D/BluetoothGatt: close()
    2021-05-18 09:47:50.861 D/BluetoothGatt: unregisterApp() - mClientIf=16
    2021-05-18 09:47:50.863 D/BluetoothGatt: close()
    2021-05-18 09:47:50.863 D/BluetoothGatt: unregisterApp() - mClientIf=0
    2021-05-18 09:47:50.868 D/Kable: Disposing of 4C:55:CC:1F:87:7E
    2021-05-18 09:47:50.868 D/BluetoothGatt: close()
    2021-05-18 09:47:50.868 D/BluetoothGatt: unregisterApp() - mClientIf=0

twyatt, 2021-05-18:
```
2021-05-18 09:47:30.858 D/BluetoothGatt: connect() - device: 4C:55:CC:1F:87:7E, auto: false
2021-05-18 09:47:30.858 D/BluetoothGatt: registerApp()
2021-05-18 09:47:30.859 D/BluetoothGatt: registerApp() - UUID=e758a6b0-fcb9-43d2-9551-170c6c2b4084
2021-05-18 09:47:30.863 D/BluetoothGatt: onClientRegistered() - status=0 clientIf=16
2021-05-18 09:47:30.872 D/Kable: Connecting to 4C:55:CC:1F:87:7E
2021-05-18 09:47:31.144 D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=16 device=4C:55:CC:1F:87:7E
2021-05-18 09:47:31.148 D/Kable: Discovering services for 4C:55:CC:1F:87:7E
2021-05-18 09:47:31.150 D/BluetoothGatt: discoverServices() - device: 4C:55:CC:1F:87:7E
2021-05-18 09:47:31.580 D/BluetoothGatt: onConnectionUpdated() - Device=4C:55:CC:1F:87:7E interval=6 latency=0 timeout=500 status=0
2021-05-18 09:47:32.038 D/BluetoothGatt: onSearchComplete() = Device=4C:55:CC:1F:87:7E Status=0
2021-05-18 09:47:32.044 D/Kable: Rewiring observers for 4C:55:CC:1F:87:7E
2021-05-18 09:47:32.045 D/Kable: Connected to 4C:55:CC:1F:87:7E
2021-05-18 09:47:32.049 I/System.out: Set notification observer
2021-05-18 09:47:32.075 D/BluetoothGatt: onConnectionUpdated() - Device=4C:55:CC:1F:87:7E interval=9 latency=0 timeout=400 status=0
2021-05-18 09:47:32.139 D/Kable: setCharacteristicNotification(true) for LazyCharacteristic(serviceUuid=175f8f23-a570-49bd-9627-815a6a27de2a, characteristicUuid=cacc07ff-ffff-4c48-8fae-a9ef71b75e26)
2021-05-18 09:47:32.139 D/BluetoothGatt: setCharacteristicNotification() - uuid: cacc07ff-ffff-4c48-8fae-a9ef71b75e26 enable: true
2021-05-18 09:47:32.142 D/Kable: Writing 2 bytes to descriptor 00002902-0000-1000-8000-00805f9b34fb
2021-05-18 09:47:32.188 D/Kable: Wrote OnDescriptorWrite(descriptor=android.bluetooth.BluetoothGattDescriptor@f82461d, status=GATT_SUCCESS(0))
2021-05-18 09:47:32.189 D/Kable: Write to 00002902-0000-1000-8000-00805f9b34fb complete
2021-05-18 09:47:32.189 D/Kable: Wrote ENABLE_NOTIFICATION_VALUE to config PlatformDescriptor(serviceUuid=175f8f23-a570-49bd-9627-815a6a27de2a, characteristicUuid=cacc07ff-ffff-4c48-8fae-a9ef71b75e26, descriptorUuid=00002902-0000-1000-8000-00805f9b34fb)
```

At this point we're part of the way spinning up the observation, when the peripheral sends us a characteristic change:

```
2021-05-18 09:47:32.189 D/Kable: Received OnCharacteristicChanged(characteristic=cacc07ff-ffff-4c48-8fae-a9ef71b75e26, value=14 bytes)
```

Kable is finishing up the observation setup:

```
2021-05-18 09:47:32.190 D/Kable: Writing 2 bytes to descriptor 00002902-0000-1000-8000-00805f9b34fb
2021-05-18 09:47:32.211 D/Kable: Wrote OnDescriptorWrite(descriptor=android.bluetooth.BluetoothGattDescriptor@f82461d, status=GATT_SUCCESS(0))
2021-05-18 09:47:32.211 D/Kable: Write to 00002902-0000-1000-8000-00805f9b34fb complete
2021-05-18 09:47:32.212 D/Kable: Wrote ENABLE_INDICATION_VALUE to config PlatformDescriptor(serviceUuid=175f8f23-a570-49bd-9627-815a6a27de2a, characteristicUuid=cacc07ff-ffff-4c48-8fae-a9ef71b75e26, descriptorUuid=00002902-0000-1000-8000-00805f9b34fb)
```

Kable has now finished setting up the observation.
It looks like you begin your writing of `SetMode: 01` (I believe):

```
2021-05-18 09:47:35.866 D/Kable: Writing 1 bytes to characteristic 20b9794f-da1a-4d14-8014-a0fb9cefb2f7
2021-05-18 09:47:35.899 D/Kable: Wrote OnCharacteristicWrite(characteristic=android.bluetooth.BluetoothGattCharacteristic@342e092, status=GATT_SUCCESS(0))
2021-05-18 09:47:35.900 D/Kable: Write to 20b9794f-da1a-4d14-8014-a0fb9cefb2f7 complete
2021-05-18 09:47:36.028 I/System.out: SetMode: 01
```

Then another write to a different characteristic (21 bytes this time):

```
2021-05-18 09:47:36.032 D/Kable: Writing 21 bytes to characteristic 1cce1ea8-bd34-4813-a00a-c76e028fadcb
2021-05-18 09:47:36.035 D/Kable: Wrote OnCharacteristicWrite(characteristic=android.bluetooth.BluetoothGattCharacteristic@6d24a63, status=GATT_SUCCESS(0))
2021-05-18 09:47:36.035 D/Kable: Write to 1cce1ea8-bd34-4813-a00a-c76e028fadcb complete
```

Another 18 bytes to same characteristic again:

```
2021-05-18 09:47:36.038 D/Kable: Writing 18 bytes to characteristic 1cce1ea8-bd34-4813-a00a-c76e028fadcb
2021-05-18 09:47:36.040 D/Kable: Wrote OnCharacteristicWrite(characteristic=android.bluetooth.BluetoothGattCharacteristic@6d24a63, status=GATT_SUCCESS(0))
2021-05-18 09:47:36.041 D/Kable: Write to 1cce1ea8-bd34-4813-a00a-c76e028fadcb complete
2021-05-18 09:47:36.041 I/System.out: Command sent
```

Then later on, the disconnect:

```
2021-05-18 09:47:50.854 D/BluetoothGatt: cancelOpen() - device: 4C:55:CC:1F:87:7E
2021-05-18 09:47:50.860 D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=16 device=4C:55:CC:1F:87:7E
2021-05-18 09:47:50.861 D/BluetoothGatt: close()
2021-05-18 09:47:50.861 D/BluetoothGatt: unregisterApp() - mClientIf=16
2021-05-18 09:47:50.863 D/BluetoothGatt: close()
2021-05-18 09:47:50.863 D/BluetoothGatt: unregisterApp() - mClientIf=0
2021-05-18 09:47:50.868 D/Kable: Disposing of 4C:55:CC:1F:87:7E
2021-05-18 09:47:50.868 D/BluetoothGatt: close()
2021-05-18 09:47:50.868 D/BluetoothGatt: unregisterApp() - mClientIf=0
```

---

It appears as though your peripheral to triggering a characteristic change immediately upon enabling Notify. This isn't inherently bad, but you never receive another characteristic change.

We're essentially seeing the following sequence:

1. enable observation (i.e. enable Notify)
2. receive characteristic change
3. write set mode
4. write command
5. disconnect

I would've expected that your peripheral would trigger more characteristic changes after you "write set mode" and/or "write command", so a sequence more like:

1. enable observation (i.e. enable Notify)
2. write set mode
3. **receive characteristic change**
4. write command
5. **receive characteristic change**
6. disconnect

Not sure if I am misinterpreting how your peripheral is supposed to operate?

nspeeD, 2021-05-18:
As you comment here

> I would've expected that your peripheral would trigger more characteristic changes after you "write set mode" and/or "write command", so a sequence more like:
> 
> enable observation (i.e. enable Notify)
> write set mode
> receive characteristic change
> write command
> receive characteristic change
> disconnect
> Not sure if I am misinterpreting how your peripheral is supposed to operate?

This is the expected behaviour and we are getting this when we use a write characteristic with a descriptor.

twyatt, 2021-05-18:
> This is the expected behavior and we are getting this when we use a write characteristic with a descriptor.

I guess my confusion is that "receive characteristic change" is data sent from your peripheral. I'm not sure what Kable would need to do differently to get your peripheral to trigger the "characteristic change" upon writing to your characteristic.

I did notice that a "characteristic change" was received prior to any data being written to the peripheral, that seems like something was sent prematurely from the peripheral? I can see in Kable where this data could've been missed, so I can make a fix in Kable to ensure that early (unprovoked) characteristic changes aren't missed, but I'm worried that won't fix your situation entirely — as you never receive data for the other 2 writes you do.

I wasn't sure of privacy concerns with the data you're passing over the wire, but I can update the logging to log the hex value of the data being sent/received so you get a clearer picture of what data went where. Then you can omit the raw data from logs you provide (if there is any privacy issues).

twyatt, 2021-05-18:
Pushed `0.5.0-issue-99-3-SNAPSHOT` which logs the data (as hex) sent over the air.

twyatt, 2021-05-18:
Pushed `0.5.0-issue-99-5-SNAPSHOT` (borked `4`, so skipping to `5` was intentional).
This version should spin up the observation in a way that it doesn't loose early characteristic changes. Let me know if it works better for you.
Thanks!

nspeeD, 2021-05-18:
I'm not seeing any message in the device logs about the "characteristic change". Let me explain the following logs comparing the Kable logs and the logs from my device:

This first few logs are from my device, take in mind that the our have a difference of 2 hours but the minutes and seconds are synchronized:

    ##ble## << recv, at Tue May 18 08:26:20 2021 (47.15 sec after start-up, 23.654776 sec after entering)
    ##ble##recv## Received message: EaE!{"met":"r_lck","par":null,"id":0}f
    ##ble## << recv, at Tue May 18 08:26:20 2021 (47.15 sec after start-up, 23.655232 sec after entering)
    ##ble## >> touch_redis_presence_lock, at Tue May 18 08:26:20 2021 (47.15 sec after start-up)
    ##ble## << touch_redis_presence_lock, at Tue May 18 08:26:20 2021 (47.15 sec after start-up, 0.000904 sec after entering)
    ##ble## >> dispatch, at Tue May 18 08:26:20 2021 (47.15 sec after start-up)
    ##ble## >> dispatch_method_selection, at Tue May 18 08:26:20 2021 (47.15 sec after start-up)
    ##ble## >> method_r_lck, at Tue May 18 08:26:20 2021 (47.15 sec after start-up)
    ##ble## << method_r_lck, at Tue May 18 08:26:20 2021 (47.15 sec after start-up, 0.004332 sec after entering)
    ##ble## << dispatch_method_selection, at Tue May 18 08:26:20 2021 (47.15 sec after start-up, 0.004809 sec after entering)
    ##ble## >> send, at Tue May 18 08:26:20 2021 (47.15 sec after start-up)
    ##ble## >> update_redis_timestamp, at Tue May 18 08:26:20 2021 (47.15 sec after start-up)
    ##ble## << update_redis_timestamp, at Tue May 18 08:26:20 2021 (47.15 sec after start-up, 0.000822 sec after entering)
    ##ble## >> message_format, at Tue May 18 08:26:20 2021 (47.15 sec after start-up)
    ##ble## << message_format, at Tue May 18 08:26:20 2021 (47.15 sec after start-up, 0.000075 sec after entering)
    ##ble##send## original: {
        "id" : 0,
        "r" : 1
    }

    ##ble##send## formatted: {"id":0,"r":1}
    ##ble## << send, at Tue May 18 08:26:20 2021 (47.15 sec after start-up, 0.001165 sec after entering)
    ##ble## << dispatch, at Tue May 18 08:26:20 2021 (47.15 sec after start-up, 0.006464 sec after entering)
    ##ble##main## Received message: {
        "id" : 0,
        "met" : "r_lck",
        "par" : null
    }

    ##ble##main## Response message: {
        "id" : 0,
        "r" : 1
    }

    ##ble##main## Loop duration (millis): 6
    ##ble##main## Loop ended at: 2021-05-18 08:26:20

and this are the Kable logs:

    2021-05-18 10:26:14.869 D/BluetoothGatt: connect() - device: 4C:55:CC:1E:73:D0, auto: false
    2021-05-18 10:26:14.869 D/BluetoothGatt: registerApp()
    2021-05-18 10:26:14.871 D/BluetoothGatt: registerApp() - UUID=c761fa3f-d3b0-4739-bf5d-93ca6a434496
    2021-05-18 10:26:14.875 D/BluetoothGatt: onClientRegistered() - status=0 clientIf=17
    2021-05-18 10:26:14.881 D/Kable: Connecting to 4C:55:CC:1E:73:D0
    2021-05-18 10:26:15.117 D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=17 device=4C:55:CC:1E:73:D0
    2021-05-18 10:26:15.121 D/Kable: Discovering services for 4C:55:CC:1E:73:D0
    2021-05-18 10:26:15.130 D/BluetoothGatt: discoverServices() - device: 4C:55:CC:1E:73:D0
    2021-05-18 10:26:15.549 D/BluetoothGatt: onConnectionUpdated() - Device=4C:55:CC:1E:73:D0 interval=6 latency=0 timeout=500 status=0
    2021-05-18 10:26:15.950 D/BluetoothGatt: onSearchComplete() = Device=4C:55:CC:1E:73:D0 Status=0
    2021-05-18 10:26:15.958 D/Kable: Rewiring observers for 4C:55:CC:1E:73:D0
    2021-05-18 10:26:15.959 D/Kable: Connected to 4C:55:CC:1E:73:D0
    2021-05-18 10:26:15.961 I/System.out: Set notification observer
    2021-05-18 10:26:16.021 D/BluetoothGatt: onConnectionUpdated() - Device=4C:55:CC:1E:73:D0 interval=9 latency=0 timeout=400 status=0
    2021-05-18 10:26:16.028 D/Kable: setCharacteristicNotification(true) for LazyCharacteristic(serviceUuid=175f8f23-a570-49bd-9627-815a6a27de2a, characteristicUuid=cacc07ff-ffff-4c48-8fae-a9ef71b75e26)
    2021-05-18 10:26:16.028 D/BluetoothGatt: setCharacteristicNotification() - uuid: cacc07ff-ffff-4c48-8fae-a9ef71b75e26 enable: true
    2021-05-18 10:26:16.032 D/Kable: Writing 2 bytes to descriptor 00002902-0000-1000-8000-00805f9b34fb
    2021-05-18 10:26:16.067 D/Kable: Wrote OnDescriptorWrite(descriptor=android.bluetooth.BluetoothGattDescriptor@f82461d, status=GATT_SUCCESS(0))
    2021-05-18 10:26:16.068 D/Kable: Write to 00002902-0000-1000-8000-00805f9b34fb complete
    2021-05-18 10:26:16.068 D/Kable: Wrote ENABLE_NOTIFICATION_VALUE to config PlatformDescriptor(serviceUuid=175f8f23-a570-49bd-9627-815a6a27de2a, characteristicUuid=cacc07ff-ffff-4c48-8fae-a9ef71b75e26, descriptorUuid=00002902-0000-1000-8000-00805f9b34fb)
    2021-05-18 10:26:16.069 D/Kable: Writing 2 bytes to descriptor 00002902-0000-1000-8000-00805f9b34fb
    2021-05-18 10:26:16.088 D/Kable: Wrote OnDescriptorWrite(descriptor=android.bluetooth.BluetoothGattDescriptor@f82461d, status=GATT_SUCCESS(0))
    2021-05-18 10:26:16.089 D/Kable: Write to 00002902-0000-1000-8000-00805f9b34fb complete
    2021-05-18 10:26:16.089 D/Kable: Wrote ENABLE_INDICATION_VALUE to config PlatformDescriptor(serviceUuid=175f8f23-a570-49bd-9627-815a6a27de2a, characteristicUuid=cacc07ff-ffff-4c48-8fae-a9ef71b75e26, descriptorUuid=00002902-0000-1000-8000-00805f9b34fb)
    2021-05-18 10:26:19.872 D/Kable: Writing 1 bytes to characteristic 20b9794f-da1a-4d14-8014-a0fb9cefb2f7
[... cut ...]
    ##ble##send## original: {
        "id" : 0,
        "r" : 1
    }

At this moment, I can't request a new device software version with more logs to know what is happening behind with the characteristics but I dare to say that the issue is in the absence of the descriptor for the "writeCharacteristic" because with the same device but with another bluetooth module, which it has a descriptor in the "writeCharacteristic", everything works correctly and Kable can see the answer in the "observer".

twyatt, 2021-05-18:
> I'm not seeing any message in the device logs about the "characteristic change"

In your previous capture of logs, there was:

```
2021-05-18 09:47:32.189 D/Kable: Received OnCharacteristicChanged(characteristic=cacc07ff-ffff-4c48-8fae-a9ef71b75e26, value=14 bytes)
```

...after having enabled Notify. I'm not seeing this characteristic change event in the latest logs you've provided though.
So, possibly a red herring. 🤷

twyatt, 2021-05-18:
> And after that the device sent the response:
>
> ```
> ##ble##send## original: {
>     "id" : 0,
>     "r" : 1
> }
> ```

So, this is your peripheral's logs which signifies that it responded by triggering a characteristic change on characteristic `cacc07ff-ffff-4c48-8fae-a9ef71b75e26`, right?

The strange part is that Android never sees this message (as I'm having Kable log in Android's [`BluetoothGattCallback#onCharacteristicChanged`](https://developer.android.com/reference/android/bluetooth/BluetoothGattCallback#onCharacteristicChanged(android.bluetooth.BluetoothGatt,%20android.bluetooth.BluetoothGattCharacteristic)) — so any characteristic changes would be logged).

twyatt, 2021-05-18:
> ```
> peripheral.write(readCharacteristic, byteArrayOf(1.toByte()), WriteType.WithResponse)
> val response = peripheral.read(readCharacteristic)
> ```

Just to make sure we're on the same page, your bi-directional communication is done by writing to a characteristic and grabbing the responses via "characteristic changes" (e.g. via Notify or Indicate), right? The above quoted code snippet you provided hints that you're writing then reading (i.e. not observing by means of Notify or Indicate)?

twyatt, 2021-05-18:
Apologies, need to head to bed (have to be up for a meeting in just over 6 hours).

Really appreciate you taking to time to diagnose this, as I always love getting help in fixing bugs in Kable. Thank you!

I'm also available on kotlinlang.slack.com in `#juul-libraries` if you want to have a real-time discussion tomorrow. Via GitHub issue is also fine if you prefer just to continue the discussion here.

Thanks again!

nspeeD, 2021-05-18:
> Is this your peripheral's logs which signifies that it responded by triggering a characteristic change on characteristic cacc07ff-ffff-4c48-8fae-a9ef71b75e26?

yes! Writing on "1cce1ea8-bd34-4813-a00a-c76e028fadcb" should notify the response on "cacc07ff-ffff-4c48-8fae-a9ef71b75e26".

Logs with the version "0.5.0-issue-99-5-SNAPSHOT"

    2021-05-18 11:58:07.084 D/BluetoothGatt: connect() - device: 4C:55:CC:1E:73:D0, auto: false
    2021-05-18 11:58:07.084 D/BluetoothGatt: registerApp()
    2021-05-18 11:58:07.085 D/BluetoothGatt: registerApp() - UUID=3aa2e63f-85ef-4d51-ad60-1b39369404f2
    2021-05-18 11:58:07.090 D/BluetoothGatt: onClientRegistered() - status=0 clientIf=17
    2021-05-18 11:58:07.099 D/Kable: Connecting to 4C:55:CC:1E:73:D0
    2021-05-18 11:58:07.312 D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=17 device=4C:55:CC:1E:73:D0
    2021-05-18 11:58:07.317 D/Kable: Discovering services for 4C:55:CC:1E:73:D0
    2021-05-18 11:58:07.321 D/BluetoothGatt: discoverServices() - device: 4C:55:CC:1E:73:D0
    2021-05-18 11:58:07.751 D/BluetoothGatt: onConnectionUpdated() - Device=4C:55:CC:1E:73:D0 interval=6 latency=0 timeout=500 status=0
    2021-05-18 11:58:08.176 D/BluetoothGatt: onSearchComplete() = Device=4C:55:CC:1E:73:D0 Status=0
    2021-05-18 11:58:08.182 D/Kable: Rewiring observers for 4C:55:CC:1E:73:D0
    2021-05-18 11:58:08.182 D/Kable: Connected to 4C:55:CC:1E:73:D0
    2021-05-18 11:58:08.185 I/System.out: Set notification observer
    2021-05-18 11:58:08.246 D/BluetoothGatt: onConnectionUpdated() - Device=4C:55:CC:1E:73:D0 interval=9 latency=0 timeout=400 status=0
    2021-05-18 11:58:08.251 D/Kable: setCharacteristicNotification(true) for LazyCharacteristic(serviceUuid=175f8f23-a570-49bd-9627-815a6a27de2a, characteristicUuid=cacc07ff-ffff-4c48-8fae-a9ef71b75e26)
    2021-05-18 11:58:08.252 D/BluetoothGatt: setCharacteristicNotification() - uuid: cacc07ff-ffff-4c48-8fae-a9ef71b75e26 enable: true
    2021-05-18 11:58:08.255 D/Kable: Writing 0100 to descriptor 00002902-0000-1000-8000-00805f9b34fb
    2021-05-18 11:58:08.294 D/Kable: OnDescriptorWrite(descriptor=00002902-0000-1000-8000-00805f9b34fb, status=GATT_SUCCESS(0))
    2021-05-18 11:58:08.296 D/Kable: Write to descriptor 00002902-0000-1000-8000-00805f9b34fb complete
    2021-05-18 11:58:08.296 D/Kable: Wrote ENABLE_NOTIFICATION_VALUE to config PlatformDescriptor(serviceUuid=175f8f23-a570-49bd-9627-815a6a27de2a, characteristicUuid=cacc07ff-ffff-4c48-8fae-a9ef71b75e26, descriptorUuid=00002902-0000-1000-8000-00805f9b34fb)
    2021-05-18 11:58:08.297 D/Kable: Writing 0200 to descriptor 00002902-0000-1000-8000-00805f9b34fb
    2021-05-18 11:58:08.347 D/Kable: OnDescriptorWrite(descriptor=00002902-0000-1000-8000-00805f9b34fb, status=GATT_SUCCESS(0))
    2021-05-18 11:58:08.348 D/Kable: Write to descriptor 00002902-0000-1000-8000-00805f9b34fb complete
    2021-05-18 11:58:08.349 D/Kable: Wrote ENABLE_INDICATION_VALUE to config PlatformDescriptor(serviceUuid=175f8f23-a570-49bd-9627-815a6a27de2a, characteristicUuid=cacc07ff-ffff-4c48-8fae-a9ef71b75e26, descriptorUuid=00002902-0000-1000-8000-00805f9b34fb)
    2021-05-18 11:58:12.099 D/Kable: Writing 01 to characteristic 20b9794f-da1a-4d14-8014-a0fb9cefb2f7
    2021-05-18 11:58:12.156 D/Kable: OnCharacteristicWrite(characteristic=20b9794f-da1a-4d14-8014-a0fb9cefb2f7, status=GATT_SUCCESS(0))
    2021-05-18 11:58:12.157 D/Kable: Write to characteristic 20b9794f-da1a-4d14-8014-a0fb9cefb2f7 complete
    2021-05-18 11:58:12.296 I/System.out: SetMode: 01
    2021-05-18 11:58:12.300 D/Kable: Writing 456145217B226D6574223A22725F6C636B222C2270 to characteristic 1cce1ea8-bd34-4813-a00a-c76e028fadcb
    2021-05-18 11:58:12.302 D/Kable: OnCharacteristicWrite(characteristic=1cce1ea8-bd34-4813-a00a-c76e028fadcb, status=GATT_SUCCESS(0))
    2021-05-18 11:58:12.304 D/Kable: Write to characteristic 1cce1ea8-bd34-4813-a00a-c76e028fadcb complete
    2021-05-18 11:58:12.305 D/Kable: Writing 706172223A6E756C6C2C226964223A307D66 to characteristic 1cce1ea8-bd34-4813-a00a-c76e028fadcb
    2021-05-18 11:58:12.307 D/Kable: OnCharacteristicWrite(characteristic=1cce1ea8-bd34-4813-a00a-c76e028fadcb, status=GATT_SUCCESS(0))
    2021-05-18 11:58:12.309 D/Kable: Write to characteristic 1cce1ea8-bd34-4813-a00a-c76e028fadcb complete
    2021-05-18 11:58:12.309 I/System.out: Command sent: EaE!{"met":"r_lck","par":null,"id":0}f
    2021-05-18 11:58:27.071 D/BluetoothGatt: cancelOpen() - device: 4C:55:CC:1E:73:D0
    2021-05-18 11:58:27.076 D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=17 device=4C:55:CC:1E:73:D0
    2021-05-18 11:58:27.077 D/BluetoothGatt: close()
    2021-05-18 11:58:27.077 D/BluetoothGatt: unregisterApp() - mClientIf=17
    2021-05-18 11:58:27.078 D/BluetoothGatt: close()
    2021-05-18 11:58:27.078 D/BluetoothGatt: unregisterApp() - mClientIf=0
    2021-05-18 11:58:27.080 D/Kable: Disposing of 4C:55:CC:1E:73:D0
    2021-05-18 11:58:27.080 D/BluetoothGatt: close()
    2021-05-18 11:58:27.080 D/BluetoothGatt: unregisterApp() - mClientIf=0

nspeeD, 2021-05-18:
> Just to make sure we're on the same page, your bi-directional communication is done by writing to a characteristic and grabbing the responses via "characteristic changes" (e.g. via Notify or [...]

We are using two ways to write/read:

- One with the classic write and then read over the readCharacteristic (20b9794f-da1a-4d14-8014-a0fb9cefb2f7)
- Second one writting on the writeCharacteristic(1cce1ea8-bd34-4813-a00a-c76e028fadcb) and waiting for the notification in (cacc07ff-ffff-4c48-8fae-a9ef71b75e26)

> I'm also available on kotlinlang.slack.com in #juul-libraries if you want to have a real-time discussion tomorrow. Via GitHub issue is also fine if you prefer just to continue the discussion here.

I was trying to join with my corporate account but I can't because I need a @jetbrains account.

> Don’t have an @jetbrains.com email address?
> Contact the workspace administrator at kotlinlang for an invitation.

As you prefer! Always glad to help!!

twyatt, 2021-05-18:
> I was trying to join with my corporate account but I can't because I need a @JetBrains account.

You shouldn't need a JetBrains account, though it is invitation based. Happy to keep the conversation going on this issue if you have trouble with kotlinlang Slack.

> We are using two ways to write/read:
>
> * One with the classic write and then read over the readCharacteristic (20b9794f-da1a-4d14-8014-a0fb9cefb2f7)
> * Second one writting on the [...]

Seems like this should work. The second method (dedicate characteristic for write and dedicated characteristic for read via notification) should definitely work, we use that method extensively on our internal libraries that use Kable and it is working well.

Based on the most recent logs you provided, everything looks good on the phone side. I'm running out of ideas why your peripheral is not triggering notifications changes. One remaining _shot-in-the-dark_ thing to try, is I'll remove enabling Indicate (perhaps enabling both Notify and Indicate is causing problems with your peripheral?):

If you could, give `0.5.0-issue-99-6-SNAPSHOT` a try?

Thanks!

nspeeD, 2021-05-18:
> You shouldn't need a JetBrains account, though it is invitation based. Happy to keep the conversation going on this issue if you have trouble with kotlinlang Slack.

If you can invite me ... jeje

Well with this last version "0.5.0-issue-99-6-SNAPSHOT", I'm able to receive the response in the "observe"!! 👏 👏

twyatt, 2021-05-18:
> If you can invite me ... jeje

You can request an invite [here](https://surveys.jetbrains.com/s3/kotlin-slack-sign-up?_ga=2.180718119.1286744683.1621365441-236598189.1596145766).

> Well with this last version "0.5.0-issue-99-6-SNAPSHOT", I'm able to receive the response in the "observe"!! 👏 👏

Yay! I'll create a fix soon that only sets one (either Notify or Indicate) and never both. Thanks so much for diagnosing the issue with me!

FilippoVigani, 2021-05-19:
I'm facing the same issue (Android client). I'm not sure whether the BLE device sets up the proper descriptors (I'm afraid not), but with the library RxAndroidBle I properly receive the notifications.

The process I follow is:
1. Set up notification on characteristic `2b3247fd-28db-413c-934f-aa4c271984c9`
2. Write to characteristic `2b3247fd-28db-413c-934f-aa4c271984c9`
3. Should receive a notification at this point

I'm not sure if having the same characteristic for write and notification is ideal, but that's how the firmware works so I can't change that.

On the console I correctly see `D/BluetoothGatt: setCharacteristicNotification() - uuid: 2b3247fd-28db-413c-934f-aa4c271984c9 enable: true`, and I also write to console a message after `peripheral.write(characteristic, bytes`, which does print out. 

Happy to help debug this if I can.

FilippoVigani, 2021-05-19:
I tried with the snapshot `0.5.0-issue-99-6-SNAPSHOT` and this is the logs I'm getting:

```
2021-05-19 14:18:01.161 6150-6193/it.labfabrici.hub.test_app D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=16 mScannerId=0
2021-05-19 14:18:01.238 6150-6150/it.labfabrici.hub.test_app D/BluetoothAdapter: isLeEnabled(): ON
2021-05-19 14:18:01.258 6150-6150/it.labfabrici.hub.test_app D/BluetoothGatt: connect() - device: DA:A9:FA:6E:D7:01, auto: false
2021-05-19 14:18:01.258 6150-6150/it.labfabrici.hub.test_app D/BluetoothGatt: registerApp()
2021-05-19 14:18:01.272 6150-6150/it.labfabrici.hub.test_app D/BluetoothGatt: registerApp() - UUID=3d3f6cf2-0372-4520-a886-8d9c37ce0dc5
2021-05-19 14:18:01.278 6150-6189/it.labfabrici.hub.test_app D/BluetoothGatt: onClientRegistered() - status=0 clientIf=16
2021-05-19 14:18:01.282 6150-6150/it.labfabrici.hub.test_app D/Kable: Connecting to DA:A9:FA:6E:D7:01
2021-05-19 14:18:01.520 6150-6189/it.labfabrici.hub.test_app D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=16 device=DA:A9:FA:6E:D7:01
2021-05-19 14:18:01.522 6150-6150/it.labfabrici.hub.test_app D/Kable: Discovering services for DA:A9:FA:6E:D7:01
2021-05-19 14:18:01.524 6150-6397/it.labfabrici.hub.test_app D/BluetoothGatt: discoverServices() - device: DA:A9:FA:6E:D7:01
2021-05-19 14:18:01.963 6150-6189/it.labfabrici.hub.test_app D/BluetoothGatt: onConnectionUpdated() - Device=DA:A9:FA:6E:D7:01 interval=6 latency=0 timeout=500 status=0
2021-05-19 14:18:02.110 6150-6189/it.labfabrici.hub.test_app D/BluetoothGatt: onSearchComplete() = Device=DA:A9:FA:6E:D7:01 Status=0
2021-05-19 14:18:02.118 6150-6150/it.labfabrici.hub.test_app D/Kable: Rewiring observers for DA:A9:FA:6E:D7:01
2021-05-19 14:18:02.119 6150-6150/it.labfabrici.hub.test_app D/Kable: Connected to DA:A9:FA:6E:D7:01
2021-05-19 14:18:02.162 6150-6436/it.labfabrici.hub.test_app D/Kable: setCharacteristicNotification(true) for LazyCharacteristic(serviceUuid=c954c6e6-3016-4416-a72f-b184a4f039e0, characteristicUuid=2b3247fd-28db-413c-934f-aa4c271984c9)
2021-05-19 14:18:02.163 6150-6436/it.labfabrici.hub.test_app D/BluetoothGatt: setCharacteristicNotification() - uuid: 2b3247fd-28db-413c-934f-aa4c271984c9 enable: true
2021-05-19 14:18:02.166 6150-6397/it.labfabrici.hub.test_app D/Kable: Writing 0100 to descriptor 00002902-0000-1000-8000-00805f9b34fb
2021-05-19 14:18:02.187 6150-6189/it.labfabrici.hub.test_app D/BluetoothGatt: onConnectionUpdated() - Device=DA:A9:FA:6E:D7:01 interval=36 latency=0 timeout=500 status=0
2021-05-19 14:18:02.236 6150-6397/it.labfabrici.hub.test_app D/Kable: OnDescriptorWrite(descriptor=00002902-0000-1000-8000-00805f9b34fb, status=GATT_SUCCESS(0))
2021-05-19 14:18:02.241 6150-6436/it.labfabrici.hub.test_app D/Kable: Write to descriptor 00002902-0000-1000-8000-00805f9b34fb complete
2021-05-19 14:18:02.246 6150-6436/it.labfabrici.hub.test_app D/Kable: Wrote ENABLE_NOTIFICATION_VALUE to config PlatformDescriptor(serviceUuid=c954c6e6-3016-4416-a72f-b184a4f039e0, characteristicUuid=2b3247fd-28db-413c-934f-aa4c271984c9, descriptorUuid=00002902-0000-1000-8000-00805f9b34fb)
2021-05-19 14:18:02.249 6150-6436/it.labfabrici.hub.test_app D/Kable: PlatformCharacteristic(serviceUuid=c954c6e6-3016-4416-a72f-b184a4f039e0, characteristicUuid=2b3247fd-28db-413c-934f-aa4c271984c9) does not support Indicate
2021-05-19 14:18:04.945 6150-6397/it.labfabrici.hub.test_app D/Kable: Writing 8A00 to characteristic 2b3247fd-28db-413c-934f-aa4c271984c9
2021-05-19 14:18:04.950 6150-6397/it.labfabrici.hub.test_app D/Kable: OnCharacteristicWrite(characteristic=2b3247fd-28db-413c-934f-aa4c271984c9, status=GATT_SUCCESS(0))
2021-05-19 14:18:04.952 6150-6436/it.labfabrici.hub.test_app D/Kable: Write to characteristic 2b3247fd-28db-413c-934f-aa4c271984c9 complete```

Still no notification though.

twyatt, 2021-05-19:
> Good news, by using `0.5.0-issue-99-6-SNAPSHOT` **and** calling `peripheral.write(characteristic, bytes, WriteType.WithResponse)` I get the notification.
> 
> At this point I'm afraid I can't tell whether the firmware just works like that, or it's some weird interaction with the library.

Thanks for sharing the workaround you found. In an internal library we use (that is powered by Kable) we've also seen the strange behavior you've described (needing to write before the observation seems to work). Not sure if we're experiencing the same issue as you, or both are facing coincidentally similar firmware issues. In either case, we'll also be investigating the issue internally and I'll report back here if/when we determine the cause.

FilippoVigani, 2021-05-19:
> Thanks for sharing the workaround you found. In an internal library we use (that is powered by Kable) we've also seen the strange behavior you've described (needing to write before the observation [...]

What did it for me was explicitly adding the `WithResponse` parameter. By default it is `WithoutResponse`. It may be worthwhile documenting this behavior in the library readme.

twyatt, 2021-05-19:
> What did it for me was explicitly adding the `WithResponse` parameter. By default it is `WithoutResponse`.

Interesting, we're using `WithoutResponse` internally, so perhaps we're experiencing similar problematic behavior but possible not related.

> It may be worthwhile documenting this behavior in the library readme.

Calling out the different write types definitely seems valuable to have in the `README`, I'll add that soon (#104). 👍

FilippoVigani, 2021-05-21:
After further testing (using 0.5.0), it still looks unreliable. Sometimes I properly get the notifications, other times I do not. Reconnecting changes whether I get them or not.

PR #121 (2021-06-23): Only write indicate or notify to CCCD on Android
Closes #99

Some peripherals have characteristics that support either notification or indication, when we're observing such a characteristic, we should write to the Client Characteristic Configuration descriptor (CCCD) either to enable notifications **or** indications, **not** both.

We default to only **notifications** if characteristic supports both. This is consistent with the documented behavior of [Core Bluetooth].

[Core Bluetooth]: https://developer.apple.com/documentation/corebluetooth/cbperipheral/1518949-setnotifyvalue
closed 2021-06-23 by twyatt