JuulLabs/kable #102: [Android] peripheral.read() hangs after manually disabling bluetooth

wrsx, 2021-05-18:
We're frequently polling `peripheral.read()` on a background coroutine and noticed that manually disabling bluetooth will not always throw an exception

An exception _will_ be thrown if the read occurs after the OS disconnects from the device:

> 2021-05-18 11:39:35.032 14297-15286/com.v I/FTRepository: start read
2021-05-18 11:39:35.471 14297-15286/com.v I/FTRepository: finish read
2021-05-18 11:39:35.481 14297-15286/com.v I/FTRepository: start read
2021-05-18 11:39:35.650 14297-15286/com.v I/FTRepository: finish read
2021-05-18 11:39:35.654 14297-16277/com.v D/BluetoothGatt: onBluetoothStateChange: up=false
2021-05-18 11:39:35.654 14297-16277/com.v D/BluetoothGatt: Bluetooth is turned off, disconnect all client connections
2021-05-18 11:39:35.654 14297-16277/com.v D/BluetoothGatt: close()
2021-05-18 11:39:35.658 14297-16277/com.v D/BluetoothGatt: unregisterApp() - mClientIf=15
2021-05-18 11:39:35.661 14297-15286/com.v I/FTRepository: start read
2021-05-18 11:39:35.662 14297-15286/com.v E/FTRepository: com.juul.kable.GattRequestRejectedException

And exception will _not_ be thrown if the read occurs before the OS disconnects:

> 2021-05-18 11:11:36.786 11390-12135/com.v I/FTRepository: start read
2021-05-18 11:11:36.956 11390-12135/com.v I/FTRepository: finish read
2021-05-18 11:11:36.966 11390-12135/com.v I/FTRepository: start read
2021-05-18 11:11:37.135 11390-12135/com.v I/FTRepository: finish read
2021-05-18 11:11:37.146 11390-12135/com.v I/FTRepository: start read <-- hangs forever
2021-05-18 11:11:37.251 11390-12401/com.v D/BluetoothGatt: onBluetoothStateChange: up=false
2021-05-18 11:11:37.251 11390-12401/com.v D/BluetoothGatt: Bluetooth is turned off, disconnect all client connections
2021-05-18 11:11:37.251 11390-12401/com.v D/BluetoothGatt: close()
2021-05-18 11:11:37.254 11390-12401/com.v D/BluetoothGatt: unregisterApp() - mClientIf=15
2021-05-18 11:11:37.255 11390-12200/com.v I/BluetoothAdapter: onBluetoothStateChange: up=false

twyatt, 2021-07-07:
@wrsx can you try the following `SNAPSHOT` and check if it resolves your issue (i.e. throws when BLE is turned off rather than hanging)?

```
repositories {
    maven("https://oss.sonatype.org/content/repositories/snapshots")
}

dependencies {
    implementation("com.juul.kable:core:0.6.0-issue-102-1-SNAPSHOT")
}
```

Thank you!

wrsx, 2021-07-12:
Just tested and seems to be disconnecting now, thanks!

twyatt, 2021-07-12:
Thanks for testing @wrsx, we'll be releasing 0.7.0 soon and it will include this fix.

twyatt, 2021-07-13:
Fix is included in 0.7.0 release.

closed 2021-07-13 by twyatt