JuulLabs/kable #105: Early characteristic changes can be missed

twyatt, 2021-05-19:
As was seen in https://github.com/JuulLabs/kable/issues/99#issuecomment-842941308, an unprovoked (prior to writing to the config descriptor to enable Notify/Indicate) characteristic may trigger a "characteristic change" event.

Code used in test branch that should be brought in to main code:

`Observers.kt`
```kotlin
    fun acquire(characteristic: Characteristic) = flow {
        try {
            characteristicChanges
                .onSubscription {
                    peripheral.suspendUntilReady()

                    if (observers.incrementAndGet(characteristic) == 1) {
                        peripheral.startObservation(characteristic)
                    }
                }
                .collect {
                    if (it.characteristic.characteristicUuid == characteristic.characteristicUuid &&
                        it.characteristic.serviceUuid == characteristic.serviceUuid
                    ) emit(it.data)
                }
        } finally {
            if (observers.decrementAndGet(characteristic) < 1) {
                try {
                    peripheral.stopObservation(characteristic)
                } catch (e: NotReadyException) {
                    // Silently ignore as it is assumed that failure is due to connection drop, in which case Android
                    // will clear the notifications.
                    Log.d(TAG, "Stop notification failure ignored.")
                }
            }
        }
    }
```

gryffindor14, 2021-05-28:
As @FilippoVigani mentioned in the other thread, this behavior works in RxAndroidBLE. This was an issue I previously ran into with the RxAndroidBLE library and I filed a bug there. It contains a similar log of events. https://github.com/Polidea/RxAndroidBle/issues/195. 

Similar to the Characteristic write chain of events in the other post, the Descriptor write and the characteristic change event come back on 2 different OS threads and have a race condition for which emits first. In our case, the characteristic doesn't have a read property so it isn't able to get the current value. The solution for this is to observe the characteristic before writing the ENABLE_INDICATIONS/ENABLE_NOTIFICATIONS descriptor value. This became tricky because with the queue there is usually an inherit FIFO contract. During testing it was at most 1 characteristic that was "lost", but there could have been more in other scenarios. In the end, I believe it was solved in the RxAndroidBLE library by adding a replay cache with a size of 1 to the characteristic observable. If the characteristic change was emitted before the descriptor write was emitted, then the latest value emitted before the descriptor write completed would be emitted again. 

Please note that I could be way off if that is what is happening here. I have not done a deep dive with Kable yet, but this is one of the first things I was looking at.

twyatt, 2021-05-28:
@gryffindor14 thanks for the additional information!
Based on what you described, I believe the proposed solution in the issue write-up should cover what you're describing.

The magic lies in the [`onSubscription`](https://kotlin.github.io/kotlinx.coroutines/kotlinx-coroutines-core/kotlinx.coroutines.flow/on-subscription.html):

> It is guaranteed that all emissions to the upstream flow that happen inside or immediately after this `onSubscription` action will be collected by this subscription.

We'll setup the observe (i.e. write the appropriate value to the config descriptor) within the `onSubscription` and we shouldn't miss any characteristic changes.

I have a local branch I've started for this effort. I'll try to get a PR (and `SNAPSHOT` available) soon with the fix so people can try and confirm if it actually fixes the issue.

twyatt, 2021-06-09:
A `SNAPSHOT` build is available for testing.

```
repositories {
    maven("https://oss.sonatype.org/content/repositories/snapshots")
}

dependencies {
    implementation("com.juul.kable:core:0.5.0-issue-105-SNAPSHOT")
}
```

PR #110 (2021-06-23): Propagate early characteristic changes
Closes #105
Closes #112

Also fixes JavaScript observations to wait until connection is ready. Only users of experimental advertisement support in Chrome would've seen this bug, as using `requestDevice` obtains devices in a connected state, i.e. already ready.
closed 2021-06-23 by twyatt