JuulLabs/kable #106: Support JavaScript IR Compiler

davidtaylor-juul, 2021-06-03:
To fully support Kotlin MPP there is a need to upgrade to Kotlin 1.5.x and enable compilation for both LEGACY and IR artifacts for the JavaScript target.

This may also require/allow changing from `binaries.executable()` to `binaries.library()` packaging in the `js() { }` Gradle block.

twyatt, 2021-06-03:
After I've tested #100 a bit more and merge it, then I'll give enabling IR a try. 👍

SebastianAigner, 2021-06-09:
Hi folks, Sebastian here, Kotlin/JS dev advocate at JetBrains! Awesome to see you're working on supporting the IR compiler. Please don't hesitate to reach out if you run into any trouble – we want to make the transition to the new compiler backend as smooth as possible :)

Feel free to reach out in the Kotlinlang Slack, for example, or via my other channels: https://kotlinlang.slack.com/team/UJ3MHFMQD

PR #126 (2021-07-02): JavaScript IR Support
Resolves #106
closed 2021-07-02 by KVelasco