JuulLabs/kable #111: Support for bonding

DoruAdryan, 2021-06-09:


twyatt, 2021-06-09:
It isn't on the short-term roadmap but I've been trying to prioritize GitHub issues by the number of 👍  it has. So, if there are others who also need this feature (and "up vote" it) then I'd be happy to tackle it sooner rather than later.

Long-term I would like to support it, so it will come eventually. Also happy to accept PRs. 😉

twyatt, 2021-07-15:
This issue appears to be getting a fair amount of attention (👍) so I'll start trying to dedicate time to researching/brainstorming this integration.

twyatt, 2021-07-25:
Found the following articles that offer awesome insight into the bonding process:

- [Making Android BLE work — part 4](https://medium.com/@martijn.van.welie/making-android-ble-work-part-4-72a0b85cb442)
- [The Ultimate Guide to Android Bluetooth Low Energy](https://punchthrough.com/android-ble-guide/#Bonding-with-a-BLE-device)

Does anyone have any recommendations on a BLE peripheral (that needs/supports bonding) that I can purchase to start developing against?

DoruAdryan, 2021-07-25:
There are multiple medical devices that I've come across requiring pairing: 
[Thermometer](https://www.amazon.com/Beurer-FT95-Non-Contact-Thermometer-Temperature/dp/B08ST6NNX1) (requires pairing at reading measurements stage, not when connecting to it),
[Pulse Oximeter](https://www.amazon.com/Beurer-Bluetooth-Smart-Fingertip-Oximeter/dp/B076DQDCPP)
but as mentioned in those articles you posted, the pairing process could be started by the OS at different moments of communication to the bluetooth device.

twyatt, 2021-09-24:
Quick update on this, I haven't made any real progress, but I have acquired both the Thermometer and Pulse Oximeter for when I do have time to investigate this further. Thanks @DoruAdryan for the recommendations.

SkyleKayma, 2022-11-03:
Has there been any progress on this part one year later ?
I don't think this has been implemented, or else I missed it.
It's the only thing currently preventing me from using this lib, even though it's really great!

twyatt, 2022-11-03:
Unfortunately I've been really busy with projects at work, and we don't have an immediate need for bonding so I haven't been able to prioritize this effort.

I know this is the most requested feature, so I would love to make some progress on it.

Some things that would really help out in the effort is:
- Create some basic sample apps for each platform that perform bonding (without Kable)
    - Would help demonstrate all the moving parts that will be needed and help drive decisions about how to make a Coroutines based API in Kable
- Alternatively, we could start a [discussion](https://github.com/JuulLabs/kable/discussions) on what the shape of the API should be
    - I have no experience around performing bonding on any of the platforms, so getting some general understanding (outline) of what is involved w/ the process would be super helpful
    - If people have worked w/ bonding before, maybe some `interfaces` (written in pseudo-Kotlin code) of what they imagine the API could look like, could help kick off this effort

SkyleKayma, 2022-12-05:
Whenever I've done pairing it's by using the Nordic library (https://github.com/NordicSemiconductor/Android-BLE-Library), I've never implemented it from scratch. So I may not be in the best position to provide a test sample.
But I will continue to follow this library and its development closely.

twyatt, 2022-12-05:
It has recently come up that we may have a need for bonding internally. Meaning that this may end up on our team's project roadmap (so I'd be able to allocate a lot of time to this effort). 🤞 

I'm not certain of timelines, but I'm hopeful the effort kicks off sometime in first half of 2023.

medmaadellaoui, 2023-06-03:
Hello @twyatt, Just wondering, how's that feature coming along :smiley:  ?

twyatt, 2023-06-03:
> Hello @twyatt, Just wondering, how's that feature coming along 😃 ?

I just last week researched what typical Android bonding looks like, here are some great articles I came across:

- [Bluetooth LE for modern Android Development - part 2](https://www.hellsoft.se/bluetooth-le-for-modern-android-development-part-2/) by [Erik Hellman](https://www.hellsoft.se/author/hellman/)
- [Making Android BLE work — part 4](https://medium.com/@martijn.van.welie/making-android-ble-work-part-4-72a0b85cb442) by [Martijn van Welie](https://medium.com/@martijn.van.welie)
- [The Ultimate Guide to Android Bluetooth Low Energy](https://punchthrough.com/android-ble-guide/) by [Chee Yi Ong](https://punchthrough.com/author/cong)

...that is to say: sorry it has taken so long, but I have finally started on this. Thanks for your patience. ❤️

twyatt, 2023-06-05:
I have a completely untested implementation (#494), based on the great articles I linked. Although I don't (yet) have any devices to test it against.

I have a Beurer PO60 Pulse Oximeter but getting the BLE working on it was super wonky (I ultimately gave up). Additionally, according to [this](https://thejeshgn.com/2020/08/05/reverse-engineering-a-bluetooth-low-energy-oximeter/), it doesn't require bonding anyways. 🤷 

Does anyone have some recommendations of devices that:
- require bonding out-of-the-box
- are simple to setup (BLE just works — no need to flash specific firmware)
- is affordable

Thank you!

YSDC, 2023-06-26:
Any update on this? Our company would like to use your library but bonding is a key requirement.

twyatt, 2023-06-26:
@YSDC do you happen to know of a publicly available (and affordable) device that requires/supports bonding? Or, is the product your company is wanting to developer for available for use as something I can develop against?

The only blocker at this point is me getting a device I can test #494 against.

DoruAdryan, 2023-06-26:
Mentioned a couple of devices that requires bonding with different behaviors [here](https://github.com/JuulLabs/kable/issues/111#issuecomment-886179302)

twyatt, 2023-06-26:
> Mentioned a couple of devices that requires bonding with different behaviors [here](https://github.com/JuulLabs/kable/issues/111#issuecomment-886179302)

Yes, thanks for those recommendations @DoruAdryan!

Unfortunately I could not get the bluetooth on the Beurer PO60 Pulse Oximeter working (the guides in the [official app](https://play.google.com/store/apps/details?id=com.beurer.connect.healthmanager) were terrible — could not even get it connected in their app). If you happen to know how to enable bluetooth on that device, then I'll give it another try.

I didn't get very far with the Beurer Thermometer either, as it was difficult to find documentation on how to communicate with it over bluetooth (i.e. what are its characteristic UUIDs, what does the data format look like to communicate with it, etc?). I was hoping to have a device that was a little more developer friendly (specifically, better documentation) but I'll go back to reverse engineering it if no better device options come up.

CristianMG, 2023-09-06:
Hi

 I'm checking these modifications on my own. The first problem I'm facing is that 'onCharacteristicRead' inside the callback is not firing. It seems like Android is waiting for pairing because my device requires pairing from the initial connection. Maybe this behavior only occurs with devices that require pairing at the initial connection, and not with devices that only require pairing for some protected characteristics.

Perhaps we should add a function to wait for pairing or consider giving access to the bonding state from an external part and keeping that responsibility outside of this library.

Added to this I'm clear code and merge with the main in my own pull request, the broadcast ACTION_BOND_STATE_CHANGED have a problem because is not called. Can you check this?

https://github.com/CristianMG/kable/tree/feature/bonding_v2

as you can see, I have modified some problematic code.

Greetings

twyatt, 2023-09-06:
@CristianMG is the peripheral you're working with publicly available?

If not, are you able to provide me with one so that I can test the bonding support against it?
Feel free to reach out to me directly on the [Kotlin Slack](https://kotlinlang.org/community/), user `@travis`.

kdkoyamatsu, 2024-05-06:
Hello @twyatt, just wanted to check-in and see how this effort was coming along. Bonding support would be an essential requirement.

twyatt, 2024-05-07:
> Hello @twyatt, just wanted to check-in and see how this effort was coming along. Bonding support would be an essential requirement.

@kdkoyamatsu, the only hold up at this time is that I don't have a device that requires bonding to test against (to complete development of #494).

If you're aware of a publicly available BLE device that requires bonding and has API documentation (i.e. something to reference to spin up a test app for BLE communication) then I'll gladly purchase it to continue development.

twyatt, 2024-10-27:
@CristianMG I tested bonding against your [kable-ble-test](https://github.com/CristianMG/kable-ble-test) project (thanks again, by the way — it has been super helpful).

Some things I've observed thus far:
1. I've been unable to get bonding status updates via the [broadcast receiver](https://github.com/CristianMG/kable/blob/4e7f5e72c036197fab9088f801202e8c3fc45534/kable-core/src/androidMain/kotlin/BluetoothDeviceAndroidPeripheral.kt#L97-L122) — is this a bug in Android, or perhaps the broadcast receiver is configured incorrectly
2. I/O (to secured characteristics) does not result in a [failure callback from Android](BondRequiredException) — instead, the I/O success callback waits until the bonding pin has been entered by the user
3. Maybe specific to `kable-ble-test`, but if the bonding PIN is never entered, eventually the connection is dropped with a GATT 133 failure

None of the above issues are show-stoppers, but it definitely isn't the behavior I was expecting. I'll continue investigating, just sharing my findings thus far.

CristianMG, 2025-01-08:
That´s weird because we use this software in multiple productions apps which use thousands of users. We have no report about problem with bonding, moreover, I tested that project against the same firmware that you are using without any problems. About first problem, we experienced that problem en Android 13, when it was released, but never again. Please could you update with the news? Tell me if I can help you with something

CristianMG, 2025-01-08:
To be honest we use firmware provided specifically for us. That´s can be a reason we have no problems, Android reacts different with weird behaviors depends on flags sent by device.

twyatt, 2025-02-01:
I've ordered a Puck.js for use with [Web Bluetooth Secure Characteristic Manual Test](https://webbluetoothcg.github.io/manual-tests/characteristic_secure/), which should provide a well defined peripheral to test against.

So, I'm hoping to pick this effort back up again soon.

twyatt, 2025-02-10:
[Puck.js](https://www.espruino.com/Puck.js) was a bust. 😢 

I was unable to get the "Secure Characteristic" test (from [Web Bluetooth Secure Characteristic Manual Test](https://webbluetoothcg.github.io/manual-tests/characteristic_secure/)) working. The `setSecurity` call would always fail, presumably because bonding isn't supported by the Puck.js? 🤷 

I feel I've exhausted my search/hope of finding an easy to use prebuilt peripheral for testing against.

Instead, I now plan to take the longer (more effort) approach and tackle #51, which I'll use to provide the peripheral to test against for bonding.

It will probably take some time to get #51 going, but I'm hopeful I'll soon enough get back to implementing bonding support. 🤞