JuulLabs/kable #112: peripheral.observe(readCharacteristic) doesn't provide a form to know when is safe to write

Devinete, 2021-06-09:
Hello there!

I'm trying to migrate a project to kable. The ble device (and others I worked with...) uses some opcode communication where you write to a writeCharacteristic and all the data is received on the readCharacteristic subscribed to notify.
After notify is subscribed I can enqueue calls to get a first response as soon as posible.

Enter kable, the peripheral.observe(readCharacteristic) provides a flow with the data, but cant signal when it is actually able to receive data. So i can't just enqueue a write to the descriptor as doing it "on the next line" because the first response will be missed as the observe isn't really ready. I can workaround that doing a delay(2000) for example between the .observe and the write.

Maybe i've missed something but can't seem to find a method or param that handles that.
Could it be possible to set the notify/observe on the new (0.6) connect params? or maybe provide a param to the observe to set an initial write?

Keep the good work, awesome library!

twyatt, 2021-06-09:
@Devinete Is it possible that the resolution to #105 would solve your issue?

I'll publish a `SNAPSHOT` build soon that you could test.

twyatt, 2021-06-09:
@Devinete `SNAPSHOT` is available for testing: https://github.com/JuulLabs/kable/issues/105#issuecomment-857930906

Devinete, 2021-06-10:
@twyatt actually i don't need to know when is spun if i'm guaranted to don't miss any response as the only reason to know it was just to time the write that provoked a possible response miss.

I Checked the snapshot. It simply works so there's no issue anymore.
Feel free to close this issue.

Thanks.

Devinete, 2021-06-10:
@twyatt whoops.... seems i talked too soon... seems like now im missing some still on at least a device (a Nokia 4.2) will update when i test on some more devices

twyatt, 2021-06-10:
@Devinete keep me posted. If you find there is still notifications that are missed then I can spin up another `SNAPSHOT` build with logging to help us diagnose.

Devinete, 2021-06-10:
@twyatt i was just editing my original comment when you replied, not sure if github notifies about edit so heres a probably totally unnecesary reply.
If you can build an snapshot i will test it.

twyatt, 2021-06-10:
Sounds good. I'll add some logging and let you know when a `SNAPSHOT` build is available.

twyatt, 2021-06-10:
@Devinete when you have a chance, can you give the following `SNAPSHOT` a try and post logs, please?

```kotlin
repositories {
    maven("https://oss.sonatype.org/content/repositories/snapshots")
}

dependencies {
    implementation("com.juul.kable:core:0.5.1-issue-112-1-SNAPSHOT")
}
```

Devinete, 2021-06-11:
Warning, Wall of text. The first one is the one dropping.

2021-06-11 09:50:34.728 19332-19332/ D/AppViewModel$connect: readConfig
2021-06-11 09:50:34.735 19332-19404/ D/Kable: Writing 0270383E03 to characteristic 6e400002-b5a3-f393-e0a9-e50e24dcca9e
2021-06-11 09:50:34.735 19332-19332/ D/AppViewModel$connect$1$1$invokeSuspend$$inlined$collect: State com.juul.kable.State$Connected@ac5a9e4
2021-06-11 09:50:34.737 19332-19332/ D/Kable: setCharacteristicNotification(true) for LazyCharacteristic(serviceUuid=6e400001-b5a3-f393-e0a9-e50e24dcca9e, characteristicUuid=6e400003-b5a3-f393-e0a9-e50e24dcca9e)
2021-06-11 09:50:34.738 19332-19332/ D/BluetoothGatt: setCharacteristicNotification() - uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e enable: true
2021-06-11 09:50:34.741 19332-19404/ D/Kable: OnCharacteristicWrite(characteristic=6e400002-b5a3-f393-e0a9-e50e24dcca9e, status=GATT_SUCCESS(0))
2021-06-11 09:50:34.798 19332-19351/ D/BluetoothGatt: onConnectionUpdated() - Device=D1:87:F9:50:23:87 interval=39 latency=0 timeout=500 status=0
2021-06-11 09:50:34.808 19332-19332/r D/Kable: Write to characteristic 6e400002-b5a3-f393-e0a9-e50e24dcca9e complete
2021-06-11 09:50:34.808 19332-19404/r D/Kable: Writing 0100 to descriptor 00002902-0000-1000-8000-00805f9b34fb
2021-06-11 09:50:34.944 19332-19404/r D/Kable: OnCharacteristicChanged(characteristic=6e400003-b5a3-f393-e0a9-e50e24dcca9e, value=3803)
2021-06-11 09:50:34.944 19332-19404/r D/Kable: OnDescriptorWrite(descriptor=00002902-0000-1000-8000-00805f9b34fb, status=GATT_SUCCESS(0))
2021-06-11 09:50:34.972 19332-19332/ D/Kable: Write to descriptor 00002902-0000-1000-8000-00805f9b34fb complete
2021-06-11 09:50:34.973 19332-19332/ D/Kable: Wrote ENABLE_NOTIFICATION_VALUE to config PlatformDescriptor(serviceUuid=6e400001-b5a3-f393-e0a9-e50e24dcca9e, characteristicUuid=6e400003-b5a3-f393-e0a9-e50e24dcca9e, descriptorUuid=00002902-0000-1000-8000-00805f9b34fb)
2021-06-11 09:50:34.973 19332-19332/ D/Kable: PlatformCharacteristic(serviceUuid=6e400001-b5a3-f393-e0a9-e50e24dcca9e, characteristicUuid=6e400003-b5a3-f393-e0a9-e50e24dcca9e) does not support Indicate
2021-06-11 09:50:34.974 19332-19332/ V/Kable: Propagating CharacteristicChange(serviceUuid=6e400001-b5a3-f393-e0a9-e50e24dcca9e, characteristicUuid=6e400003-b5a3-f393-e0a9-e50e24dcca9e, data=3803)
2021-06-11 09:50:35.049 19332-19332/ D/AppViewModel$connect$1$2$invokeSuspend$$inlined$collect: addding to buffer 2 buffer length: 2
2021-06-11 09:50:35.050 19332-19332/ D/AppViewModel$connect$1$2$invokeSuspend$$inlined$collect: [Ble Data] Length: 2 String: 8
2021-06-11 09:50:54.295 19332-19353/r D/BluetoothGatt: onConnectionUpdated() - Device=D1:87:F9:50:23:87 interval=159 latency=0 timeout=400 status=0

2021-06-11 09:53:46.370 22172-22172/e D/AppViewModel$connect: readConfig
2021-06-11 09:53:46.378 22172-22317/e D/Kable: Writing 0270383E03 to characteristic 6e400002-b5a3-f393-e0a9-e50e24dcca9e
2021-06-11 09:53:46.378 22172-22172/e D/AppViewModel$connect$1$1$invokeSuspend$$inlined$collect: State com.juul.kable.State$Connected@b0b4a73
2021-06-11 09:53:46.380 22172-22172/ D/Kable: setCharacteristicNotification(true) for LazyCharacteristic(serviceUuid=6e400001-b5a3-f393-e0a9-e50e24dcca9e, characteristicUuid=6e400003-b5a3-f393-e0a9-e50e24dcca9e)
2021-06-11 09:53:46.380 22172-22172/ D/BluetoothGatt: setCharacteristicNotification() - uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e enable: true
2021-06-11 09:53:46.383 22172-22317/ D/Kable: OnCharacteristicWrite(characteristic=6e400002-b5a3-f393-e0a9-e50e24dcca9e, status=GATT_SUCCESS(0))
2021-06-11 09:53:46.431 22172-22172/eD/Kable: Write to characteristic 6e400002-b5a3-f393-e0a9-e50e24dcca9e complete
2021-06-11 09:53:46.431 22172-22317/eD/Kable: Writing 0100 to descriptor 00002902-0000-1000-8000-00805f9b34fb
2021-06-11 09:53:46.438 22172-22190/ D/BluetoothGatt: onConnectionUpdated() - Device=D1:87:F9:50:23:87 interval=39 latency=0 timeout=500 status=0
2021-06-11 09:53:46.534 22172-22317/ D/Kable: OnCharacteristicChanged(characteristic=6e400003-b5a3-f393-e0a9-e50e24dcca9e, value=02702D574353454E534F52206465207072756562)
2021-06-11 09:53:46.534 22172-22317/ D/Kable: OnCharacteristicChanged(characteristic=6e400003-b5a3-f393-e0a9-e50e24dcca9e, value=612D323130353132303331363537323530303030)
2021-06-11 09:53:46.534 22172-22317/ D/Kable: OnDescriptorWrite(descriptor=00002902-0000-1000-8000-00805f9b34fb, status=GATT_SUCCESS(0))
2021-06-11 09:53:46.535 22172-22317/e D/Kable: OnCharacteristicChanged(characteristic=6e400003-b5a3-f393-e0a9-e50e24dcca9e, value=30303130323531303230333E323330313530303F)
2021-06-11 09:53:46.536 22172-22317/ D/Kable: OnCharacteristicChanged(characteristic=6e400003-b5a3-f393-e0a9-e50e24dcca9e, value=3903)
2021-06-11 09:53:46.546 22172-22172/ D/Kable: Write to descriptor 00002902-0000-1000-8000-00805f9b34fb complete
2021-06-11 09:53:46.546 22172-22172/ D/Kable: Wrote ENABLE_NOTIFICATION_VALUE to config PlatformDescriptor(serviceUuid=6e400001-b5a3-f393-e0a9-e50e24dcca9e, characteristicUuid=6e400003-b5a3-f393-e0a9-e50e24dcca9e, descriptorUuid=00002902-0000-1000-8000-00805f9b34fb)
2021-06-11 09:53:46.547 22172-22172/e D/Kable: PlatformCharacteristic(serviceUuid=6e400001-b5a3-f393-e0a9-e50e24dcca9e, characteristicUuid=6e400003-b5a3-f393-e0a9-e50e24dcca9e) does not support Indicate
[... cut ...]
2021-06-11 09:53:46.623 22172-22172/e D/AppViewModel$connect$1$2$invokeSuspend$$inlined$collect: addding to buffer 20 buffer length: 58
2021-06-11 09:53:46.623 22172-22172/e D/AppViewModel$connect$1$2$invokeSuspend$$inlined$collect: [Ble Data] Length: 20 String: 00102510203>2301500?
2021-06-11 09:53:46.624 22172-22172/e V/Kable: Propagating CharacteristicChange(serviceUuid=6e400001-b5a3-f393-e0a9-e50e24dcca9e, characteristicUuid=6e400003-b5a3-f393-e0a9-e50e24dcca9e, data=3903)
2021-06-11 09:53:46.624 22172-22172/e  D/AppViewModel$connect$1$2$invokeSuspend$$inlined$collect: addding to buffer 2 buffer length: 60

twyatt, 2021-06-11:
> The first one is the one dropping.

@Devinete in the first one, I only see a single `OnCharacteristicChanged` (provided via Android's callback) with a value of `38 03`. Then it makes it to your buffer and is logged as having a length of 2. So, it is looking like Kable properly propagated all the data it received from Android. Is it possible your peripheral didn't send the additional data you expected?

Devinete, 2021-06-14:
That would be very strange.
If i set an arbitaty delay between the collect of the write it will allways read the whole data.
Maybe I'm handling this incorrectly at coroutine level?

`    fun connect(advertisement: Advertisement) = viewModelScope.launch{
        disconnect()

        try {
            val peripheral = viewModelScope.peripheral(advertisement)
            device = deviceDevice(peripheral)
            device.connect()

            launch {
                device.state
                    .catch { throwable -> Timber.e(throwable,"device STATE CATCHED") }
                    .collect() { state ->
                    Timber.d("State " + state)
                    _state.emit(state)
                }
            }

            launch {
                device.observe() //this is just  peripheral.observe(readCharacteristic)
                    .catch { throwable -> Timber.e(throwable,"device OBSERVE CATCHED") }
                    .collect { data ->
                    when {
                         //do sometyhing with data...
                    }
                    Timber.d("[Ble Data] Length: ${data.size} String: ${String(data)}")
                }
            }
           
            
            device.readConfig() //this is just an initial msg encoded and peripheral.write(writeCharacteristic, msg)

        } catch (e: Exception) {
            Timber.w("connect catch")
            Timber.w(e)
        }

    }`

twyatt, 2021-06-14:
@Devinete on the `Flow` that you want to grab the response (via `observe`) perform your write operation in an `onSubscription` operator on the `Flow`, for example:

```kotlin
peripheral
    .observe(...)
    .onSubscription { peripheral.write(...) }
    .collect { // or `first`, or similar terminal operator.
        // Handle "response".
    }
```

Devinete, 2021-06-14:
Oh, so it seems i was doing it wrong then if it's supposed to work with the onSubscribe operator.

About the terminal collect, thanks for the ideas. I will definetly use it on other use cases. On this specific one, the "channel" is allways open because there are multiple data/response/types going on, it just happens to be writen by the client on the write descriptor and responses are available via de observe, the buffer is to decode responses because they are of multiple sizes/types based on app interaction.

About the onSubscription:
What i'm seeing is that maybe you are using internally a Shared or State flow but observe exposes a plain ByteArray flow with no onSubscription operator. (Maybe i'm expected to just cast, transform, shareIn, wathever to make a SharedFlow/StateFlow)

twyatt, 2021-06-14:
> About the onSubscription:
> What i'm seeing is that maybe you are using internally a Shared or State flow but observe exposes a plain ByteArray flow with no onSubscription operator. (Maybe i'm expected to just cast, transform, shareIn, wathever to make a SharedFlow/StateFlow)

Oh, I'll change the `observe` to return a `SharedFlow` and push another `SNAPSHOT` soon for you to try.

twyatt, 2021-06-14:
> Oh, I'll change the `observe` to return a `SharedFlow` and push another `SNAPSHOT` soon for you to try.

Apologies, I misspoke; realizing now that it would be better for you to use the `shareIn` operator rather than Kable expose `observe` as a `SharedFlow` (having consumer convert it to `SharedFlow` allows you to decide the `Flow`'s lifecycle).

So, could you try something like:

```kotlin
val observation = peripheral.observe(...).shareIn(scope, started = WhileSubscribed())
observation
    .onSubscription {
        peripheral.write(...) // i.e. `device.readConfig()`
    }
    .collect { // or `first`, or similar terminal operator.
        // Handle "response".
    }
```

Assuming this is for write-for-response style I/O, you could encapsulate it nicely as (for example):

```kotlin
val observation = peripheral.observe(...).shareIn(scope, started = WhileSubscribed())
val expectedBufferLength = 2 // bytes

suspend fun writeForResponse(request: ByteArray): ByteArray {
    val buffer = ByteArray(expectedBufferLength)
    observation
        .onSubscription {
            // i.e. `device.readConfig()`
            peripheral.write(..., request)
        }
        .transformWhile { bytes ->
            if (position >= buffer.size) {
                emit(buffer)
                false // Terminate `Flow`.
            } else {
                position += bytes.size
                // Copy `bytes` into `buffer`.
                true // Continue collecting bytes into buffer.
            }
        }
        .first()
}
```

Devinete, 2021-06-15:
I've changed to a shared flow and put the peripheral initial write onSubscription like your first example. It's still missing pakets on a Nokia and a Xiaomi.

I'm not quite sure but could be that the onSubscription write is not actually waiting for the notify observer to spun up?. If i just do the basic collect, delay(insertArbitraryValue), and do the initial write, there is no missing data.

2021-06-15 10:30:09.553 5698-5766/D/Kable: Writing 0270383E03 to characteristic 6e400002-b5a3-f393-e0a9-e50e24dcca9e
2021-06-15 10:30:09.558 5698-5766/D/Kable: OnCharacteristicWrite(characteristic=6e400002-b5a3-f393-e0a9-e50e24dcca9e, status=GATT_SUCCESS(0))
2021-06-15 10:30:09.609 5698-5722/D/BluetoothGatt: onConnectionUpdated() - Device=D1:87:F9:50:23:87 interval=39 latency=0 timeout=500 status=0
2021-06-15 10:30:09.617 5698-5698/D/Kable: setCharacteristicNotification(true) for LazyCharacteristic(serviceUuid=6e400001-b5a3-f393-e0a9-e50e24dcca9e, characteristicUuid=6e400003-b5a3-f393-e0a9-e50e24dcca9e)
2021-06-15 10:30:09.617 5698-5698/D/BluetoothGatt: setCharacteristicNotification() - uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e enable: true
2021-06-15 10:30:09.619 5698-5698/D/Kable: Write to characteristic 6e400002-b5a3-f393-e0a9-e50e24dcca9e complete
2021-06-15 10:30:09.620 5698-5766/D/Kable: Writing 0100 to descriptor 00002902-0000-1000-8000-00805f9b34fb
2021-06-15 10:30:09.759 5698-5766/D/Kable: OnCharacteristicChanged(characteristic=6e400003-b5a3-f393-e0a9-e50e24dcca9e, value=3703)
2021-06-15 10:30:09.760 5698-5766/D/Kable: OnDescriptorWrite(descriptor=00002902-0000-1000-8000-00805f9b34fb, status=GATT_SUCCESS(0))
2021-06-15 10:30:09.761 5698-5698/D/Kable: Write to descriptor 00002902-0000-1000-8000-00805f9b34fb complete
2021-06-15 10:30:09.762 5698-5698/D/Kable: Wrote ENABLE_NOTIFICATION_VALUE to config PlatformDescriptor(serviceUuid=6e400001-b5a3-f393-e0a9-e50e24dcca9e, characteristicUuid=6e400003-b5a3-f393-e0a9-e50e24dcca9e, descriptorUuid=00002902-0000-1000-
2021-06-15 10:30:09.762 5698-5698/D/Kable: PlatformCharacteristic(serviceUuid=6e400001-b5a3-f393-e0a9-e50e24dcca9e, characteristicUuid=6e400003-b5a3-f393-e0a9-e50e24dcca9e) does not support Indicate
2021-06-15 10:30:09.763 5698-5698/V/Kable: Propagating CharacteristicChange(serviceUuid=6e400001-b5a3-f393-e0a9-e50e24dcca9e, characteristicUuid=6e400003-b5a3-f393-e0a9-e50e24dcca9e, data=3703)
2021-06-15 10:30:09.843 5698-5698/D/appViewModel$connect$1$2$invokeSuspend$$inlined$collect: addding to buffer 2 buffer length: 2
2021-06-15 10:30:09.844 5698-5698/D/appViewModel$connect$1$2$invokeSuspend$$inlined$collect: [Ble Data] Length: 2 String: 7

twyatt, 2021-06-15:
I have a suspicion that `setCharacteristicNotification(true)` is occurring too late for you. I'll look into if there is a way I can get that operation to execute sooner when spinning up observations.

twyatt, 2021-06-16:
@Devinete, I investigated timings of `onSubscription` calls and it does seem to explain why your write (`device.readConfig()`) is happening before the observation is wired up; illustrated with the following simple test:

```kotlin
suspend fun main() = coroutineScope<Unit> {
    val upstream = MutableSharedFlow<Int>()
    val sf = upstream.onSubscription { println("upstream onSubscription") }

    launch {
        repeat(100) {
            delay(1_000)
            upstream.emit(it)
        }
    }

    val downstream = sf
        .shareIn(this, started = Lazily)
        .onSubscription { println("downstream onSubscription") }
    downstream.onEach { println("← $it") }.launchIn(this)
}
```

**Output**

```
downstream onSubscription
upstream onSubscription
← 0
← 1
...
```

In the above test, "upstream" is emulating Kable wiring up the observation and "downstream" emulates your attempt at invoking `readConfig()`. Clearly the `readConfig()` is occurring _before_ the observation is wired up (and ready) — which is not what you want.

I can't think of an elegant way to get the desired ordering using standard `Flow` operators, so (instead) I've added an _optional_ action lambda to `observe` function (this lambda is invoked right _after_ the observation is wired up), for example:

```kotlin
device.observe(characteristic) {
    // todo: Perform any operations you want to occur immediately after the observe is active, for example:
    device.readConfig()
}
```

If you could give it a try and let me know if it works for you: `0.5.1-issue-112-2-SNAPSHOT`.

_You shouldn't need to do the `shareIn` approach I previously mentioned with this change._

Devinete, 2021-06-16:
Tested multiple times on all my test devices. It works.

twyatt, 2021-06-16:
> Tested multiple times on all my test devices. It works.

Thanks for reporting back. I'll work on updating the PR with some additional documentation then get it merged. Only a few [quick items I want to get into 0.6.0](https://github.com/JuulLabs/kable/milestone/9), so I'm hoping by end-of-week I'll be able to push out that release.

twyatt, 2021-06-18:
> I'm hoping by end-of-week I'll be able to push out that release

Ran into some additional complexities around invoking the `observe` lambda (`onObservationStarted`) upon reconnection or when `observe` is configured _before_ connection is established (i.e. when re-wiring observations). I have a plan for implementation but would like to add unit tests to validate — so, integration of the fix may continue through this weekend / next week.

Devinete, 2021-06-18:
I'm ready to test wathever you throw at me.
Thanks!

twyatt, 2021-06-21:
@Devinete would be great to get some feedback on potential design choices I'm facing:

With `observe`, Kable ensures that the notification or indication is only enabled once per characteristic, so someone calling `observe` twice with the same characteristic will get two separate `Flow`s. Subscribing to both will make sure that only one of them will trigger the enabling of notifications/indications.

This brings us to `onObservationStarted`, it will be triggered upon subscription to the `Flow` returned by `observe` (with the special timing of occurring _after_ the enabling of notifications/indications — yet capturing any characteristic changes during that process) but should it trigger for every subscription to `observe`, or only if the notification/indication was enabled (i.e. when transitioning from `0` to `1` subscribers for a specific characteristic)?

Illustrating via pseudo code:

```kotlin
peripheral.observe(characteristicA) {
    onObservationStarted {
        peripheral.write(characteristicA, "hello".toByteArray())
    }
}.onEach { ... }.launchIn(scope) // causes subscription count to transition from 0 to 1 and `onObservationStarted` is invoked

peripheral.observe(characteristicA) {
    onObservationStarted {
        peripheral.write(characteristicA, "bye".toByteArray())
    }
}.onEach { ... }.launchIn(scope) // should `onObservationStarted` be invoked for this subscription (we've already setup the notifications/indications — so Flow's `onSubscription` operator would've sufficed)?
```

Perhaps (to simplify the internals), Kable could restrict to only a single `onObservationStarted` per characteristic (throwing an exception if additional observations with `onObservationStarted` are attempted) — but that makes the API less friendly.

Additionally, I presume the `onObservationStarted` should be invoked again when re-wiring the observation (i.e. upon reconnection)?

twyatt, 2021-06-21:
@Devinete you can probably disregard my previous comment, thinking about it more, I've settled on the following behavior (copied from `observe`'s KDoc on a local branch I'm working on):

```
The optional [onSubscription] parameter is functionally identical to using the
[onSubscription][kotlinx.coroutines.flow.onSubscription] operator on the returned [Flow] except it has the
following special properties:

- It will be executed whenever [connection][connect] is re-established (while the returned [Flow] is active); and
- It will be executed _after_ the observation is spun up (i.e. after enabling notifications or indications)
```

I'll be renaming `onObservationStarted` to `onSubscription`, as its behavior will be similar to using the standard Flow `onSubscription` operator.

Let me know if you see any potential issues with this design/behavior. I'll try to get a `SNAPSHOT` with these changes up soon.

twyatt, 2021-06-22:
@Devinete when you have a chance, it'd be must appreciated if you gave `0.5.1-issue-112-3-SNAPSHOT` a spin.

Notable changes since `0.5.1-issue-112-2-SNAPSHOT`:

- `observe`'s `onObservationStarted` parameter renamed to `onSubscription`
- `onSubscription` lambda is invoked upon connection being established (see `observe` KDoc for more details)

These changes mean that the `onSubscription` should execute (at the appropriate time) whether or not the `observe` is setup before or after the connection is made.

Being that it behaves similar to the standard [`onSubscription`](https://kotlin.github.io/kotlinx.coroutines/kotlinx-coroutines-core/kotlinx.coroutines.flow/on-subscription.html) operator, it will be executed for every new Flow collector, for example:

```kotlin
val observation = peripheral.observe(characteristic) {
    println("hello")
}
observation.onEach(::println).launchIn(scope)
observation.onEach(::println).launchIn(scope)
```

In the above example, since 2 collectors are started, `"hello"` will be printed twice. If only a single invocation of `onSubscription` is desired, then you can either use [`shareIn`](https://kotlin.github.io/kotlinx.coroutines/kotlinx-coroutines-core/kotlinx.coroutines.flow/share-in.html):

```kotlin
val shared = observation.shareIn(scope, started = WhileSubscribed())
shared.onEach(::println).launchIn(scope)
shared.onEach(::println).launchIn(scope)
```

...or call `observe` once with the `onSubscription` lambda and once without:

```kotlin
val observation1 = peripheral.observe(characteristic) {
    println("hello")
}
val observation2 = peripheral.observe(characteristic)
observation1.onEach(::println).launchIn(scope)
observation2.onEach(::println).launchIn(scope)
```

Devinete, 2021-06-22:
I can't seem to trigger on my use case any change between 112-2 and 112-3. It works for me.

twyatt, 2021-06-22:
> I can't seem to trigger on my use case any change between 112-2 and 112-3. It works for me.

Great! That's what I want to hear. 😄  Thanks for confirming it still works.

PR #110 (2021-06-23): Propagate early characteristic changes
Closes #105
Closes #112

Also fixes JavaScript observations to wait until connection is ready. Only users of experimental advertisement support in Chrome would've seen this bug, as using `requestDevice` obtains devices in a connected state, i.e. already ready.
closed 2021-06-23 by twyatt