JuulLabs/kable #118: notification observe returning less data than characteristic read

nathan-castlehow, 2021-06-21:
Hi,

Apologies if this is the wrong place. Feels like it might be an issue with the library but not 100% sure.

The issue I am having feels similar to https://github.com/JuulLabs/kable/issues/112

We have a characteristic we start to observe when connected to our device.
It eventually gets written to, however when observing the flow it only returns 20 bytes (this feels oddly specific).
If we read straight from the characteristic it gets the full value of 74 bytes (We have a delay which lets our device write to the characteristic after we have asked it to). 

When poking at logs the only interesting thing i could see was below where it changes the notification settings.
 D/BluetoothGatt: setCharacteristicNotification() - uuid: a7d06ce0-2e84-485f-9c25-3d4ba6fe7319 enable: false

Do you have any insights on this one?

Thanks a ton!

twyatt, 2021-06-21:
Hi @nathan-castlehow, can you try the latest snapshot in the issue you mentioned (`0.5.1-issue-112-2-SNAPSHOT`) and use the `onObservationStarted` option in the `observe` lambda? And report back if it works as expected, if not, then logs produced from that snapshot would be helpful.

Thanks!

nathan-castlehow, 2021-06-22:
Hmm so the example you have is writing to the same characteristic we are then observing. We only read from this characteristic. The observing gets setup as soon as we connect to the device and then later on it gets written to.

I added the snapshot but doesn't appear to log anything as we are interested in the reading (and the logs from the other issue appear to be for the writing).

[image]

Let me know if I'm missing some obvious step to help trouble shoot this.

Thanks again!

twyatt, 2021-06-22:
@nathan-castlehow can you make sure that under "External Libraries" it is showing the `SNAPSHOT` version, for example:

[image]

nathan-castlehow, 2021-06-22:
[image]

Hmmm seems to match.

The only thing interesting seems to be the fact that "setCharacteristicNotification" is being called with false
[image]

Not sure if this adds any value but if I both read from and observe the characteristic I get a not ready exception.

twyatt, 2021-06-22:
Bizarre that you're not getting logging from Kable, it is using the standard Android `Log.*` methods. You're able to log via Android's `Log.*` from your app?

nathan-castlehow, 2021-06-28:
Haven't had any issues with logging.

I was chatting to a colleague and he suggested that this might actually be the desired / default behavior.
The size of the value we want is 74 bytes which is larger than our MTU (we can't increase the MTU due to a bug on some samsung devices with BLE). He suggested that notifications can only return values up to the MTU size and for larger observing the notification and then reading the characteristic is the default behaviour.

Does that make sense to you?

twyatt, 2021-06-29:
> we can't increase the MTU due to a bug on some samsung devices with BLE

I personally haven't encountered issues with increasing the MTU, but BLE on some phones is super wonky, so it doesn't surprise me that there are limitations on some Samsung phones. It may be worth trying to figure out what specific Samsung models suffer from the MTU limits? Perhaps (and perhaps just wishful thinking) that the offending models are pre-Android 21, or their prevalence is so low it may not be worth supporting them?

> He suggested that notifications can only return values up to the MTU size and for larger observing the notification and then reading the characteristic is the default behaviour.

Seems like a reasonable workaround (if needed), you could fairly easily write something in Kable to implement that workaround, for example:

```kotlin
peripheral
    .observe(characteristic)
    .map { peripheral.read(characteristic) }
    .collect { data -> /* todo: process `data` */ }
```

nathan-castlehow, 2021-06-30:
I ended up with something pretty close to what you suggested above.
We can probably close this one as "by design"
I wish we could drop support but some specific customers were using the phones in question. One can dream ;)

twyatt, 2021-06-30:
> I ended up with something pretty close to what you suggested above.
> We can probably close this one as "by design"

Sounds good. 👍 

> I wish we could drop support but some specific customers were using the phones in question. One can dream ;)

Haha, yeah, I know how that goes. 😉 

> Thanks again for all your help and for maintaining such a great library! Will try give back when I can.

Thanks for the kind words. 😄 
Feel free to reach out if you have any other questions or issues.

closed 2021-06-30 by twyatt