JuulLabs/kable #119: JVM shows NoSuchErrorMessage for overloaded Expect functions

burnhamd, 2021-06-22:
After building latest branch I have started getting the following error:

`    java.lang.NoSuchMethodError: No static method peripheral$default(Lkotlinx/coroutines/CoroutineScope;Lcom/juul/kable/Advertisement;Lkotlin/jvm/functions/Function1;ILjava/lang/Object;)Lcom/juul/kable/Peripheral; in class Lcom/juul/kable/PeripheralKt; or its super classes (declaration of 'com.juul.kable.PeripheralKt' appears in /data/app/~~3Aev88ZqQF1v99H8zS59Yw==/com.example.kabletest.android-8adKCQ6PYRgbM-h0HQ6Lgg==/base.apk)`

At first I thought it was due to my own changes, but after reverting back to main branch the issue remains.
Has anyone else seen this?

twyatt, 2021-06-22:
Is this in a project you're able to share?

burnhamd, 2021-06-22:
Yes. I’ll remove some things and then provide a link.

burnhamd, 2021-06-23:
After a fresh look, Ive realized it was a simple issue.  My android project was not flushing the cached version of its dependency, but the common library was.  So when it came to running the app, the android common library was trying to call methods that didn't exist in the 0.5.1 version.

Deleting android build folder and running grade sync fixed the issue.

closed 2021-06-23 by burnhamd

twyatt, 2021-06-23:
Great to hear you got it working. Thanks so much for sharing what you did to fix it, so if others encounter the same issue they have some steps they can try. 😄