JuulLabs/kable #127: Characteristics with duplicating UUIDs are not supported

solvek, 2021-06-26:
I have a device which have two different characteristics with the same UUID:
[image]

One of them should be used for writing data and another for notifications.

Seems kable uses standard android method for getting characteristic and it returns first one by UUID:
`service.getCharacteristic(BLE_DATA_STREAM_CHARACTERISTIC_DATA_AVAILABILITY)`

For my device I wrote a pure app without libraries and I loop the service characteristic and choose proper one by checking its properties:
`    private fun BluetoothGattService.getCharaWithProperty(uuid: UUID, property: Int) =
        characteristics.first { it.uuid == uuid && it.properties and property != 0 }`

This pure app works fine but if I use kable my device is not working properly.

Is there any way to choose service/characteristic from discovered but not just by UUIDs?

twyatt, 2021-06-26:
Yes, it is planned to be fixed in the future. 👍 
Duplicate of #93.

solvek, 2021-06-27:
Thanks for your great library and quick response.
I am not absolutely sure this is the same issue because #93 is about duplicated services while in my case there are characteristic duplicated.
I have an idea how to fix the issue.
I looked in two code and seems like kable just searches the first characteristic by uuid.
What if we search for the first by uuid and having proper property? For exame If we are doing `write` then the characteristic should have `WRITE` propery, if we are doing `observe` then characteristic should have `NOTIFY` or `INDICATE`, not only matching uuid.

twyatt, 2021-06-27:
Thanks for the clarification, I see now how the issues may differ. 👍

Clever idea on how to differentiate the characteristics by their supported properties.

I need to look into it further, but I think the `instanceId` that Android provides would be a reliable differentiator in the case where a library consumer iterates over discovered services/characteristics to find the characteristics of interest.

In either case, do need to look into it further, but your suggestion will help in the brainstorming process.

twyatt, 2022-01-06:
@solvek when you have a chance, can you try the following version and see if it addresses this issue?:

```
repositories {
    maven("https://oss.sonatype.org/content/repositories/snapshots")
}

dependencies {
    implementation("com.juul.kable:core:0.10.4-issue-93-1-SNAPSHOT")
}
```

See #238 for more details.

solvek, 2022-01-10:
@twyatt 
Thanks for the fix and sorry for delay.
Seems it works for me.

Thanks a lot.

twyatt, 2022-01-10:
@solvek thanks so much for all your help!
I'll try to get a version with this update released soon (expect it in the [0.12.0 release](https://github.com/JuulLabs/kable/milestone/21)).

closed 2022-02-04 by twyatt

twyatt, 2022-02-04:
@solvek I'll put together a write up for users re: how characteristic and descriptor lookups have changed to have in the 0.13.0 release notes. I expect to cut the release in the coming days. Sorry it took so long to get this in, thanks for your patience!

twyatt, 2022-02-05:
Support shipped in [0.13.0](https://github.com/JuulLabs/kable/releases/tag/0.13.0).