JuulLabs/kable #132: ACCESS_FINE_LOCATION permission requirement

Jahor, 2021-07-05:
Is it possible to remove 
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
from AndroidManifest.xml ?

In my app I use Companion Device Setup to avoid requesting location permission from user. But now I've noticed that my app still has that permission in the resulting manifest.

My understanding that scanning for BLE devices requires ACCESS_COARSE_LOCATION, but if I don't use scanning I can just  have BLUETOOTH and BLUETOOTH_ADMIN.

Is there any reason to force library users to request ACCESS_FINE_LOCATION?

twyatt, 2021-07-05:
I suspect that a majority of Kable library consumers would prefer having Kable specify the permissions it may use (i.e. less setup), but for your use case, it does make sense that if you aren't scanning that you don't want the `ACCESS_*_LOCATION` permission, I believe you can add the following to your apps' manifest?:

```xml
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" tools:node="remove"/>
```

— [Remove extra unwanted permissions from manifest android](https://stackoverflow.com/a/33105591)

Jahor, 2021-07-05:
Thanks, that trick works. I still think you should at ask for coarse location if fine is not required.
I was just very surprised that library can silently require permissions in the first place.

twyatt, 2021-07-05:
> I still think you should at ask for coarse location if fine is not required.

Agreed, I'll look into if `ACCESS_COURSE_LOCATION` would be sufficient. Thanks!

mostthingsweb, 2021-07-06:
> Agreed, I'll look into if `ACCESS_COURSE_LOCATION` would be sufficient. Thanks for pointing that out!

Here's some info that may be helpful for Bluetooth permissions.

From https://developer.android.com/guide/topics/connectivity/bluetooth/permissions: 

> Note: If your app targets Android 9 (API level 28) or lower, you can declare the ACCESS_COARSE_LOCATION permission instead of the ACCESS_FINE_LOCATION permission.

In other words (from https://developer.android.com/reference/android/bluetooth/le/BluetoothLeScanner#startScan(java.util.List%3Candroid.bluetooth.le.ScanFilter%3E,%20android.bluetooth.le.ScanSettings,%20android.app.PendingIntent)):

> An app must have ACCESS_COARSE_LOCATION permission in order to get results. An App targeting Android Q or later must have ACCESS_FINE_LOCATION permission in order to get results.

However, Android 12 (recently in Beta) changes up the permissions altogether: https://developer.android.com/about/versions/12/features/bluetooth-permissions

BluestormDNA, 2021-07-14:
Continuing this conversation as Kable is moving towards handling "all the things" (wich is awesome btw) and seeing #137, i'm gonna leave a question here:
_There should be another trow if Location is off?_

You basically can't scan without the GPS ON (havent't checked to connect to an already bonded device).

If yes, the rabbit hole goes a bit deeper, because on API levels below 28 users could actually disable the GPS but retain Location Services ON.

Consider this pictures:
[image]

In this picture Location Services, labeled as GPS are ON but the actual GPS behind the scenes is OFF so BLE scan will silently fail.

Furthermore and just because i recently lose some time on this:
Latest AndroidX Core 1.6.0 `LocationManagerCompat.isLocationEnabled()` is implemented on those api levels as GPS || Network as seen on:
https://github.com/androidx/androidx/blob/c5f6a749b139b898b5be5376b257868347b4f223/core/core/src/main/java/androidx/core/location/LocationManagerCompat.java#L120-L121

So in the end you need to end with something like:

```
       //But is really the GPS on?
        return if(Build.VERSION.SDK_INT > 28) {
            locationManager.isLocationEnabled
        } else {
            locationManager.isProviderEnabled(LocationManager.GPS_PROVIDER)
        }
```

Keep the good work, awesome library.

twyatt, 2021-07-14:
@BluestormDNA we do plan to add checks to confirm that permissions and location services are in a state that is usable by Kable (and throw if not), so this is super helpful information for when we tackle that. Thank you!

mostthingsweb, 2021-07-16:
> @BluestormDNA we do plan to add checks to confirm that permissions and location services are in a state that is usable by Kable (and throw if not), so this is super helpful information for when we tackle that. Thank you!

I don't know if it helps at all, but I took a slightly different approach in the app I'm working on based on kable and SensorTag: https://github.com/mostthingsweb/esp_provisioning_coroutines/blob/main/lib/src/main/java/io/laplante/epc/EpcService.kt#L77.  I expanded the ScanStatus Failed to include particular failure reasons, e.g. location permissions not granted, Bluetooth off. 

By using a flow instead of exceptions it is very easy to observe inside an Activity and then perform the proper actions in response: https://github.com/mostthingsweb/esp_provisioning_coroutines/blob/main/app/src/main/java/io/laplante/epc_demo/MainActivity.kt#L102

twyatt, 2021-07-19:
Thanks for sharing the code links @mostthingsweb, it is always helpful to see how others approach problems and also use Kable in their projects, helps to drive future API design.

I'm pretty busy on some other projects for a bit, but as soon as I find some more free time, I'll dive back into this issue.

Thanks again!

PR #866 (2025-04-02): Provide default Android permissions via `kable-default-permissions` artifact
Closes #865
Closes #132

Removed Kable permission declarations and makes it the responsibility of the consumer to setup appropriate permissions. `kable-default-permissions` artifact provides sane defaults (same as previously) and can optionally be used by consumers.

Rendered markdown of the relevant updated documentation can be found [here](https://github.com/JuulLabs/kable/tree/twyatt/android-default-permissions?tab=readme-ov-file#android-permissions).
closed 2025-04-02 by twyatt