JuulLabs/kable #143: Consider exposing uncached ble device name

BluestormDNA, 2021-07-14:
Hello there!

Not sure if related to #59 so i opened a new issue, feel free to close it if it actually is.

Because Ble is so weird, at least on Android (and probably on iOS too) the name returned by the peripheral is the name the BLE Device had when it connected the first time. Then is cached internally (and is reseted on certain conditions...)

This is usually not a problem, but on some scenarios BLE Devices change names to signal things and you actually need the name the device is actually advertising and not the one you connected to originally. On Android that uncached name is on `scanRecord?.deviceName?` and on iOS is on `kCBAdvDataLocalName` but at least on android that field on Advertisement is private.

Even thought i can get them with reflection it could be nice to maybe have something like an `advertisedName` field or actually the scanRecord data exposed.

twyatt, 2021-07-19:
On Android, is [`scanRecord.deviceName`](https://developer.android.com/reference/android/bluetooth/le/ScanRecord#getDeviceName()) more accurate than [`BluetoothDevice.name`](https://developer.android.com/reference/android/bluetooth/BluetoothDevice#getName())? Is that what you're saying?

If so, then perhaps it would make sense to have the following:

https://github.com/JuulLabs/kable/blob/f34f54e0a9125a637f6ea4615b4028ccf0a1f3b8/core/src/androidMain/kotlin/Advertisement.kt#L24-L25

...instead be written as?:

```kotlin
public actual val name: String?
    get() = scanRecord?deviceName
```

Or, are there use-cases for using one of the names vs. the other?

Devinete, 2021-07-19:
> On Android, is scanRecord.deviceName more accurate than BluetoothDevice.name? Is that what you're saying?

For scanning purpose it actually is. It haves the current emitted deviceName wich may be the same OR NOT of what you have cached on BluetoothDevice.name

This is some niche use case and relies on ble devices that change names. (My use case itself is a device that acts as a bridge between a server and a client to send data and it changes names dinamically, conserving the same mac, to expose to clients or server...)

Consider a BLE device that changes its name every 24 hours and it starts as "Monday Device" then "Tuesday Device" and so on...
If you happen to connect to it the first time on Wednesday. The bluetoothDevice.name will be "Wednesday Device".
If you come back on Sunday, assuming here the ble hasn't invalidated the cache on reboot or magic, bluetoothDevice.name will still be "Wednesday Device" and scanRecord?deviceName will actually have "Sunday Device".

I can foresee some use cases where some people may be expeting allways the uncached name and others the cached name. 

iOS seems to have the same quirks.
https://developer.apple.com/forums/thread/72343

PR #403 (2022-12-27): Retrieve `Advertisement.name` from advertisement data
Adjusts the value returned by `Advertisement.name` that was recently changed (unreleased) via #401.

As @BluestormDNA kindly pointed out in https://github.com/JuulLabs/kable/pull/401#issuecomment-1319638377, `Advertisement.name` was not sourcing the name from the most up-to-date sources (and that Kable should provide the uncached value, as well as the previously provided cached value).

The `Advertisement.name` now pulls the name from advertisement data. Previously it used the value provided by the platform peripheral object (but in many ca [... cut]
closed 2022-12-27 by twyatt