JuulLabs/kable #147: Observe flow not observing all values.

Alexminator99, 2021-07-23:
I have this piece of code:

     viewModelScope.launch {
                val advertisement = Scanner()
                    .advertisements
                    .catch { cause -> Log.d(TAG, "scanPeripherals: $cause") }
                    .filter { it.name?.equals("myDdeviceName") == true }
                    .first()

                val peripheral = peripheral(advertisement) {
                    onServicesDiscovered {
                        Log.d(TAG, "services discovered")
                    }

                }

                val charToObserve = characteristicOf(
                    service = Services.service,
                    characteristic = Characteristcs.readerSomething
                )
                val charToObserve2 = characteristicOf(
                    service = Services.eTectService,
                    characteristic = Characteristcs.readerSomethingElse
                )

                try {
                    peripheral.connect()

                    peripheral.observe(charToObserve).catch {
                        Log.d(TAG, "error detected: $it")
                    }.onEach { characteristic ->
                        // Process data.
                        Log.d(TAG, "char detected: ${characteristic.contentToString()}")
                    }.launchIn(this)

                    peripheral.observe(charToObserve2).catch {
                        Log.d(TAG, "error detected: $it")
                    }.onEach { characteristic ->
                        // Process data.
                        Log.d(TAG, "char2 detected: ${characteristic.contentToString()}")
                    }.launchIn(this)
                } catch (e: IOException) {
                    Log.d(TAG, "scanPeripherals: $e", e)
                }
             }...   

My **problem** is the next: every time the device is **connected/disconnected** it sends a **signal** in form of characteristic, so it supports notifications. The problem is that if I do that **10 times** quickly the best i got is **8 times** the log print... **the worse is only 1 time**. 

Im doing something wrong maybe? Or in a rare case is a lib problem (which I dont think it is)?

twyatt, 2021-07-23:
Are you saying that your remote peripheral send to the phone a "characteristic change" immediately upon the phone connecting to it? And that it appears that Kable is missing that really early "characteristic change"?

Alexminator99, 2021-07-23:
I'm going to perform some more testing and be back with more info to share. Thks

twyatt, 2021-08-28:
If you're still experiencing this issue please re-open and use the new [logging feature](https://github.com/JuulLabs/kable#logging) to provide logs.

closed 2021-08-28 by twyatt