JuulLabs/kable #156: Bluetooth device name on iOS is not updated

ln-12, 2021-08-31:
I am developing a bluetooth device which my app should connect to. During testing, I changed the name of my bluetooth device. On Android, it works absolutely fine and as expected. However, as iOS always does things differently, it does not work.  In fact, iOS caches bluetooth names according to this [SO answer](https://stackoverflow.com/a/47129088/13440564). The author mentions a different approach  of getting the real (updated) name which could be used [here](https://github.com/JuulLabs/kable/blob/main/core/src/appleMain/kotlin/Advertisement.kt#L23). What do you think?

twyatt, 2021-08-31:
I think this is either a duplicate of, or closely related to, #143.

In either case, yes, we need to make some changes there to make the uncached name available.
We've been swamped with some deadlines lately, but I hope to be able to put some more time to Kable in the coming weeks.

ln-12, 2021-08-31:
Alright, I'll close this issue in favor of #143. Thanks!

closed 2021-08-31 by ln-12