JuulLabs/kable #165: (iOS) Peripheral.state not changing on turning off Bluetooth

lammertw, 2021-09-28:
I'm presenting a device connection state to the user. For that I'm observing `Peripheral.state`. But when I run my app with a device connected and then turn off Bluetooth on my iPhone, the `Peripheral.state` does not change and stays `Connected`. When I turn off my Bluetooth device it does change to `Disconnected` so it would appear I'm not doing anything wrong.

twyatt, 2021-09-28:
Thanks for reporting the issue.

We probably need to do something similar to what was done for Android in #133 (where we monitor BLE state and explicitly update the state when we detect that BLE has been turned off).

burnhamd, 2022-03-18:
Looking at this it seems like the best thing is to just add something like this to the constructor of the peripheral
`        centralManager.delegate.state.onEach { event ->
            if(event == CBManagerStatePoweredOff){
                _state.value = State.Disconnected()
            }
        }.launchIn(scope)`

What Im not are about is how to properly handle state resetting. My thought is that this should also be considered disconnected. If anyone else knows if this is appropriate please let me know.

burnhamd, 2022-03-24:
I have this pretty much ready to go.  Will create PR once #280 is merged in since it is branched from that

closed 2022-03-29 by twyatt