JuulLabs/kable #189: peripheral.observe only emits once

AyoTheDev, 2021-11-15:
Hello, I've connected my app to a heart rate monitor device successfully, however when calling observation. collect it only collects once and then nothing after.

My expectation is that I should be getting updating data via this flow for heart rate, any help with this, please?

twyatt, 2021-11-15:
Can you provide logs with Kable logging enabled, as well as the snippet of code you're using to `observe`, please?

AyoTheDev, 2021-11-15:
Thanks for speedy response

2021-11-15 17:25:17.299 15112-15112/com.withutraining.tv.hrm I/Surface: opservice is null false
2021-11-15 17:25:17.345 15112-15112/com.withutraining.tv.hrm D/BluetoothAdapter: isLeEnabled(): ON
2021-11-15 17:25:17.351 15112-16210/com.withutraining.tv.hrm D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=11 mScannerId=0
2021-11-15 17:25:20.220 15112-16210/com.withutraining.tv.hrm D/BluetoothDevice: getService, sService=null
2021-11-15 17:25:20.221 15112-16210/com.withutraining.tv.hrm W/BluetoothAdapter: getBluetoothService(), client: android.bluetooth.BluetoothDevice$1@dd76d19
2021-11-15 17:25:20.221 15112-16210/com.withutraining.tv.hrm D/BluetoothDevice: getService, adapter sService=android.bluetooth.IBluetooth$Stub$Proxy@24d41de
2021-11-15 17:25:20.290 15112-16356/com.withutraining.tv.hrm D/BluetoothGatt: connect() - device: E7:BB:EB:BD:A0:A2, auto: false
2021-11-15 17:25:20.290 15112-16356/com.withutraining.tv.hrm D/BluetoothGatt: registerApp()
2021-11-15 17:25:20.291 15112-16356/com.withutraining.tv.hrm D/BluetoothGatt: registerApp() - UUID=654de77d-4a7f-4d39-a270-a5d52792f159
2021-11-15 17:25:20.294 15112-16210/com.withutraining.tv.hrm D/BluetoothGatt: onClientRegistered() - status=0 clientIf=12
2021-11-15 17:25:21.216 15112-16210/com.withutraining.tv.hrm D/BluetoothDevice: getService, sService=android.bluetooth.IBluetooth$Stub$Proxy@24d41de
2021-11-15 17:25:21.217 15112-16210/com.withutraining.tv.hrm D/BluetoothDevice: getService, adapter sService=android.bluetooth.IBluetooth$Stub$Proxy@24d41de
2021-11-15 17:25:21.245 15112-16210/com.withutraining.tv.hrm D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=12 device=E7:BB:EB:BD:A0:A2
2021-11-15 17:25:21.250 15112-16365/com.withutraining.tv.hrm D/BluetoothGatt: discoverServices() - device: E7:BB:EB:BD:A0:A2
2021-11-15 17:25:21.265 15112-16210/com.withutraining.tv.hrm D/BluetoothGatt: onSearchComplete() = Device=E7:BB:EB:BD:A0:A2 Status=0
2021-11-15 17:25:21.285 15112-16356/com.withutraining.tv.hrm D/BluetoothGatt: setCharacteristicNotification() - uuid: 00002a37-0000-1000-8000-00805f9b34fb enable: true
2021-11-15 17:25:21.932 15112-16356/com.withutraining.tv.hrm I/System.out: got new heart rate data
2021-11-15 17:25:22.542 15112-16210/com.withutraining.tv.hrm D/BluetoothGatt: onConnectionUpdated() - Device=E7:BB:EB:BD:A0:A2 interval=400 latency=0 timeout=500 status=0
2021-11-15 17:25:24.326 15112-16210/com.withutraining.tv.hrm D/BluetoothDevice: getService, sService=android.bluetooth.IBluetooth$Stub$Proxy@24d41de
2021-11-15 17:25:24.327 15112-16210/com.withutraining.tv.hrm D/BluetoothDevice: getService, adapter sService=android.bluetooth.IBluetooth$Stub$Proxy@24d41de
2021-11-15 17:25:25.366 15112-16210/com.withutraining.tv.hrm D/BluetoothDevice: getService, sService=android.bluetooth.IBluetooth$Stub$Proxy@24d41de
2021-11-15 17:25:25.366 15112-16210/com.withutraining.tv.hrm D/BluetoothDevice: getService, adapter sService=android.bluetooth.IBluetooth$Stub$Proxy@24d41de
2021-11-15 17:25:30.503 15112-16210/com.withutraining.tv.hrm D/BluetoothDevice: getService, sService=android.bluetooth.IBluetooth$Stub$Proxy@24d41de
2021-11-15 17:25:30.503 15112-16210/com.withutraining.tv.hrm D/BluetoothDevice: getService, adapter sService=android.bluetooth.IBluetooth$Stub$Proxy@24d41de
2021-11-15 17:25:35.591 15112-16210/com.withutraining.tv.hrm D/BluetoothDevice: getService, sService=android.bluetooth.IBluetooth$Stub$Proxy@24d41de
2021-11-15 17:25:35.592 15112-16210/com.withutraining.tv.hrm D/BluetoothDevice: getService, adapter sService=android.bluetooth.IBluetooth$Stub$Proxy@24d41de
2021-11-15 17:25:40.730 15112-16210/com.withutraining.tv.hrm D/BluetoothDevice: getService, sService=android.bluetooth.IBluetooth$Stub$Proxy@24d41de
2021-11-15 17:25:40.731 15112-16210/com.withutraining.tv.hrm D/BluetoothDevice: getService, adapter sService=android.bluetooth.IBluetooth$Stub$Proxy@24d41de
2021-11-15 17:25:45.912 15112-16210/com.withutraining.tv.hrm D/BluetoothDevice: getService, sService=android.bluetooth.IBluetooth$Stub$Proxy@24d41de
2021-11-15 17:25:45.912 15112-16210/com.withutraining.tv.hrm D/BluetoothDevice: getService, adapter sService=android.bluetooth.IBluetooth$Stub$Proxy@24d41de
2021-11-15 17:25:47.914 15112-16230/com.withutraining.tv.hrm D/ViewRootImpl[HrmTestActivity]: windowFocusChanged hasFocus=false inTouchMode=true

```
    fun startScan() {
        launch {
            scanner.advertisements.collect { advertisement ->
                foundDevices.add(
                    Device(
                        name = advertisement.name ?: "no name for device",
                    )
                )
                availableDevices.send(foundDevices)

                val peripheral = this.peripheral(advertisement) {
                    logging {
                        engine = SystemLogEngine
                        level = Logging.Level.Warnings
                        format = Logging.Format.Multiline
                        data = Hex
                    }
                }

                peripheral.connect()

                val characteristic = characteristicOf(
                    service = hrService,
                    characteristic = hrCharacteristic
                )

                val observation = peripheral.observe(characteristic)
                observation.collect { data ->
                    println("got new heart rate data")
                    val flag: Int =
[... cut ...] oothUtils.getIntValue(BluetoothGattCharacteristic.FORMAT_UINT8, 0, data)

                    val format = if (flag and 0x01 != 0) {
                        BluetoothGattCharacteristic.FORMAT_UINT16
                    } else {
                        BluetoothGattCharacteristic.FORMAT_UINT8
                    }
                    val heartRate: Int = BluetoothUtils.getIntValue(format, 1, data)
                    bpm.send(heartRate)
                }
            }
        }
    }

```
here is

AyoTheDev, 2021-11-15:
@twyatt 
After setting the logging level to Events, I'm seeing the onCharacteristicChanged being called as expected multiple times but its not coming through to the observation

twyatt, 2021-11-15:
The fact that Kable is logging the `onCharacteristicChanged` does seem like everything is working as expected.
Can you try changing your `collect` to read:

```kotlin
observation.collect { data ->
    println("got new heart rate data")
}
```

_In other words, nothing else in that `collect` other than the log indicating that it got data._
I'm mostly wondering if the `bpm.send(heartRate)` is problematic. Is that a `Channel`? Perhaps it is full and suspending there, thus preventing further collection?

---

As an aside: since you have the connection and data collection within the `scanner.advertisements.collect`, you're essentially having scanning remain active during the connection — I seem to remember that some Android phones have trouble with having connections open while scanning. It would probably be better to use a terminal operator on the scan (e.g. `first`) to find the peripheral of interest, then connect to that.

AyoTheDev, 2021-11-15:
Thanks for that having my collect read:

observation.collect { data ->
    println("got new heart rate data")
}

has it consistently updating

closed 2021-11-15 by AyoTheDev