JuulLabs/kable #190: Should not stop notifications/indications if not previously started

twyatt, 2021-11-16:
For `Peripheral.observe`, when subscribers goes from `1` to `0`, then underlying notifications/indications are stopped. Although this will also occur even if starting notifications/indications previously failed, or if connection drops. In both cases, stopping notifications/indications should not be attempted (the underlying BLE system will automatically clear notifications/indications).

Because these I/O operations are occurring when then shouldn't (aren't needed), the following exception is being thrown from the `observe` flow:

```
Caused by com.juul.kable.GattRequestRejectedException
       at com.juul.kable.PeripheralKt.setCharacteristicNotification(PeripheralKt.java:471)
       at com.juul.kable.PeripheralKt.access$setCharacteristicNotification(PeripheralKt.java:1)
       at com.juul.kable.AndroidPeripheral.stopObservation$core_release(AndroidPeripheral.java:391)
       at com.juul.kable.AndroidPeripheral$stopObservation$1.invokeSuspend(AndroidPeripheral.java:12)
       at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(BaseContinuationImpl.java:33)
       at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.java:104)
       at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.java:571)
       at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.java:750)
       at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.java:678)
       at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.java:665)
```

twyatt, 2021-12-05:
A potential fix is available as a `SNAPSHOT` build:

```
repositories {
    maven("https://oss.sonatype.org/content/repositories/snapshots")
}

dependencies {
    implementation("com.juul.kable:core:0.10.3-issue-190-3-SNAPSHOT")
}
```

ln-12, 2021-12-06:
Looks good in my use case, thanks :)

twyatt, 2021-12-07:
On an internal project, noticed that some BLE connections stalled after app was in the background for a long period of time with device coming in and out of range.

Not sure if it is a regression due to the changes in the `SNAPSHOT` mentioned above, but I'm going to do more testing before merging #193.

PR #193 (2022-01-18): Optimize observation handling
Observation handling has been reworked/optimized and moved to common code.

- `Peripheral.state` promoted to `StateFlow` (from `Flow`)
- ~Replaced usage of [`stately-isolate`](https://github.com/touchlab/Stately#stately-isolate) with Tuulbox's [synchronized map](https://github.com/JuulLabs/tuulbox#synchronizedmap)~
  - We still need stately until we #237
- `ConnectionLostException` and `NotReadyException` were changed to extend `NotConnectedException` (which extends `IOException`) — they previously directly extended `IOException`
- Connection related failures duri [... cut]
closed 2022-01-18 by twyatt