JuulLabs/kable #198: Query: Windows support?

nathan-castlehow, 2021-11-24:
Hey, I was just wondering how much investigation has been done in order to add windows support?
From my rudimentary google it looks like windows will be a giant pain so I presume that's why its not there already.

Is there interest in a windows version?

twyatt, 2021-11-24:
> it looks like windows will be a giant pain so I presume that's why its not there already.

Exactly this, or at least that is my understanding as well.

> Is there interest in a windows version?

There definitely is interest, though any near-term Windows support would have to come from outside contributors/PRs, as internally we don't have any near-term plans to support Windows.

twyatt, 2021-11-24:
If you happen to find, or create, a simple project that demonstrates using BLE from Kotlin on the Windows platform, then I'd use it as a reference for integrating support into Kable. 😄

nathan-castlehow, 2021-11-25:
@twyatt sweet. No worries. That all makes sense.
I have a couple of ideas so might give them a stab. It ultimately does look painful though so even if I get something working I'm not convinced it will be elegant.

I'll let you know how I go! :)

closed 2021-11-25 by nathan-castlehow