JuulLabs/kable #199: Keep Kable running within a foreground service when device screen is off

somq, 2021-11-25:
Hi there,

I'm trying to run Kable within a `LifecycleService` and collect advertisements while the device screen is off (sleeping).

I have to admit I'm having a hard time understanding how to perform this, though I'm able to run a custom Flow while the device screen is off, I can't get the same success with a simple scan.
It works while the app is in background but as soon as I shut the screen off or turn in back on, advertisements stop and start again.

```kt
val scanner = Scanner {}
scanner
.advertisements
.onEach { Log.i("TAG", "Scan advertisement => $it") }
.launchIn(lifecycleScope)
```

Here is a sample repo as an [example](https://github.com/somq/Stopwatch-Using-Coroutine-and-Foreground-Service)  
And the interesting part is [here](https://github.com/somq/Stopwatch-Using-Coroutine-and-Foreground-Service/blob/master/app/src/main/java/com/srmstudios/stopwatchtimer/service/StopwatchService.kt#L68)

Any help would be appreciated,
Thanks!

twyatt, 2021-11-28:
The flow will live as long as the scope that you launch it in.

I'm not certain of when `lifecycleScope` cancels within an Android service, but as you described it, it sounds like it cancels when the screen is off.

Do you have another scope available to `launchIn` that outlives the `lifecycleScope`? Perhaps try launching in `GlobalScope`? If you do use `GlobalScope`, you'll have to be careful to make sure you cancel the flow when you no longer want to scan.

somq, 2021-12-06:
Ok, after some debugging I've found a resolution.  
It appears that the Android BLE stack won't scan when device is sleeping if there is no services filtered.  
And this happens even with a foreground service running, which seems kind of counter-intuitive to me.  

It seems that additionally there is limitation applied on scans, because even with `ScanSettings` sets to `SCAN_MODE_LOW_LATENCY` there is a lot less scan results with screen off than with screen on.   

I say it "seems" because unfortunately I haven't been able to find any official documentation regarding this behaviour and I'm running out of time to jump into the Android code to find out...   

Anyway, thanks for your help @twyatt

twyatt, 2021-12-06:
Thanks @somq for following up with your findings! Really good info to know.
Sounds like the documentation should be updated to warn users of these pitfalls with Android scanning.
I'll re-open issue until someone has submitted a PR, or I've had time, to update the documentation.

Thanks again for reporting the issue and what ended up being the cause. ❤️

somq, 2021-12-06:
Your welcome!
This behavior exists since Android 8.1.
You can find some refs. [here](https://stackoverflow.com/questions/48077690/ble-scan-is-not-working-when-screen-is-off-on-android-8-1-0/48079800#48079800), and here is the [culprit](https://android.googlesource.com/platform/packages/apps/Bluetooth/+/319aeae6f4ebd13678b4f77375d1804978c4a1e1).   

I'm not sure how you'd like to shape this info into the docs so I'll let you do it.

twyatt, 2022-01-06:
I believe this has [been addressed](https://github.com/JuulLabs/kable/blob/8854ba2a1dc2460b3f4c9374a744ea27d67dbdae/core/src/androidMain/kotlin/Scanner.kt#L80), although the fix is not yet released (currently on the `main` branch).

Once the issues for [0.11.0 milestone](https://github.com/JuulLabs/kable/milestone/16) have been resolved, then that release _should_ have this fix.

twyatt, 2022-01-13:
[0.11.0](https://github.com/JuulLabs/kable/releases/tag/0.11.0) has been released and should include this fix.
Please re-open issue if you find it doesn't work.

closed 2022-01-13 by twyatt