JuulLabs/kable #200: Support for writing UBytes

boggsey, 2021-11-26:
I'm building an android app using Kable and need to send an unsigned 16-bit value to set a port number. I'm still getting used to working in Bytes, but I seem to be running into a problem. I attempted to use peripheral.write to write a uByteArray. However, it seems as though this is not currently possible as .write requires a ByteArray. Is there any interest in supporting uBytes in the near future?

Here is a basic example of how you can get different values for a port when using Byte vs UByte:
```
val port = 8883
    
asByte(port)
asUByte(port)

fun asByte(port: Int) {
    val arr = byteArrayOf(port.toByte())
    println(arr.contentToString())
}

fun asUByte(port: Int) {
    val arr = ubyteArrayOf(port.toUByte())
    println(arr.contentToString())
}
```
You'll get -77 asByte and 179 asUByte, with 179 being the correct value.

twyatt, 2021-12-15:
For your `asByte` and `asUByte` functions, I don't believe this will work as you expect. Since your `port` value is `8883` it can't be represented as either a `Byte` (`-128` to `127`) or `UByte` (`0` to `255`). So the conversion to `Byte` or `UByte` (`port.toByte` and `port.toUByte`) won't be able to hold the `8883` value in a single `Byte` or `UByte`.

That being said, I'm not aware of a simpler way of building a `ByteArray` from an `Int` (or `UInt`) other than doing a manual bit shifting technique as described [here](https://play.google.com/apps/internaltest/4698441401988954960) (a [similar technique](https://github.com/square/okio/blob/79656d80e812c36342add36580ca5784e5b8f637/okio/src/commonMain/kotlin/okio/internal/-Buffer.kt#L1119-L1130) is used in the popular [Okio](https://github.com/square/okio) library).

Here is a sample that demonstrates that the same bytes will be written regardless if your `port` is converted to an array of `Byte`s or `UByte`s:

```kotlin
fun main() {
    val port = 8883
    val hex = port.toByteArray().toHexString(separator = " ")
    val uhex = port.toUByteArray().toByteArray().toHexString(separator = " ")
    println(hex)
    println(uhex)
}

private fun Int.toByteArray() = ByteArray(4).also { array ->
    array[0] = (this shr 0).toByte()
    array[1] = (this shr 8).toByte()
    array[2] = (this shr 16).toByte()
    array[3] = (this shr 24).toByte()
}

private fun Int.toUByteArray() = UByteArray(4).also { array ->
    array[0] = (this shr 0).toUByte()
    array[1] = (this shr 8).toUByte()
    array[2] = (this shr 16).toUByte()
    array[3] = (this shr 24).toUByte()
}

// https://github.com/JuulLabs/tuulbox#encoding
fun ByteArray.toHexString(
    separator: String? = null,
    prefix: String? = null,
    lowerCase: Boolean = false
): String {
    if (size == 0) return ""
    val hexCode = if (lowerCase) "0123456789abcdef" else "0123456789ABCDEF"
    val capacity = size * (2 + (prefix?.length ?: 0)) + (size - 1) * (separator?.length ?: 0)
    val r = StringBuilder(capacity)
    for (b in this) {
        if (separator != null && r.isNotEmpty()) r.append(separator)
        if (prefix != null) r.append(prefix)
        r.append(hexCode[b.toInt() shr 4 and 0xF])
        r.append(hexCode[b.toInt() and 0xF])
    }
    return r.toString()
}
```

Outputs:

```
B3 22 00 00
B3 22 00 00
```

...all that is to say, unless Kotlin's [`UByteArray.toByteArray()` extension function](https://kotlinlang.org/api/latest/jvm/stdlib/kotlin.collections/to-byte-array.html) does some undesirable manipulation, I don't think there is any reason to provide functionality for writing unsigned byte arrays. As library consumers can convert unsigned byte arrays to signed byte arrays before writing.

Android does provide helper functions for setting characteristic values of unsigned types (so you don't have to manually do the bit shifting as shown above). It may make sense to have Kable provide some helper functions for building `ByteArray` data that will be written.

Something like:

```kotlin
interface ByteArrayBuilder {
    fun append(value: Int)
    fun append(value: UInt)
    fun append(value: Long)
    // ..
    fun append(text: String)
}

fun buildByteArray(builder: ByteArrayBuilder.() -> Unit): ByteArray = TODO()
```

This would allow library consumers to easily create `ByteArray`s for writing, in your example, this would allow you to do something like:

```kotlin
val port = 8883
val bytes = buildByteArray {
    append(port)
}
peripheral.write(bytes)
```

Did I address your issue? Would a helper like I mentioned be useful to you?

boggsey, 2022-02-10:
Sorry for my delayed response and thank you for your detailed response and examples. I ended up solving this via bit-shifting, as you suggested. The toHexString extension function was very helpful in visualizing what was being generated. Thanks for the help. It seems as though nothing needs to be done here until the need becomes stronger for some other reason.

closed 2022-02-10 by boggsey