JuulLabs/kable #208: CompletedContinuation cannot be cast to DispatchedContinuation

joharei, 2021-12-09:
I'm seeing this crash in production, but I haven't been able to reproduce it myself. I added more details to this issue already: Kotlin/kotlinx.coroutines#3005, but I wanted to hear if you have some insight.

<details>
<summary>Stacktrace</summary>

```
Fatal Exception: kotlin.NotImplementedError: Fatal exception in coroutines machinery for DispatchedContinuation[g0@e10a02f, Continuation at com.juul.kable.AndroidPeripheral.write(Peripheral.kt)@55aed1a]. Please read KDoc to 'handleFatalException' method and report this incident to maintainers
       at kotlinx.coroutines.DispatchedTask.handleFatalException(DispatchedTask.java:144)
       at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.java:115)
       at androidx.compose.ui.platform.AndroidUiDispatcher.performTrampolineDispatch(AndroidUiDispatcher.java:81)
       at androidx.compose.ui.platform.AndroidUiDispatcher.access$performTrampolineDispatch(AndroidUiDispatcher.java:41)
       at androidx.compose.ui.platform.AndroidUiDispatcher$dispatchCallback$1.run(AndroidUiDispatcher.java:57)
       at android.os.Handler.handleCallback(Handler.java:938)
       at android.os.Handler.dispatchMessage(Handler.java:99)
       at android.os.Looper.loop(Looper.java:247)
       at android.app.ActivityThread.main(ActivityThread.java:8656)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:602)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1130)
    Caused by java.lang.ClassCastException: kotlin.coroutines.jvm.internal.CompletedContinuation cannot be cast to kotlinx.coroutines.internal.DispatchedContinuation
       at kotlinx.coroutines.CoroutineDispatcher.releaseInterceptedContinuation(CoroutineDispatcher.java:107)
       at kotlin.coroutines.jvm.internal.ContinuationImpl.releaseIntercepted(ContinuationImpl.java:118)
       at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(BaseContinuationImpl.java:39)
       at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.java:233)
       at androidx.compose.ui.platform.AndroidUiDispatcher.performTrampolineDispatch(AndroidUiDispatcher.java:81)
       at androidx.compose.ui.platform.AndroidUiDispatcher.access$performTrampolineDispatch(AndroidUiDispatcher.java:41)
       at androidx.compose.ui.platform.AndroidUiDispatcher$dispatchCallback$1.run(AndroidUiDispatcher.java:57)
       at android.os.Handler.handleCallback(Handler.java:938)
       at android.os.Handler.dispatchMessage(Handler.java:99)
       at android.os.Looper.loop(Looper.java:247)
       at android.app.ActivityThread.main(ActivityThread.java:8656)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:602)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1130)
```
</details>

twyatt, 2021-12-09:
I don't know the cause, but can you provide the Kable version and Android version you're running?

Not sure how helpful it will be, but with that info I'll at least know the code path that is being taken in Kable and can try to create a reproducer.

joharei, 2021-12-10:
This was Kable version 0.10.3, and the error was reported from devices running Android 8-12.

Interestingly, I updated kotlin to 1.6.0 and coroutines to 1.6.0-RC and pushed a new release yesterday. So far this crash hasn't appeared in that release. I'll wait and see a bit longer, but that update might have fixed this. Will keep you posted.

joharei, 2021-12-16:
I can confirm that this crash disappeared after updating to Kotlin 1.6.0 and coroutines 1.6.0-RC.

closed 2021-12-16 by joharei