JuulLabs/kable #210: Expose Non-Connectable Advertising

miwright2, 2021-12-17:
Similar to Issue #65.  Is it possible to expose the connectable state of an advertising packet?  For example:

```
public actual class Advertisement(
    private val scanResult: ScanResult,
) {

    internal val bluetoothDevice: BluetoothDevice
        get() = scanResult.device

    public actual val isConnectable: Boolean
        get() = scanResult.isConnectable
...
}
```

[Similar information on iOS.](https://developer.apple.com/documentation/corebluetooth/cbadvertisementdataisconnectable)

twyatt, 2021-12-17:
Seems easy enough. Thanks for the link to iOS docs. I'll have to check if Web Bluetooth also offers this boolean, to allow for a common API across all the platforms.

twyatt, 2023-01-12:
Available in [0.21.0](https://github.com/JuulLabs/kable/releases/tag/0.21.0).

PR #404 (2022-12-14): Add property to `Advertisement` to indicate if peripheral is connectable
Closes #210
closed 2023-01-12 by twyatt