JuulLabs/kable #226: Can't reconnect after disconnect

TSkovsgaard, 2021-12-21:
Hi

On iOS when I call disconnect() the state changes directly to Disconnected, if I Connect again the state changes to Connected but on write if get the com.juul.kable.NotReadyException because the peripheral is disconnected

```
Debug: (Kermit) New state Disconnected(status=null)
Debug: (Kermit) New state com.juul.kable.State.Connecting.Bluetooth@83761768
Debug: (Kermit) New state com.juul.kable.State.Connecting.Services@83761528
Debug: (Kermit) New state com.juul.kable.State.Connected@83761718```

Now I call disconnect() the state never updates, if I call connect() again it updates directly to Connected

Debug: (Kermit) New state com.juul.kable.State.Connected@83761718

Now I write to the peripheral but I get the exception because the peripheral is disconnected.

com.juul.kable.NotReadyException

twyatt, 2021-12-21:
Thanks for the report. This is likely a regression from recent connection handling changes. I'll try to find some time to look into this soon.

twyatt, 2022-01-10:
Was blocked briefly due to https://github.com/JuulLabs/sensortag/issues/50. Found workaround, so I can again debug this issue. After I get #193 merged, I'll look into this again.

TSkovsgaard, 2022-01-18:
Hi @twyatt did you have any time looking into my issue yet ?

twyatt, 2022-01-18:
Just wrapping up on #193, so I'll try to look into it next.

twyatt, 2022-01-18:
@TSkovsgaard fixed in #246.

If you have a chance, can you confirm that #246 fixes the issue for you?:

```kotlin
repositories {
    maven("https://oss.sonatype.org/content/repositories/snapshots")
}

dependencies {
    implementation("com.juul.kable:core:0.11.0-issue-226-1-SNAPSHOT")
}
```

Thanks!

closed 2022-01-18 by twyatt

twyatt, 2022-01-18:
Fixed in [0.12.0](https://github.com/JuulLabs/kable/releases/tag/0.12.0).

TSkovsgaard, 2022-01-20:
Hi @twyatt thank you, I can confirm it works in version 0.12.0, great job 👍

PR #246 (2022-01-18): Prevent connection state monitor from being cancelled on disconnect
Added a dedicated connection `CoroutineScope`, so that in-flight connection attempts can be cancelled (via `Job.cancelChildren`) without cancelling the Coroutine that pipes `CentralManager` delegate callbacks to the `Peripheral`'s connection state `Flow`.

Closes #226
commit 99d71700 (2022-01-18): Prevent connection state monitor from being cancelled on disconnect (#246)

Closes #226