JuulLabs/kable #237: Have project to use the new memory model

twyatt, 2022-01-06:
https://blog.jetbrains.com/kotlin/2021/08/try-the-new-kotlin-native-memory-manager-development-preview/

volo-droid, 2023-03-01:
There's a new version of Stately available `2.0.0` (so far there's a release candidate `2.0.0-rc2`), which uses the new memory model, deprecates `stately-iso-collections`, and suggests to use [`stately-concurrent-collections`](https://github.com/touchlab/Stately#stately-concurrent-collections) instead. 

It would be cool if Kable migrates to that version (or alternatively removes Stately all together) otherwise it's impossible to use Stately concurrent collections together with Kable in one project because of a linking error for iOS/Mac targets:
```
e: Module "co.touchlab:stately-isolate" has a reference to symbol co.touchlab.stately.concurrency/GuardedStableRef|null[0]. Neither the module itself nor its dependencies contain such declaration.
```

twyatt, 2023-03-01:
Thanks for the info, I had planned to remove Stately entired but I was unaware of `stately-concurrent-collections`. That looks like a good choice. Super helpful that you pointed that out!
It is high up on my to do list to get these changes in.

twyatt, 2023-03-03:
@volo-droid I published a `SNAPSHOT` version for testing (from #437) that has Stately removed:

```
repositories {
    maven("https://oss.sonatype.org/content/repositories/snapshots")
}

dependencies {
    implementation("com.juul.kable:core:0.21.0-issue-237-1-SNAPSHOT")
}
```

volo-droid, 2023-03-08:
Thanks @twyatt . The project can be built with that version, Mac version crashes on consuming `scanner.advertisements` but I have to figure out if the issue is on our side or caused by the library.

PR #437 (2023-03-13): Remove `co.touchlab:stately-iso-collections` dependency
Closes #237
closed 2023-03-13 by twyatt