JuulLabs/kable #239: Scanner requires background location permission on Android 10 and 11

mark-dumontier, 2022-01-07:
When running on a device using Android 10 or 11, no advertisements are detected by `Scanner` unless the user has set their location permission to "Allow all the time" (i.e. the ACCESS_BACKGROUND_LOCATION permission). This is inconvenient for apps which only use scanning while in the foreground, particularly when [the user is required to manually change this permission in settings on Android 11](https://developer.android.com/training/location/permissions#request-background-location). 

Is it possible to configure `Scanner` to require only foreground location permissions on Android 10 and 11? If not, it might be helpful to include this permission requirement in the README. (It was not immediately obvious that detection of no advertisements was a permission issue, as documented in #103, especially since this issue only occurs on certain Android versions.)

twyatt, 2022-01-07:
I'm not super familiar with foreground vs. background permissions w/ respect to location permissions.

> Is it possible to configure `Scanner` to require only foreground location permissions on Android 10 and 11?

This seems feasible, assuming we can check/confirm we have [`ACCESS_BACKGROUND_LOCATION`](https://developer.android.com/reference/android/Manifest.permission#ACCESS_BACKGROUND_LOCATION] permission when the advertisements flow is subscribed to (i.e. in `onStart`) while the app is in the background.

The checking if the app is currently in the background looks like it may be the tricky part. [This](https://stackoverflow.com/questions/8489993/check-android-application-is-in-foreground-or-not) Stackoverflow post has various approaches, not sure if any work as would be needed for integrating into Kable, but I can definitely check it out once I have some time.

BluestormDNA, 2022-01-07:
Scanner doesn't need any background location permission by default. That would be overkill. In fact by today Google Play standards is a "delicate permission" where you need to go to https://support.google.com/googleplay/android-developer/answer/9799150?hl=en and ask for it manually to Google...

You should be ok with FINE_LOCATION on Android 10 and 11.

What are you trying exactly?

twyatt, 2022-01-07:
I have have misunderstood the original issue comment. @mark-dumontier, were you asking if we could remove the background permission requirement from Kable? It doesn't currently differentiate between foreground/background location permission. It is simply configured for fine location permission.

Though, you mentioned that there are situations where you weren't getting scan results. It would be good (if feasible) for Kable detect that situation to either throw and exception (if appropriate), or log a warning. If this is what you were referring to, then could you give some more detail as to what the scenario is that you might not get scan results, and ideally how to detect that situation would be super helpful. If it's not easily detectable then knowing a bit more would help me write some documentation about it in the readme (as you had mentioned) to help others.

mark-dumontier, 2022-01-08:
After investigating further, I've discovered that I was mistaken, and BACKGROUND_LOCATION wasn't the issue. 
(A different KMP library I was using for permissions handling was only asking for COARSE_LOCATION permission when getting bluetooth permissions, despite FINE_LOCATION permission being required on Android 10 and 11. My testing of granting BACKGROUND_LOCATION permission was accidentally having the side-effect of granting FINE_LOCATION, which caused the confusion.)

> Though, you mentioned that there are situations where you weren't getting scan results? It would be good (if feasible) for Kable detect that situation, to either throw an exception (if appropriate), or log a warning. If this is what you were referring to, then do you have more detail as to what the scenario it is that you might not get scan results? And, perhaps how to detect that situation would be super helpful. If it's not easily detectable, then knowing a bit more would help me write some documentation about it in the readme (as you had mentioned) to help others.

#103 describes my scenario exactly. My permissions were not correctly set (FINE_LOCATION was not allowed, which is required for bluetooth on Android 10 and 11) but no warning or error was being thrown by `Scanner`. Instead, `Scanner` appeared to be working, except no advertisements were being published by the advertisement flow. Having `Scanner` throw an error on initialization if the required permissions aren't set would be ideal. However, I'm not sure that even Android's `BluetoothLeScanner` throws an error; it looks like it might also just fail silently when permissions aren't set correctly. [Their documentation](https://developer.android.com/reference/android/bluetooth/le/BluetoothLeScanner#startScan(java.util.List%3Candroid.bluetooth.le.ScanFilter%3E,%20android.bluetooth.le.ScanSettings,%20android.app.PendingIntent)) says:

> An app must have ACCESS_COARSE_LOCATION permission in order to get results. An App targeting Android Q or later must have ACCESS_FINE_LOCATION permission in order to get results.

Doing a quick test myself by just calling ```BluetoothAdapter.getDefaultAdapter().bluetoothLeScanner.startScan(callback)```
while having all permissions denied doesn't throw an error. So getting proper feedback might be impossible without inspecting permissions directly. The best solution might be to just include a warning in the README that `Scanner` will fail silently if permissions aren't set correctly.

twyatt, 2022-01-12:
Thanks @mark-dumontier for the thorough investigation!

Despite Android silently failing when permissions are missing, I'd like Kable to be a bit more helpful in that regard, so I'll see if it is feasible to detect absence of permissions and throw an exception.

If it isn't feasible to detect missing permissions, then I'll update the README with an appropriate warning to users.

closed 2024-08-24 by twyatt