JuulLabs/kable #243: Drop JS `LEGACY` support

twyatt, 2022-01-12:


burnhamd, 2022-01-13:
Curious what is driving the decision to drop javascript?

twyatt, 2022-01-13:
Not dropping Javascript support, just dropping support for `LEGACY` compiler (and will only support [IR compiler](https://kotlinlang.org/docs/js-ir-compiler.html) moving forward).

twyatt, 2022-02-04:
Going to postpone dropping `LEGACY` support, as IR compiler is still considered Beta. Assuming no major hurdles/headaches arise, I'd like to continue to support `LEGACY` until IR compiler is announced as stable.

twyatt, 2022-03-23:
Watching https://youtrack.jetbrains.com/issue/KT-42289 for status updates on Kotlin/JS IR compiler becoming stable.

twyatt, 2022-12-30:
Kotlin/JS IR backend is considered stable w/ 1.8.0, and legacy backend is now deprecated.
Support for legacy backend will be removed once Kable upgrades its Kotlin dependency to 1.8.0.

PR #419 (2023-01-12): Drop `LEGACY` JavaScript compiler support
According to [KT-42289 comment](https://youtrack.jetbrains.com/issue/KT-42289/Make-the-new-JS-IR-backend-Stable#focus=Comments-27-6622643.0-0), Kotlin/JS IR backend is considered stable w/ 1.8.0, and legacy backend is now deprecated.

Closes #243
closed 2023-01-12 by twyatt