JuulLabs/kable #277: WillRestoreState for IOS CBCentral Manager

burnhamd, 2022-03-11:
I need to implement the willRestoreState delegate for the CBCentralManager for an application which makes use of background mode state restoration on IOS.

Has any thought been done on the architecture of this?

Currently I am thinking to add the following to the code base where the delegate produces a shared flow 
```
public data class CBManagerRestoredState(
    val peripherals: List<CBPeripheral>? = null,
    val scannUUIDs: List<CBUUID>? = null,
    //TBD Scan Options
)

val _willRestoreSharedFlow = MutableSharedFlow<CBManagerRestoredState?>()
```

My first thought is to expose this as a package wide variable since there is no Adapter/Manager class.

Alternatively I think it may also be useful to have an Adapter/Manager class for shared code to view things like bluetooth status and this might could go in the IOS implementation.

Lastly, implementing this may break MacOSX compatibility since to restore I will need to create the CBCentralManager with a key for restoring state and this is not accepted on OS X since the app is never suspended.

Wanted to get feedback before implementing in case there is a thought already for this

twyatt, 2022-03-11:
Thanks for creating this issue. You've identified an apparent gap in the current Kable API. I'm not sure the best way to approach it, but I'll try to find some time soon to dig deeper and give meaningful feedback.

Phoenix7351, 2023-08-21:
Hi, @burnhamd. We've added iOS background functionality in https://github.com/JuulLabs/kable/pull/551 . `willRestoreState` was implemented as a `No-op`, though, as it was required by the OS to be implemented by the delegate, but there was (seemingly) no work to be done to set back up the BLE stack. Does this cover this issue?

closed 2024-09-11 by twyatt